esp8266/Arduino #6750: New binary does not fit SPI Flash size even if free Flash size is 8x bigger!

jbrepogmailcom, 2019-11-10:
Please see following debug message, what do you recommend please? Thank you
Wemos D1 mini (4MB, no SPIFFS), Arduino IDE 1.8.10

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 459248
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 3710976
[httpUpdate]  - current Sketch Size: 462640
[httpUpdate] runUpdate flash...
[httpUpdate] New binary does not fit SPI Flash size

devyte, 2019-11-10:
Please take a moment to gather your thoughts, then read your post above, then compare it to the issue instructions you were presented when opening this.
Closing due to non compliant.

closed 2019-11-10 by devyte

jbrepogmailcom, 2019-11-11:
Hello I do not understand your advice. According to error message there is 3,710,976 bytes free space, the OTA update is 459,248 yet there is error that the binary does not fit. So what is wrong with my error report please?

devyte, 2019-11-11:
Open a new issue, and read the instructions carefully. Following the template and providing the required info is the minimum to accept an issue report.