esp8266/Arduino #6751: EEPROM size updated in latest core to 8192 bytes from previous 4096 bytes. Is it a breaking change ?

arihantdaga, 2019-11-11:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			bc60e97
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Nodemcu
- Flash Size: 		        4MB, FS 1MB
- lwip Variant:                v2 Higher Bandwidth]

### Problem Description
Prior to the version 2.6.0 EEPROM sector had size of 4096 bytes, same is in the docs also. I am using EEPROM Rotate library. And EEPROMr.reserved() function was returning value 1, Now its returning 2.
This function's definition has this - 
```
uint8_t EEPROM_Rotate::reserved() {
    uint32_t fs_end = (uint32_t)&_FS_end - 0x40200000;
    uint8_t sectors = (ESP.getFlashChipSize() - fs_end) / SPI_FLASH_SEC_SIZE - 4;
    return sectors;
}
```
I suspect that now value of FS_end has been changed ? If that happens it may affect the previous values store in EEPROM at particular positions. ? Is this a breaking change ? 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

arihantdaga, 2019-11-11:
In my espurna code where I used to get this earlier - 
```
[002150] [INIT] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[002157] [INIT] Flash size (SDK)    :  4194304 bytes / 1024 sectors (   0 to 1023)
[002164] [INIT] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[002171] [INIT] Firmware size       :   535712 bytes /  131 sectors (   1 to  131)
[002178] [INIT] Max OTA size        :  2605056 bytes /  636 sectors ( 132 to  767)
[002185] [INIT] SPIFFS size         :  1028096 bytes /  251 sectors ( 768 to 1018) // NOTE: THE LAST SECTOR IS TILL 1018. 
[002192] [INIT] EEPROM size         :     4096 bytes /    1 sectors (1019 to 1019)
[002200] [INIT] Reserved            :    16384 bytes /    4 sectors (1020 to 1023)
```

But now with the same code I get this - 
```
[075637] [INIT] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[075644] [INIT] Flash size (SDK)    :  4194304 bytes / 1024 sectors (   0 to 1023)
[075651] [INIT] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[075658] [INIT] Firmware size       :   529424 bytes /  130 sectors (   1 to  130)
[075665] [INIT] Max OTA size        :  2609152 bytes /  637 sectors ( 131 to  767)
[075673] [INIT] SPIFFS size         :  1024000 bytes /  250 sectors ( 768 to 1017) // NOTE THAT HERE SPIFFS Size is 250 sector and the last sector is 1017. 
[075680] [INIT] EEPROM size         :     8192 bytes /    2 sectors (1018 to 1019)
[075687] [INIT] Reserved            :    16384 bytes /    4 sectors (1020 to 1023)

```

earlephilhower, 2019-11-11:
Your code is making an incorrect assumption, unfortunately.  The EEPROM sector has always been defined in the Arduino core as the last sector of flash before the reserved area, and always 4K.

FS_end was busted in earlier releases because it could end up giving an incomplete final block to SPIFFS or LittleFS.  That half-size last block, in best case, was never used by the FS, but in the worst case might end up confusing the FS or overwriting the EEPROM sector.

devyte, 2019-11-11:
To be clear, with the FS_END fix it is possible in some cases for a gap to exist between the FS and the EEPROM start. The reserve calculation is running into that gap and incorrectly assuming it is part of the EEPROM space.

closed 2019-11-11 by devyte

d-a-v, 2019-11-11:
@arihantdaga Are you aware there is an `_EEPROM_start` symbol ?

arihantdaga, 2019-11-12:
Thank you for pointing at this. It'll help me fixing this.