esp8266/Arduino #6752: New binary does not fit SPI Flash size even if free Flash size is 8x bigger

jbrepogmailcom, 2019-11-11:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [-] I have tested that the issue is present in current master branch (aka latest git). - NO, LATEST GIT DOES NOT COMPILE, ENDS WITH ERROR - SEE ISSUE #9416
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 mini
- Core Version:			2.5.2
- Development Env:		[rduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			tried Generic ESP8266 and Wemos D1 mini r2
- Flash Mode:			no options given
- Flash Size: 		        tried all combinations [4MB/1MB]
- lwip Variant:                  v2 Lower Memory
- Reset Method:		no options given
- Flash Frequency:	        no options given
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:              http

### Problem Description

Detailed problem description goes here.

binary downloaded over OTA does not fit into available space. bin is 450kB, available space 3.7MB. The problem may be related to switching compile options from esp8266 to esp32 and back. It stopped working for no apparent reason - at some time it was upgrading OTA corerctly, then it stopped - with same binary:

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate] - code: 200
[httpUpdate] - len: 459248
[httpUpdate] ESP8266 info:
[httpUpdate] - free Space: 3710976
[httpUpdate] - current Sketch Size: 462640
[httpUpdate] runUpdate flash...
[httpUpdate] New binary does not fit SPI Flash size

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

```

### Debug Messages

```
Debug messages go here
```

devyte, 2019-11-19:
@jbrepogmailcom please reduce your sketch to something "M"inimal and "C"omplete with which we can "V"erify the issue.

jbrepogmailcom, 2019-11-19:
Problem solved. It turned out that after flash on first http update the ESP module downloaded whole update over 400kB correctly and updated. However, on all subsequential updates, it only downloaded first 10-15kB and then reported error "New bin does not fit". It is quite strange, because http server's error log did not show any problem. ESP just cut off downloading of bin and gave an error. I fixed it by changing this:

HTTPClient httpClient;
httpClient.begin( fwVersionURL );
.... some code ....
t_httpUpdate_return ret = ESPhttpUpdate.update( fwImageURL );

to this

WiFiClient wclient;
t_httpUpdate_return ret = ESPhttpUpdate.update( wclient, fwImageURL );

I would be glad if someone can explain me why it helped

jbrepogmailcom, 2019-12-23:
Hello, unfortunately the issue is not resolved. I am not using any SPIFFS. I would say that main problem is that download from server gets cut off after about 13kB, server responds with code 200 (OK) and ESP reports the bin does not fit. In the reality the bin should have around 450kB.

jbrepogmailcom, 2019-12-23:
I will reopen after I confirm it is not server or router problem

closed 2019-12-23 by jbrepogmailcom

lightning003, 2019-12-24:
Facing the same issue, have you solved it?

jbrepogmailcom, 2019-12-25:
Not yet. Seems that OTA example is working, but when used in real sketch it does not. If you can, please check your apache log to see if the bin file is sent completely or if server logs smaller file with OK 200 response code. Thanks

jbrepogmailcom, 2019-12-26:
In one case it clearly helped adding **wclient** to ESPhttpUpdate.update( wclient, fwImageURL ); as in post above, in other case the update is still not working

KimAnhDuc, 2024-10-17:
I use ESP8266 and 
I get an error when updating the firmware HTTP_UPDATE_FAILED Error (-107): New Binary Does Not Fit Flash Size while Sketch uses 326240 bytes (31%) of program storage space. Maximum is 1044464 bytes. Is there any way to fix it? Thank