esp8266/Arduino #6756: strange compile error using double: collect2.exe: error: ld returned 1 exit status

dsyleixa, 2019-11-11:
strange compile error using double: 
collect2.exe: error: ld returned 1 exit status
does not happen with float vs. double

Basic Infos

    [x ] This issue complies with the issue POLICY doc.
    [x ] I have read the documentation at readthedocs and the issue is not addressed there.
    [x ] I have tested that the issue is present in current master branch (aka latest git).
    [x ] I have searched the issue tracker for a similar issue.
    [ x] If there is a stack dump, I have decoded it.
    [x ] I have filled out all fields below.

Platform

    Hardware: ESP-12E
    Core Version: 2.5.2
    Development Env: [Arduino IDE
    Operating System: [Windows| 7/64 pro

Settings in IDE

    Module: nodemcu 1.0
    Flash Mode: ???
    Flash Size: 4MB no SPIFFS
    lwip Variant: ???
    Reset Method: ???
    Flash Frequency: [ 40Mhz]
    CPU Frequency: [80Mhz|]
    Upload Using: [SERIAL]
    Upload Speed: [115200|

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
// Neural Network for Arduino

// Ref.:
// http://robotics.hobbizine.com/arduinoann.html
// www.cs.bham.ac.uk/~jxb/NN/nn.html

// 3-layer Backpropagation net

// modified, extended, and enhanced by dsylexia
// required MCUs: ESP8266, ESP32
// version 0.1.1

// (C) 2018 by dsyleixa
// (C) of processed 3rd party code: see original sources.
// This example code is in the public domain for private use, 3rd party code not affected.
// Use for professional or business purpose only by personal written permission
// by the author.

// change log:
// 0.1.1: input matrix=void => output array=void {0,0,0,0,0,0,0,0,0,0}
//        input matrix pattern="0" => output array={1,0,0,0,0,0,0,0,0,0}
// 0.1.0: debug mode

#include <math.h>

// debug mode: uncomment, regular mode: outcomment!
//#define  DEBUG

#define  REPORT_N   20    // for terminal log
int32_t  ReportInterval;  // for terminal log
uint32_t timestamp;       // for terminal log

/******************************************************************
   Network Configuration, customizable
 ******************************************************************/
[... 1069 lines cut ...]

#ifdef DEBUG
  Serial.println ("DEBUG MODE: ACTIVE! ");
#endif

  while (true) {        // debug: loop forever
    delay(1000); // training finished
  }

}

```

### Debug Messages

```

d:/arduino/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x3fffd150 of C:\Users\hw\AppData\Local\Temp\arduino_build_160748/esp8266_BPN_011.ino.elf section `.bss' is not within region `dram0_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

```

earlephilhower, 2019-11-11:
You are out of RAM due to RODATA constants and very large heap arrays, so BSS can't fit w/in the ~50KB free for core use.

Try `float` instead of `double` (4 bytes vs. 8 bytes), moving constant arrays to PMEM (and using appropriate pgm_read_xxx macros), etc.

closed 2019-11-11 by earlephilhower

dsyleixa, 2019-11-11:
float has too few digits for convergence.
the error msg is confusing and ambguous:
How to retrieve the free RAM/stack/heap?

earlephilhower, 2019-11-11:
That's some serious dynamic range problem in the algorithm, ouch.

The error message is from GCC, not us, and it's pretty clear.  BSS == your uninitted memory arrays.  It doesn't fit in the RAM segment, so it can't link properly.

Just build an empty sketch and you'll see the total heap space available printed as part of the Arduino output ("Program uses XXX bytes out of YYYY of RAM" or similar).  Assume ~50KB total w/o any WiFi.  And that's obviously not going to include any dynamically allocated stuff.

Stack is 4K always.

dsyleixa, 2019-11-11:
I actually do not see or remember havon used dynamic memory at all.
A free mem function during runtime would be fine too (does actually exist for Arduino Due (Cortex M3 ATSAM38X), but does not work for ESP8266

earlephilhower, 2019-11-11:
`ESP.getFreeHeap()` is the runtime command.  It's not going to help you, though, if your heap variables are too big since you can't make an executable in that case.

devyte, 2019-11-11:
Your hidden weights matrix alone is 57KB or so, plus all the other stuff.

dsyleixa, 2019-11-11:
yes, Isee thanks.
I'll try to migrate to esp32.
At least I finally know now what the  starnge error msg means, thank you!