esp8266/Arduino #6760: ESP8266SdFat and SDFS linked when not used

TD-er, 2019-11-12:
Similar to #6691 
The PlatformIO inspect tool shows both libraries `ESP8266SdFat` and `SDFS` are linked in the binary and thus taking both around 20k of flash size. (debug build, release size may differ)

[image]

As far as I can see, all includes for `SD.h` in my code are in #ifdef wrappers and I cannot see where the file `SD.h` is included in the code of esp8266/Arduino.

Right now, I am using https://github.com/Jason2866/platform-espressif8266.git#core_2_6_0 as platform (has the latest build toolchain included).

Does any of you have an idea why it is linked?

Jason2866, 2019-11-12:
Not linked in Tasmota using https://github.com/Jason2866/platform-espressif8266.git#core_2_6_0 as platform
[image]

TD-er, 2019-11-12:
@Jason2866 You are not using SPIFFS in Tasmota, right?

Jason2866, 2019-11-12:
Ahhh, sorry overlooked (read not exactly...)
It is NOT the SPIFFS / LittleFs stuff. It is about `SDFS` and `ESP8266SdFat`

TD-er, 2019-11-12:
> It is NOT the SPIFFS / LittleFs stuff. It is about `SDFS` and `ESP8266SdFat`

Yep, but I wanted to check if there may be a link somewhere.
Like I said, I did check the source of this repo also and could not find an include of SD.h

d-a-v, 2019-11-12:
SDFS.h replaces SD.h

TD-er, 2019-11-12:
> SDFS.h replaces SD.h

Do you have any idea what these include and if it is needed when you don't use the SD functionality?

d-a-v, 2019-11-12:
I find none.
Can you copy-paste the linker line, and run it without SDFS library/object files, in the hope the linker will tell where the symbols are missing ?

TD-er, 2019-11-13:
I just renamed the files `SD.h` and `SDFS.h` so they would not be compiled and then the build completes just fine but without the libraries.

Clean build with the original file names:
[image]

Clean build with the 2 .h files renamed:
[image]

So I wonder, what could be wrong here? (maybe this is a PlatformIO bug? Linker bug?)

devyte, 2019-11-13:
> PlatformIO bug

Maybe retest building with Arduino IDE?

TD-er, 2019-11-13:
That's easier said than done.
The only way I know I could check then is by comparing sizes.
And is the build toolchain the same?

Jason2866, 2019-11-13:
@TD-er  Build chain for platformIO used (with my setup for core 2.6.0) is
https://github.com/earlephilhower/esp-quick-toolchain/releases/tag/2.5.0-4

TD-er, 2019-11-13:
Just as a test, I added those libraries to the platformIO `lib_ignore` list, which does help in no longer including it.
```
lib_ignore                = SD(esp8266), SDFS
```

Jason2866, 2019-11-13:
In Tasmota the behaviour you have does not occur
[image]

Misiu, 2019-11-14:
@TD-er adding those libraries to `lib_ignore` is more of a workaround. Maybe this is a bug in PlatformIO or build chain?

TD-er, 2019-11-14:
@Misiu Yep, I agree.
It was not reported as the final fix, but more to collect data for someone to interpret what may cause this issue.
And it does help me to get the build size down.

This issue (and the one with loosing all available RAM when including SD.h) has already taken a lot of time, so for now it is a quick work-around to make sure I can continue.

earlephilhower, 2019-11-14:
@TD-er, I think this is probably best handled at the Platform.IO GH site, no?  We're not seeing it on the Arduino IDE builds, and it seems like not everyone with PIO is seeing it, either.

TD-er, 2019-11-14:
I guess so. I will open one on the PIO github repo
Not sure where it is caused.

At least something is fishy with SD.h, see my other issue.  (which is reproducible in Arduino IDE)

earlephilhower, 2019-11-15:
OK, I'll close this on the Arduino side.  If something comes up w/PIO and they need Core changes, we can re-open and track.  The SD.h F() problem is repro'd and on its way in #6767 .

closed 2019-11-15 by earlephilhower

TD-er, 2019-11-15:
Let's first see if this linking is still an issue after the #6767 is fixed and if it is still an issue, I will add an issue on PIO's issues list for it.