esp8266/Arduino #6775: LittleFS files can not be accesses using ESP8266FtpServer

bill-orange, 2019-11-14:
- Hardware:				Adafruit Huzzah ESP8266
- Core Version:			2.6.1
- Development Env:		Arduino IDE 1.8.9
- Operating System:		Windows| 10
### Settings in IDE

- Module:  			        Huzzah ESP8266 Module
- Flash Mode:			        unknown
- Flash Size: 		                4MB
- lwip Variant:                          v2 Lower Memory
- Reset Method:		        unknown
- Flash Frequency:	                40Mhz
- CPU Frequency:		        80Mhz|
- Upload Using:		        SERIAL
- Upload Speed:                      115200

Problem Description: Files system created with LittleFS can not be accessed with FTPServer

To avoid the possibility of weird memory interactions, I started with the Huzzah completely erased from the command line.

Detailed problem description: Both LittleFS and FTP file Server appear to be running normally.  Files in LittleFS can be created and appended as expected.  FTP Server appears to run normally.  A connection to it can be quickly achieved.  The problem is that the files created with LittleFS are not 'seen' by the FTP Sever.  In other words the FTP server directory is empty.

Example: I am not providing a debug screen since no errors are reported.  The offending sketch requires serial hardware not likly to be possessed by a tester. 

Additional Information:

Is LittleFS not designed to work with FTP Server?  Should another FTP Server library be used? Is there an FTP setting beyond the defaults that must be adjusted.

earlephilhower, 2019-11-14:
Without an example you're not going to be able to get much help.

That said, the FTP server library probably simply needs to have a search/replace done on SPIFFS->LittleFS.  File opens, directory lists, etc. all come from the FS object, and it probably is hardcoded in the library as SPIFFS.  The LittleFS API is the same, so it's a matter of search and replace (possibly including <LittleFS.h> in the source files of course).

bill-orange, 2019-11-14:
@earlephilhower So you are saying the LittleFS library needs to be modified for my needs?

earlephilhower, 2019-11-14:
FTP library needs to use LittleFS.openDir()/etc. instead of SPIFFS.openDir().

A good library would take the FS* as an input on creation, and just use that and let you go between as you choose (it would also then work with SD cards, too, with SDFS).

bill-orange, 2019-11-14:
@earlephilhower Got It, I will write up a sketch illustrating the problem.  It should only take a few minutes.

earlephilhower, 2019-11-14:
@bill-orange Got it, but the problem lies not in the Arduino code, but in the ftpServer itself.

I assume the lib is here:   https://github.com/nailbuster/esp8266FTPServer

If you look at the library, it hardcodes SPIFFS.xxxx everywhere.  Just edit that library to swap in LittleFS.xxx and you have a reasonably good chance of making it work.  Nothing we can do on our end if the library is specifically calling SPIFFS.xxxx.

devyte, 2019-11-14:
I don't see a core issue here. Closing.

On a side note, I wasn't aware of that ftp lib. It looks good, but needs some work to improve it in terms of language use, heap use, generalization of the FS to use, and externalization of the specific WiFiClient/server object to use (e.g.: allow passing a WiFiClientSecure, like HttpClient). It would be interesting if somebody pursued that.
Reference: #1183 .

closed 2019-11-14 by devyte

bill-orange, 2019-11-14:
@earlephilhower Sadly, it does not appear to be that simple.  I substituted LittleFS for SPIFFS wherever it was used and added the LittleFS library to the .cpp file.  It compiles and runs but the directory lists hello.txt as ello.txt and it is inaccessible. This is likely to be beyond my skill level to fix.

hojatm61, 2020-02-10:
Hi.
Inside the ESP8266FtpServer.cpp file
// fn.remove (0, 1);
Comment. Your problem will be solved.

jmwislez, 2020-10-13:
Updated it to do LITTLEFS/SPIFFS on ESP8266/ESP32: https://github.com/jmwislez/ESP32FtpServer

bill-orange, 2020-11-01:
Works like a charm. Thanks!

kmsmi, 2020-11-10:
> Updated it to do LITTLEFS/SPIFFS on ESP8266/ESP32: https://github.com/jmwislez/ESP32FtpServer
I have a problem with this library.
I can't upload files that are slightly larger than 4 KB.
This is the problem with SPIFFS and LittleFS.

bill-orange, 2020-11-10:
I am using FTP for files around 120k just fine.  Could you post a sketch that is not working for you?  I can give it a try.

kmsmi, 2020-11-10:
A simple sketch.
`#include <ESPFtpServer.h>`
`#include <LittleFS.h>`
`ftpSrv.begin(config_sys.ftp_login, config_sys.ftp_pass);`
`ftpSrv.handleFTP(LittleFS);`

ftp works and all files are visible, but files larger than 4 KB are written with an error.

bill-orange, 2020-11-10:
Ah, I am using a different littleFS library.  Try this one (note the caps in the .h file).

https://github.com/lorol/LITTLEFS

This code runs fine with bigger files.

```
#include <ESPFtpServer.h>
#include <LITTLEFS.h> 
#include <WiFi.h>  
#define FS_NAME "LittleFS"
#define FS_ID LITTLEFS
#define FORMAT_LITTLEFS_IF_FAILED false  // if false don't reformat if something goes wrong

const char* ssid = "yours";          // WiFi credentials
const char* password = "yours";     // WiFi credentials
FtpServer ftpSrv;  

void setup() {
  Serial.begin (115200);  // Standart intro boilerplate follows
  WiFi.begin (ssid, password);
  Serial.println ("");
  Serial.print ("Opening WiFi Connection ");

  // Wait for connection
  while (WiFi.status () != WL_CONNECTED) {
    delay (500);
    Serial.print (".");
  }
  Serial.println ("");
  Serial.print ("Connected to ");
  Serial.println (ssid);
  Serial.print ("IP address: ");
  Serial.println (WiFi.localIP ());

  if (!LITTLEFS.begin(FORMAT_LITTLEFS_IF_FAILED)) {
    Serial.println("LITTLEFS Mount Failed");
    return;
  }
  // Set-up FTP
  if (FS_ID.begin ()) {
    Serial.println ("File system opened (" + String (FS_NAME) + ")");
    ftpSrv.begin ("esp32", "esp32");    //username, password for ftpon default port 21
  }
  else {
    Serial.println ("File system could not be opened; ftp server will not work");
  }
}
void loop() {
  ftpSrv.handleFTP(LITTLEFS);
}
```

kmsmi, 2020-12-17:
I solved my problem with this library [https://github.com/jmwislez/ESP32FtpServer](url)
Changed the boolean `boolean FtpServer::doStore()`

`boolean FtpServer::doStore()`
`{`
`  // Avoid blocking by never reading more bytes than are available`
`  int navail = data.available();`

`  if (navail > 0)`
`  {`
`    // And be sure not to overflow buf.`
`    if (navail > FTP_BUF_SIZE)`
`      navail = FTP_BUF_SIZE;`
`    int16_t nb = data.read((uint8_t *)buf, navail);`
`    // int16_t nb = data.readBytes((uint8_t*) buf, FTP_BUF_SIZE );`
`    if (nb > 0)`
`    {`
`      // Serial.println( millis() << " " << nb << endl;`
`      file.write((uint8_t *)buf, nb);`
`      bytesTransferred += nb;`
`    }`
`  }`
`  if (!data.connected() && (navail <= 0))`
`  {`
 `   closeTransfer();`
 `   return false;`
`  }`
`  else`
`  {`
`    return true;`
`  }`
`}`

Now it works fine.

jmwislez, 2020-12-18:
Thank you @kmsmi ! I applied your patch in https://github.com/jmwislez/ESPFtpServer.