esp8266/Arduino #6777: Pull (#6765) breaks arduino-esp8266fs-plugin functoinality

gnu-emu, 2019-11-15:
Hi,
First time as active github contributor here so please forgive if not everything is correct.

The change in upload.py has breaks arduino-esp8266fs-plugin functoinality. 

The plugin esp8266fs.jar executes:

....upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 921600 write_flash 0x200000/tmp/something.bin --end

... so that esptool gives out:

esptool write_flash: error: argument <address> <filename>: Must be pairs of an address and the binary filename to write there

The problem seems to be that the change in upload.py.

cmdline in "esptool.main(cmdline)" becomes:

--chip', 'esp8266', '--port', '/dev/ttyUSB0', '--baud', '460800', 'write_flash', '0x200000', '--end', '0x0', '/tmp/something.bin'

... so that '--end' is passed to esptool and '0x0' is passed aditionally but should not.

devyte, 2019-11-15:
It's not correct, you ignored the issue template which requires certain information. Please edit your post above, add the template, follow the instructions in it, and fill out the required info.

devyte, 2019-11-15:
@earlephilhower I think a new plugin version is needed

gnu-emu, 2019-11-15:
Looks like changing some lines in ESP8266FS.java of arduino-esp8266fs-plugin to directly invoke esptool.py is the best and easiest solution. For me it's working. I'll try to send my proposal there.

gnu-emu, 2019-11-15:
> It's not correct, you ignored the issue template which requires certain information. Please edit your post above, add the template, follow the instructions in it, and fill out the required info.

I hope it's ok now. Thanks.

devyte, 2019-11-15:
It's enough, and much better than other cases. Thank you! I fixed the one thing missing: a reference to the PR that broke the plugins.

Strictly speaking you're correct to close this issue and raise the problem in the other repo(s). However, I consider this case to be a bit special:
- development is hampered while the plugins are broken. This makes for a bit of urgency. I'm in this boat.
- there are two plugins involved: the SPIFFS one and the LittleFS one. I believe @earlephilhower is handling them. 
- I don't know if the SPIFFS plugin is being updated in the original repo or in @earlephilhower 's fork. 

All of this means that special coordination and tracking is needed, so I'm keeping this issue open.

In addition, the README needs to be updated:
 * add a link to both plugins in the Other links section
 * replace the esptool-ck link with esptool.py, or add the esptool.py link in addition to the other
 * add a quick link at the top to jump to the Other Useful Links section at the bottom

devyte, 2019-11-15:
LittleFS plugin repo: https://github.com/earlephilhower/arduino-esp8266littlefs-plugin
SPIFFS plugin repo fork: https://github.com/earlephilhower/arduino-esp8266fs-plugin

d-a-v, 2019-11-15:
We can also signore `--end` in `upload.py`.
PR is welcome

gnu-emu, 2019-11-15:
Then there is still:

cmdline = cmdline + ['0x0', binary]

... that adds the '0x0'...

Changing plugins to directly invoke esptool.py seems to me to be the easiest fix (of course it's a beginners opinion :)

gnu-emu, 2019-11-15:
And now looking a second time at the code my problem description in the first post seems to be incomplete.
It should be:

cmdline in "esptool.main(cmdline)" becomes:

['--chip', 'esp8266', '--port', '/dev/ttyUSB0', '--baud', '460800', 'write_flash', '0x200000', '--end', 'write_flash', '0x0', '/tmp/arduino_build_949720/something.bin']

...so 'write_flash' is passed two times too because of:

cmdline = cmdline + ['write_flash'].

gnu-emu, 2019-11-15:
Changing:
File uploadPyFile = new File(platform.getFolder()+"/tools", "upload.py"); 

to
   
File uploadPyFile = new File(platform.getFolder()+"/tools/esptool", "esptool.py");

and

sysExec(new String[]{pythonCmd, uploadCmd, "--chip", "esp8266", "--port", serialPort, "--baud", uploadSpeed, "write_flash", uploadAddress, imagePath, "--end"});

to

sysExec(new String[]{pythonCmd, uploadCmd, "--chip", "esp8266", "--port", serialPort, "--baud", "460800", "write_flash", uploadAddress, imagePath});

in ESP8266FS.java of the arduino-esp8266fs-plugin was the way to fix it for me. It needs further investigation if it works for every OS and every condition.

Two lines of code :)

d-a-v, 2019-11-15:
It is now time to make your pull-requests on these repositories !

earlephilhower, 2019-11-18:
Closing as fixed w/#6788

closed 2019-11-18 by earlephilhower