esp8266/Arduino #6778: redirect location gets cleared in setURL(const String& url) function of ESP8266HTTPClient.cpp

Suraj151, 2019-11-15:
i updated to latest release 2.6.0 before my last ota update to devices and now devices are not updating as they do before. i go through debug process where i found subjected issue. 

below line return false for redirects in ota update

https://github.com/esp8266/Arduino/blob/05454df1641b47e99a03de848e64936e3cf4b62f/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L555

so i did serial debug at
```
bool HTTPClient::setURL(const String& url)
{
    // if the new location is only a path then only update the URI
    if (url && url[0] == '/') {
        _uri = url;
        clear();
        return true;
    }

    if (!url.startsWith(_protocol + ':')) {
        DEBUG_HTTPCLIENT("[HTTP-Client][setURL] new URL not the same protocol, expected '%s', URL: '%s'\n", _protocol.c_str(), url.c_str());
        return false;
    }
    // disconnect but preserve _client
    Serial.printf("check url1: %s\n", url.c_str());
    disconnect(true);
    Serial.printf("check url2: %s\n", url.c_str());
    clear();
    Serial.printf("check url3: %s\n", url.c_str());
    return beginInternal(url, nullptr);
}

```
and at

```

bool HTTPClient::beginInternal(const String& __url, const char* expectedProtocol)
{
    String url(__url);

    DEBUG_HTTPCLIENT("[HTTP-Client][begin] url: %s\n", url.c_str());
    clear();
    Serial.printf("check url4: %s\n", url.c_str());
    // check for : (http: or https:
    int index = url.indexOf(':');
    if(index < 0) {
        DEBUG_HTTPCLIENT("[HTTP-Client][begin] failed to parse protocol\n");
        return false;
    }

```

where i found serial debug output as
```
check url4: http://something
check url1: http://something
check url2: http://something
check url3: 
check url4: 

```
and resulted that **clear()** function clears the location url before pass it to **beginInterval** function.
don't know much why it is cleared before passing, since **beginInterval** function calls **clear** at beginning.
i get it done by commenting clear().

d-a-v, 2019-11-15:
@liebman can you please have a look to this issue ?
https://github.com/esp8266/Arduino/blame/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L554

liebman, 2019-11-15:
OK - this is because the parameter to `setURL`was changed to to a `const String&` in #6571. And given that `beginInternal` both forces a local copy of the url and calls `clear` I'll put a PR togeather that removes the `clear` just before `beginInternal` from `setURL`.

PR #6786 (2019-11-18): fix http client redirects (broken from const String& changes)
remove `clear()` before `beginInternal()` so _location is not cleared before its passed to `beginInternal()`

Fixes #6778
closed 2019-11-18 by devyte