esp8266/Arduino #6779: fatal: not a git repository (or any of the parent directories): .git on esp8266

GeorgeMpouk, 2019-11-15:
Hi , i just bought the esp8266 so i'm really new to this kind of stuff , after i did everything that i saw on a youtube video on how to set it up i tried to upload the blink sketch to it as video said , but it threw this error at first : "fatal: not a git repository (or any of the parent directories): .git "
while compiling and then it threw this
[image]

i have a macbook pro 2013 running mojave and the arduino ide version is the 1.8.10

Thank you in advance

d-a-v, 2019-11-15:
Are you using the esp8266 arduino core version 2.6.1 ?

TD-er, 2019-11-15:
This also happens if I use this one in PlatformIO:  (made by @Jason2866 )
```
platform                  = platform = https://github.com/Jason2866/platform-espressif8266.git#core_2_6_1
```

Build error in PIO:
```
Processing debug_custom_ESP8266_4M1M (platform: platform = https://github.com/Jason2866/platform-espressif8266.git#core_2_6_1; board: esp12e; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

PlatformManager: Installing platform 
git version 2.21.0.windows.1
Cloning into 'C:\Users\gijs\.platformio\platforms\_tmp_installing-u2kv50i3-package'...
fatal: protocol ' https' is not supported
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'core_2_6_1', ' https://github.com/Jason2866/platform-espressif8266.git', 'C:\\Users\\gijs\\.platformio\\platforms\\_tmp_installing-u2kv50i3-package']
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

d-a-v, 2019-11-15:
@TD-er 
maybe with :?
```
platform=https://github.com/Jason2866/platform-espressif8266.git#core_2_6_1
```
instead of
```
platform = https://github.com/Jason2866/platform-espressif8266.git#core_2_6_1
```
I don't think it is related with OP anyway

TD-er, 2019-11-15:
@d-a-v You are right. I had an extra `platform =` statement in my line.

I will get some coffee now....

GeorgeMpouk, 2019-11-15:
> @GeorgeMpouk Are you using the esp8266 arduino core version 2.6.1 ?

No , i'm using the 2.6.0

GeorgeMpouk, 2019-11-15:
@d-a-v hi , thank you for your response , i'm using the 2.6.0 version .

d-a-v, 2019-11-15:
This issue is fixed in version 2.6.1.

GeorgeMpouk, 2019-11-15:
hi i no longer have the git repository issue but i have this issue still
[image]

earlephilhower, 2019-11-16:
Does `ls -l /dev/cu.usbmodem14101` actually show a port?

If not, there's nothing to do here, you just need to select the proper port from the IDE (it can change if you plug/unplug things).

If it does exist, make sure your user on the Mac has access permissions.  Your user might not be part of the CU group or whatever.

The best way to get additional help with this would be to go to https://gitter.im/esp8266/Arduino or use the forum on https://esp8266.com

Closing as I'm not seeing a core issue here.  If some other Mac users are hitting the same thing with the latest release, do chime in here and we'll see if it merits reopening.

closed 2019-11-16 by earlephilhower