esp8266/Arduino #6799: Slow Webserver response after OTA update

caipifrosch, 2019-11-18:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12, custom board]
- Core Version:			[2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module, custom board]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[none]
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA]
- Upload Speed:   

### Problem Description

OTA update with new device firmware results in slow station mode webserver response. Also connection with AP is not reliable after update.

In general, both firmware versions do work smoothly. Just after OTA update firmware 2 will be slow. Firmware 2 is ok again when first serial port flashing firmware 1 and firmware 2 afterwards. 

1. Firmware 1 is stable (core 2.4.0)
2. After OTA update with Firmware 2 webserver is slow (core 2.5.2)
3. Repeated flash update with firmware 2 via serial port will still be slow
4. Serial flash update firmware 1 will then be ok again
5. Finally flash update with firmware 2 will still be ok

Why does it require to first flash firmware 1 before firmware 2 works fine again?

Are there i.e any changes in flash wifi settings made by the core libraries?

Thanks a lot for your ideas in advance.

d-a-v, 2019-11-18:
Please try again with latest release 2.6.1

caipifrosch, 2019-11-19:
Same after OTA update to release 2.6.1.

jandrassy, 2019-11-19:
I suspect something is different in SDK init data or init data location after 2.4.2
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_phy.cpp
https://github.com/esp8266/Arduino/commit/f706c83b66261df58b7d6148e343fd21cbd77206#diff-db6c0fed822f374f3e4a504405bdf278

devyte, 2019-11-19:
What happens if you have 2.4.2, then you flash 2.6.x over serial with full flash wipe enabled?

caipifrosch, 2019-11-27:
@devyte: Yes, that seemed to work. Erasing flash on laggy firmware 2 device and serially updating to firmware 2 worked. 

With this in mind, it looks like OTA update with firmware 2 works fine again when disabling these two lines of code in firmware 2 sketch:

//WiFi.setAutoConnect(true);
//WiFi.persistent(false);

Firmware 1 has WiFi.persistent(true);
Firmware 2 OTA with WiFi.persistent(false); did not work
Firmware 2 OTA with WiFi.persistent(true); seems to work ok again

Any idea how this could make a difference? 

Are maybe settings written earlier by firmware 1 (core 2.4.0) somehow incompatible to firmware 2 (core 2.6.1)?

Any help/hint is highly appreciated.

closed 2021-03-23 by earlephilhower