esp8266/Arduino #6803: WPA2 enterprise with credentials auth ok with radius but still get STATION_WRONG_PASSWORD after 30s

simkard69, 2019-11-18:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12E / NodeMCU v3]
- Core Version:			[2.6.1] (esp8266 by ESP8266 Community)
- Development Env:		[Arduino IDE 1.8.10]
- Operating System:		[Windows 10 x64]

### Settings in IDE

- Module:  			[NodeMCU 1.0 (ESP-12E Module)]
- Flash Mode:			[???]
- Flash Size: 			[4MB]
- lwip Variant:   		[v2 Lower Memory]
- Reset Method:		[??? (probably NodeMCU)]
- Flash Frequency:		[???]
- CPU Frequency:		[80Mhz]
- Upload Using:		[USB/SERIAL]
- Upload Speed:		[115200 and/or 256000] (serial upload only)

### Problem Description

Wifi router is a Synology RT1900ac with WPA2 Enterprise authentication enabled.
RADIUS server is a NPS running on Windows Server 2019.
Policies are configured with :
	- Conditions > NAS Port Type > Wireless - Other OR Wireless - IEEE 802.11
	- Conditions > NAS Identifier > <MAC ADDRESS FROM ROUTER WIFI INTERFACE>
	- Conditions > Calling Station ID > <MAC ADDRESS FROM NodeMCU/ESP8266>
	- Conditions > Windows Groups > <ACTIVE DIRECTORY GROUP THAT CONTAINS THE AUTHENTICATING USER>
	- Constraints > Authentication Methods > Microsoft Secure Password (EAP-MSCHAP v2)

Behavior of code (chronologically) :
	- [ESP8266] boots up then tries to associate/authenticate
	- [RADIUS/NPS] receives AUTH and grant access
	- [ESP8266] On the first 30 seconds, "wifi_station_get_connect_status" == 1 ()
	- [ESP8266] Then after 2nd AUTH which takes place at 30s after 1st one : "wifi_station_get_connect_status" switch to == 2 (STATION_WRONG_PASSWORD)
[... 148 lines cut ...]
      Serial.print(WiFi_status, DEC);
    } else {
      Serial.print(".");
    }

    // Restart ESP8266 if no WiFi association has been successful
    if (millis() - millis_WiFi_association_start >= millis_WiFi_association_reset){
      ESP.restart();
    }
    
    delay(500);
  }
  
  
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  IPAddress myAddr = WiFi.localIP();
}

void loop() {

}
```

d-a-v, 2019-11-18:
We have no control on this API.
Just to be sure, you may try again with lwip1 and also with this call anywhere in `setup()`:
```
disable_extra4k_at_link_time();
```
If it still does not work, what you would have to do is to make a native nonos-sdk test (branch 2.2.x - the one we use) and check if this issue is still there. And if it is, try with their branch 3.x. And if it is still there, then open an issue on their repository.
You can also ask for support and examples in [their rtos-sdk](https://github.com/espressif/ESP8266_RTOS_SDK/issues/635).

devyte, 2019-11-19:
WPA2 isn't fully supported by the sdk (we're still waiting for a working example of the limited use case), and we don't have core code on our side to wrap it. The one thing to try is what @d-a-v says above, which disables an optimization that returns the stacks to their normal places. That allows more stack space for SYS, just in case the optimized reduced stack being blown up is what you're running into.
Closing due to feature not supported by core.

closed 2019-11-19 by devyte

simkard69, 2019-11-19:
> We have no control on this API.
> Just to be sure, you may try again with lwip1 and also with this call anywhere in `setup()`:
>
> ```
> disable_extra4k_at_link_time();
> ```
>
> If it still does not work, what [...]

Just for me to understand what I should be doing regarding the nonos-sdk versions.
Can you point me to some tutorial regarding that part ?

Thanks

simkard69, 2019-11-19:
I had to include this code in order for the Arduino IDE to find the function "disable_extra4k_at_link_time()"
`
#include <coredecls.h>
`

... But unfortunately, no change to be reported on the wifi association/authentication behavior.

d-a-v, 2019-11-19:
This is not lwip1, lwip1 is selectable in arduino IDE menus or environment variables in PIO
```
disable_extra4k_at_link_time(); // use official/legacy ram location for user stack
```

nonos-sdk example: https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/examples/wpa2_enterprise

maybe it can help (maybe not): https://github.com/d-a-v/esp8266-nonos-easy-sdk (unmaintained)

simkard69, 2019-11-19:
I tried to enable "debug level" on "Wifi" to check for any further/complementary messages, and I got these :

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld
⸮#d`;⸮olph
SDK:2.2.2-dev(5ab15d1)/Core:2.6.1=20601000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 
state: 5 -> 0 (0)
rm 0
WPA2 ENTERPRISE VERSION: [v2.0] enable

Connecting to SSID [<AP SSID GOES HERE>] : .wifi evt: 1
STA disconnect: 8
wifi evt: 8
.....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 
EAP-MSCHAPV2: RX identifier 1 mschapv2_id 1
EAP-MSCHAPV2: Generate Challenge Response
EAP-MSCHAPV2: RX identifier 2 mschapv2_id 1
....................pm open,type:2 0
.......................................state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
wifi evt: 1
STA disconnect: 204
2scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 
EAP-MSCHAPV2: RX identifier 1 mschapv2_id 1
EAP-MSCHAPV2: Generate Challenge Response
EAP-MSCHAPV2: RX identifier 2 mschapv2_id 1
222 (These "2" are here to indicate the value of "wifi_station_get_connect_status" which corresponds to "STATION_WRONG_PASSWORD")
```

Seems like EAP-MSCHAPV2 is taking place correctly, but then, nothing happens.

I have some doubts about the fact that "wifi_station_get_connect_status" updates correctly its status value once associated.
Can this be possible ?

Or maybe I'm wrong and the following sequency just means that there is some problem with the authentication on RADIUS side (which on RADIUS event log doesn't seem to be ...) :
```
EAP-MSCHAPV2: RX identifier 1 mschapv2_id 1
EAP-MSCHAPV2: Generate Challenge Response
EAP-MSCHAPV2: RX identifier 2 mschapv2_id 1
```

d-a-v, 2019-11-19:
I *personally* never observed WPA2 working.
This code is FW-closed-source so
- we cannot do anything
- the best move is to show them a pure non-working nonos-sdk program like in OP+1, with maximum informations. In some places they propose the linux-hostapd configuration they use to test their with WPA2 FW. I think one should try with it and report with the same configuration ([one of the places](https://github.com/espressif/ESP8266_RTOS_SDK/issues/442#issuecomment-451918005)).

simkard69, 2019-11-19:
There is some interesting information on this link
[joostd/esp8266-eduroam](https://github.com/joostd/esp8266-eduroam)

Seems like it is stated that :
```
Notes:
use version 1.5.2 or later of the espressif sdk for EAP-TLS support.
use version 2.0.0 or later of the espressif sdk for PEAP or EAP-TTLS support.

PEAP still has issues that need to be resolved:
the EAP outer identity seems fixed to anonymous@espressif.com.
MS-Chapv2 authentication fails for some reason.
```

Something is strange by the way because it seems that I can't change the authentication protocol used from EAP-MSCHAPV2 to something else.

Actually my wifi clients authenticates fine with "Microsoft : Protected EAP (PEAP)".
On NPS, when I disable the "Microsoft: Secured password (EAP-MSCHAP v2), I get access denied for the ESP8266 trying to authenticates with the following reason :
`Reason:				The client and server cannot communicate, because they do not possess a common algorithm.`

simkard69, 2019-11-19:
Seems like the only to get past this problem is to upgrade to NONOS SDK v3 / v3.0.0
I know that there might be some paths to go there, but just wondering when should we expect this SDK release coming to this esp8266 arduino package ?

Thanks for your hard work guys !