esp8266/Arduino #6805: Board specific defines are generic for some boards

sprior, 2019-11-19:
I have noticed that ARDUINO_BOARD and the ARDUINO_**** #defines are nicely specific for some variants of ESP8266 but frustratingly generic on others.  

Note that there is no way to use the current definitions to differentiate between NodeMCU 0.9 and NodeMCU 1.0 or the Adafruit HUZZAH.

Unless there is a compelling reason not to I'd suggest that it is desirable to be able to use preprocessor ifdefs to detect each variant of device.

Some examples from boards.txt are:

huzzah.name=Adafruit Feather HUZZAH ESP8266
huzzah.build.board=ESP8266_ESP12

inventone.name=Invent One
inventone.build.board=ESP8266_GENERIC

cw01.name=XinaBox CW01
cw01.build.board=ESP8266_GENERIC

nodemcu.name=NodeMCU 0.9 (ESP-12 Module)
nodemcu.build.board=ESP8266_NODEMCU

nodemcuv2.name=NodeMCU 1.0 (ESP-12E Module)
nodemcuv2.build.board=ESP8266_NODEMCU

espino.name=ESPino (ESP-12 Module)
espino.build.board=ESP8266_ESP12

espinotee.name=ThaiEasyElec's ESPino
espinotee.build.board=ESP8266_ESP13

d-a-v, 2021-01-28:
#7839 fixed this, no ?
@earlephilhower

earlephilhower, 2021-02-13:
Only the ones as ESP8266_GENERIC were updated.  A new PR is forthcoming for the rest.

sprior, 2021-02-13:
Nice! I look forward to the release.

PR #7875 (2021-02-14): Update board-specific defines
Fixes #6805

Update the build.board for several boards which were originally added
with a generic type.
commit e4435fa3 (2021-02-14): Update board-specific defines (#7875)

Fixes #6805

Update the build.board for several boards which were originally added
with a generic type.
closed 2021-02-14 by d-a-v