esp8266/Arduino #6810: Crash inside WiFiClientSecure connect

fmgomes, 2019-11-19:
I had a small SMTP client implementation done more than 1 year ago that was tested at that time with several email providers. Currently, I've identified that with some of the same email providers it crashes inside the connect call:

`  WiFiClientSecure client;
  char smtpUserBase64[128];
  char smtpPassBase64[128];

  base64_encode(strlen(smtpUser), (unsigned char*)smtpUser, sizeof(smtpUserBase64), smtpUserBase64);
  base64_encode(strlen(smtpPass), (unsigned char*)smtpPass, sizeof(smtpPassBase64), smtpPassBase64);

  Serial.println("Connecting to smtp server");

  if (client.connect(smptServerName, smptServerPort) == 1) {
    Serial.println(F("connected"));
  } else {
    Serial.println(F("connection failed"));
    return 0;
  }
  Serial.flush();
`
I can see the log "Connecting to smtp server" but not the "connected" or "connection failed"

Nevertheless this still works with some smtp email servers (for example it works with gmail (I have to configure 'less secure apps') and yahoo, but crashes using aol or smtp2go.

Any idea of what could be the problem? Could it be a memory limitation for processing the certificates? I get an exception 29, and decoding the stack dump, the first line refers AES decryption

Thanks in advance!

Fernando

Jeroen88, 2019-11-19:
You should at least set one of the validation methods, see [here](https://github.com/esp8266/Arduino/blob/919c7535631d60e085f87d2b4c72d05a88651eed/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino#L158) for a certificate example (other validation methods in the same example). If you use certificates, you should also set the time, so that the validity of the certificate can be verified.

d-a-v, 2019-11-19:
@fmgomes 
Please fill the initial issue template (presented to you when a new issue is opned), try again with latest release (2.6.1) or latest git,
run at 160MHz, enable all debug options, show us the log and the decoded stack,
You also may retry with the "Basic ciphers" options.
(and what @Jeroen88 told)

devyte, 2019-11-19:
Which core version? Also, no MCVE.

earlephilhower, 2019-12-08:
No update in 20 days, closing.  If it is still there with git head, please open a new issue with an MCVE so we can reproduce it locally.

closed 2019-12-08 by earlephilhower