esp8266/Arduino #6811: User exception (panic/abort/assert) after random number of successful requests

NiallRees, 2019-11-19:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[Feather HUZZAH ESP8266]
- Core Version:			[2.6.1]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Feather HUZZAH ESP8266]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I'm using the WiFiClientSecure client and sometimes during connection, other times after a request, User exception (panic/abort/assert) is raised and the board restarts. If it successfully makes a request, it crashes on the next attempt.

I've attached a decoded stack trace along with heap memory debug. Would really appreciate some help with this.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch (WiFi creds left blank)

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
 
const char* ssid     = "";
const char* password = "";
 
[... 125 lines cut ...]
0x4021d844: br_ssl_engine_recvrec_ack at src/ssl/ssl_engine.c line 290
0x40205ca4: operator+(StringSumHelper const&, String const&) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/WString.cpp line 410
0x4020513d: fs::FS::open(char const*, char const*) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h line 781
0x40203cbc: BearSSL::WiFiClientSecure::available() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 386
0x402036d2: std::__shared_ptr ::operator=(std::__shared_ptr &&) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h line 858
0x4021d21e: cc_do_sign at src/ssl/ssl_ccert_single_rsa.c line 131
0x40203910: BearSSL::WiFiClientSecure::~WiFiClientSecure() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 119
0x40203af3: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 533
0x40205d7f: String::equals(String const&) const at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/WString.cpp line 501
0x402026e0: ClientContext::_write_some() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/include/ClientContext.h line 504
0x40203c11: BearSSL::WiFiClientSecure::_write(unsigned char const*, unsigned int, bool) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 288
0x40207928: WiFiClient::peek() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 284
0x40207928: WiFiClient::peek() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 284
0x4020118d: connectToWifi() at /Users/niallwoodward/silverton-dev/smart-crunch/smart-crutch-huzzah/smart-crutch-huzzah.ino line 45
0x402077c8: _GLOBAL__sub_D__Z15settimeofday_cbOSt8functionIFvvEE() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/sntp-lwip2.cpp line 494
0x40204eb9: TwoWire::write(unsigned char) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/Wire/Wire.cpp line 191
0x40204be4: BearSSL::CertStore::freeHashedTA(void*, br_x509_trust_anchor const*) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/CertStoreBearSSL.cpp line 219
0x40204f34: _GLOBAL__sub_I__ZN7TwoWire8rxBufferE() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/Wire/Wire.cpp line 333
0x40204fa8: fs::File::peek() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/FS.cpp line 73
0x402010d4: connectToWifi() at /Users/niallwoodward/silverton-dev/smart-crunch/smart-crutch-huzzah/smart-crutch-huzzah.ino line 38
0x40207928: WiFiClient::peek() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 284
0x402058dc: String::init() at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/WString.cpp line 131
0x40100ea5: umm_realloc(void*, size_t) at /Users/niallwoodward/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/umm_malloc/umm_malloc.cpp line 804

```

NiallRees, 2019-11-19:
I've also just found that using host as script.google.com in https://github.com/esp8266/Arduino/blob/919c7535631d60e085f87d2b4c72d05a88651eed/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino#L158 causes the exact same error.

earlephilhower, 2019-11-19:
@NiallRees can you try something simple on your end?  I'm traveling for a few days and may not be able to run anything on real HW for a while.

Can you make:
https://github.com/esp8266/Arduino/blob/919c7535631d60e085f87d2b4c72d05a88651eed/cores/esp8266/StackThunk.cpp#L39
into
````
#define _stackSize (6748/4) 
````

and re-try the connections with your app or the Validation example?  That would add an additional 1K to the available stack for BearSSL and narrow it down to a stack overflow issue vs. something else.

NiallRees, 2019-11-19:
@earlephilhower thank you so much! It's working absolutely fine now. I've seen you've increased the stack size in previous releases, is the problem that certificates are getting larger and larger?

earlephilhower, 2019-11-19:
Could you add the following lines to your sketch to figure out exactly how much bigger the stack usage was?

First, add 
````
#include <StackThunk.h>
````
at the top of your sketch, then after connect:
````
Serial.printf("BSSL stack: %d\n", stack_thunk_get_max_usage());
````

It's not really a certificate size thing (they're processed byte-by-byte as they come in), but it could be a new encryption mode which is using more stack than others we've seen.

Thx!

NiallRees, 2019-11-19:
BSSL stack: 5636

earlephilhower, 2019-11-20:
Thanks, @NiallRees .  Unfortunately, that's actually saying that the required stack is *smaller* than the default value (5636 vs. 5748) which gives more questions than answers. 

Is that # stable through multiple connections?  Do you see some fluctuations between runs/connections?

My thinking is that there might be some ISR called which requires more than (5748-5636 = 112 bytes) of space, but it's not always called so you have the case where the unmodified code works *most* times, but when that ISR happens the SYS needs >112 bytes of stack...

The alternative is that it's not the stack, but memory corruption somewhere (possible, but we do run valgrind on the host to check for stuff like that).  By increasing the stack you don't avoid the corruption, but the corruption ends up in some part of the stack that's not used anyway so it's invisible to the app.

TD-er, 2019-11-20:
What part is using so much stack?
Is it in the client object?
What if you declare the client object on the heap?

devyte, 2019-11-20:
@TD-er it's the BSSL stack, different from the SYS or CONT. There's a stack thunk mechanism involved to dynamically allocate and destroy it.

TD-er, 2019-11-20:
Ah, OK, didn't know that.

earlephilhower, 2019-11-21:
@NiallRees thanks for the detailed info.  I was able to reproduce it using the BSSL_Validation.ino like you said.  Sometimes it worked fine w/o any changes, but other times it would crash.

The issue is sometimes the max BSSL stack usage is 5636, which is fine, but other times I get that it's used **5828** which is past the stack end.

I'll bump up the stack allocation and tweak the validation exmable to dump the BSSL stack as well.

closed 2019-11-21 by earlephilhower

rch16, 2019-11-27:
Hi @earlephilhower ,

I am working with @NiallRees on his project and therefore have been encountering the same problems. I completed your fix, i.e. editing line 39, but am still intermittently experiencing the same error. Any suggestions?

Thanks,
Becca

jeffbuot, 2019-12-23:
Hi, I'm a totally newbie about this @earlephilhower I just started learning Arduino using nodemcu esp8266. Do you have details on how to fix this?

> @NiallRees can you try something simple on your end? I'm traveling for a few days and may not be able to run anything on real HW for a while.
>
> Can you make: [...]

Teddyz, 2020-01-01:
FYI

I have (had) the same problem. I am connected to my local MQTT-broker, but close this connection for a short connection to Pushover (to send messages to my phone). This led to crash after a number of connects (in the range 2 to 20-something). 

So far:
01 BSSL stack: 0 -> 4616 
02 BSSL stack: 4616 ->5736 
03 BSSL stack: 5736 -> 6088 

And no crashes yet.

earlephilhower, 2020-01-01:
@Teddyz, you're saying you saw `stack_thunk_get_max_usage` return 6088?  Like I said, the # in the code is empirically determined, so it's possible to need more than the 5900.

I think `6100` would be a better number.  Could you give it a try in your app and report back?  If that's good, we can update the main code to give it another 200bytes and hopefully get this all over with.

Teddyz, 2020-01-01:
I had this BSSL-check in my program running today:

```CPP
#include <StackThunk.h>
uint16_t max_stack_usage = 0;
...
// In loop
  if (now() - prev1s >= 1UL) {
    //checkDelay("", 1);
    prev1s = now();
    relay (4); // update relay
    yield();
    //checkDelay("do every 1 seconds", 2);
    uint16_t stack_usage = stack_thunk_get_max_usage();
    static uint16_t usage_increases = 0;
    if ( stack_usage > max_stack_usage ) {
      Serial.printf("%02d BSSL stack: %d -> %d \n", ++usage_increases, max_stack_usage, stack_usage);
      max_stack_usage = stack_usage;
    }
    return;
  }

```

01 BSSL stack: 0 -> 4484 
02 BSSL stack: 4484 -> 5796 
03 BSSL stack: 5796 -> 5928 
04 BSSL stack: 5928 -> 6056 
05 BSSL stack: 6056 -> 6136 

Soft WDT Crash after > 100 connections to api.pushover.net/1/messages.json. Plus the same number of connections to my local MQTT-broker.

01 BSSL stack: 0 -> 4516 
02 BSSL stack: 4516 -> 5736 
03 BSSL stack: 5736 -> 5796 
04 BSSL stack: 5796 -> 5828 
05 BSSL stack: 5828 -> 6104 
06 BSSL stack: 6104 -> 6120 a

So far no crash.

The Soft WDT reset had the explanation below. I do not understand where it went wrong, but I also do not think it is related to this problem.
```
0x40100984: millis at C:\Users\Dick\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_wiring.cpp line 188
0x40100969: millis at C:\Users\Dick\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_wiring.cpp line 185
0x4020a4ea: lwmqtt_arduino_timer_get(void*) at D:\Mina Dokument\Arduino\libraries\MQTT\src/MQTTClient.cpp line 25
0x4010088c: ets_post at C:\Users\Dick\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 160
0x4020abe8: now() at D:\Mina Dokument\Arduino\libraries\Time/Time.cpp line 252 (discriminator 1)
0x40205afc: loop at D:\Mina Dokument\Arduino\MQTT\mqtt_pocsag_v001/mqtt_pocsag_v001.ino line 252
0x4010088c: ets_post at C:\Users\Dick\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 160
0x4020e1d8: loop_wrapper() at C:\Users\Dick\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 180
0x40101675: cont_wrapper at C:\Users\Dick\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/cont.S line 81
```

Thanasist2, 2020-04-27:
i had problem with resets on boot after "Refreshing JWT" but since the mod everything works fine for 2 days now.
BSSL stack: 6132

stasysdr, 2020-12-28:
Changing BSSL stack to 6648 was not enough for me, BSSL was timing out.
I tried to use netClient.setBufferSizes(1024, 1024) as explained here https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html#setbuffersizes-int-recv-int-xmit
and it works now without timeout.

JamesDelfini, 2023-10-16:
> Changing BSSL stack to 6648 was not enough for me, BSSL was timing out. I tried to use netClient.setBufferSizes(1024, 1024) as explained here [...]

I have similar issue on this thread causes `exception (panic/abort/assert)` but with different cause. Thanks to @stasysdr I also set the setBufferSizes to 1024, the default is 512 at `WiFiClientBearSSL.cpp`.

The error was due to the conflict between `ESPAsyncWebServer`, which receives web requests from the client and ESP8266 as the web server. During the `client.connect` if it receives a request then the exception will occur, maybe the connection needed more buffer than the default due to the "hanging process that got stuck" based on my testing.

I did not change the default _stack size_ defined as `#define _stackSize (6200/4)` on `StackThunk.cpp`. Since the I was receiving BSSL stack size around 4300 to 4400.

No more crashes have occurred after I adjusted the buffer size. Before, sometimes it would go through, and sometimes it wouldn't when `client.connect` was hanging which is not good for ESP to get stuck on a process.

PR #6819 (2019-11-21): Expand BSSL stack
Fixes #6811 which found an issue where connecting to scripts.google.com
would *occasionally* cause a crash.  On inspection, it was found that up
to 5828 bytes of stack were used once in a while, so expand the stack to
5900 bytes to cover this case plus a little extra.
commit ebae47c1 (2019-11-21): Expand BSSL stack (#6819)

Fixes #6811 which found an issue where connecting to scripts.google.com
would *occasionally* cause a crash.  On inspection, it was found that up
to 5828 bytes of stack were used once in a while, so expand the stack to
5900 bytes to cover this case plus a little extra.