esp8266/Arduino #6815: Exception number 3 in versions higher than 2.50

franciscogimeno2000, 2019-11-20:
Hi, I have a problem with versions higher than 2.50

What I am going to describe compiles and works without problems in version 2.50 but in the following ones, that exception number jumps in a part of the code that I have isolated.

This is the code:

`
webconfig()
{
String webcon;
webcon = CONFIG_page;  <-----------When you try to do this the exception is triggered.
....
.
..
.
`

Clarification on "CONFIG_page"
It is the htlm code that is defined in an "include" file , basically it is the web code to be displayed. The size of this code is about 3000 characters approximately.

As I say, this in version 2.50 compiles and works without problems.

That has changed since 2.50 so that it doesn't work properly, or that I do wrong?

Greetings and thanks for your work
Fran

d-a-v, 2019-11-20:
The issue template that you didn't fill would have told us many things that we consequently don't know.
What version of the core are you using ?

How is the html object declared in your include file ?

Can you run the issue again and decode the stack strace (https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html).

From your explanation it should not fail like that (even if you don't have enough heap (you can check it with `ESP.getFreeHeap()`)).

On a side note there is surely a possibility to send your html page without putting it in a String first, avoiding a useless evil copy.

devyte, 2019-11-20:
@franciscogimeno2000 the issue template is there for a reason. The instructions require you to follow a checklist and to provide specific information before reporting. In many cases this guides the user to a resolution before actually submitting the report. At the very least, it provides us with beeded information without wasting time going back and forth asking questions. 
Please open a new issue and comply with the instructions.
Closing.

closed 2019-11-20 by devyte