esp8266/Arduino #6818: Exception number 3 in versions higher than 2.50

franciscogimeno2000, 2019-11-20:
### Basic Infos

- [ *] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [* ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [* ] I have tested that the issue is present in current master branch (aka latest git).
- [* ] I have searched the issue tracker for a similar issue.
- [* ] If there is a stack dump, I have decoded it.
- [* ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-01]
- Core Version:			[2.6.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  	[Generic ESP8266]
- Flash Mode:	[dout compatible]
- Flash Size: 		[1MB fs:none ]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:	[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description
Exception number 3 in versions higher than 2.50

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

franciscogimeno2000, 2019-11-20:
Hello, is the description wrong?

franciscogimeno2000, 2019-11-20:
OK thanks.

earlephilhower, 2019-11-20:
Looks like you were editing when I closed.  You will need to decode that stack dump and provide a *full* compilable, working MCVE so that we can look at it (as requested in the template)

As of now, there's nothing usable for debug given here yet.

earlephilhower, 2019-11-20:
And, if I had to guess, you're running out of memory copying a massive const char * stored in RAM into an equally massive String also stored in RAM.

franciscogimeno2000, 2019-11-20:
ok , Two questions
1 How is it decoded?
2 Why does version 2.50 work?

earlephilhower, 2019-11-20:
Sorry, but without a full MCVE and a properly decoded stack trace including the callbacks there is nothing to be done here.

devyte, 2019-11-20:
@franciscogimeno2000 this is better than your other attempt, but it is still not quite there.
The requirement is that you provide an MCVE: Minimal Complete Verifiable Example. In other words, a complete sketch where you have pulled out all unnecessary code, that can be loaded into the IDE and built, then flashed to a board to verify the issue you face. This is so that we can reproduce the issue on our end and investigate the cause.
Please edit your OP and replace the sketch with

franciscogimeno2000, 2019-11-20:
The problem is the size, the number of characters that happened to the string.
I have left a few characters and it works.
The question is, because in version 2.50 it works without problems and in the following ones it does not. That has changed in the updates.

franciscogimeno2000, 2019-11-20:
Can you tell me where I find information about what the configuration parameters of the chip do in the arduino ide?

devyte, 2019-11-21:
Available heap changes a bit from release to release. Also, you could be running into fragmentation, i.e.: you have enough free heap, but no single piece of it is big enough to hold a required allocation.
If it works with a slightly smaller html string, then a memory issue is most likely.
One thing you can do is use the ESPClass, there are methods to get free heap and a fragmentation metric. Look at Esp.h.
It is in general not a good idea to server html from embedded strings. Instead, put static content in files in a (small) FS, and serve dynamic data as a json to be loaded by the browser into the static html. This is what I do.

I don't see a core issue, so closing for now. If further info is provided that shows otherwise, then I'll reopen.

closed 2019-11-21 by devyte

franciscogimeno2000, 2019-11-21:
Thank you for your help, I have achieved a good decoding.

CODE:

``
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4000bf64
EXCVADDR: 0x40242cf3

Decoding stack results
0x40209aa3: String::String(char const*) at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\WString.cpp line 34
0x40209220: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/HardwareSerial.h line 159
0x402014ae: Montarwebconfig() at C:\Users\IEUser\Documents\Arduino\AVERESTESIVAOQUEREDUCI/AVERESTESIVAOQUEREDUCI.ino line 696
0x40209500: Print::print(String const&) at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\Print.cpp line 116
0x402098a1: String::invalidate() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\WString.cpp line 140
0x402045d2: std::_Function_handler ::_M_invoke(const std::_Any_data &) at C:\Users\IEUser\Documents\Arduino\AVERESTESIVAOQUEREDUCI/AVERESTESIVAOQUEREDUCI.ino line 480
0x402098a1: String::invalidate() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\WString.cpp line 140
0x402011f8: esp8266webserver::FunctionRequestHandler ::canHandle(HTTPMethod, String) at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 23
0x401000e1: std::function ::operator()() const at c:\users\ieuser\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2465
0x40202f3e: esp8266webserver::FunctionRequestHandler ::handle(esp8266webserver::ESP8266WebServerTemplate &, HTTPMethod, String) at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 47
0x4020490a: esp8266webserver::ESP8266WebServerTemplate ::_handleRequest() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 665
0x40100278: millis() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40206b20: WiFiServer::available(unsigned char*) at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WiFi\src\WiFiServer.cpp line 116
0x401009ef: free(void*) at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40204adc: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 335
0x40206e7f: WiFiUDP::parsePacket() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WiFi\src\WiFiUdp.cpp line 199
0x402075dc: DNSServer::processNextRequest() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\DNSServer\src\DNSServer.cpp line 166
0x40204b86: loop() at C:\Users\IEUser\Documents\Arduino\AVERESTESIVAOQUEREDUCI/AVERESTESIVAOQUEREDUCI.ino line 624
0x4020aa61: esp_schedule() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_main.cpp line 112
0x4020ab70: loop_wrapper() at C:\Users\IEUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_main.cpp line 179
``