esp8266/Arduino #6821: HTTP OTA UPDATE on NODEMCU - ESPhttpUpdate library is not continuously updating the firmware.

Lakshmi-aloore, 2019-11-21:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:			         nodemcu0.9
- Core Version:			v2.4.0
- Development Env:		Arduino IDE
- Operating System:	       Ubuntu 18.04

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		nodemcu
- Flash Frequency:	80Mhz
- CPU Frequency:		80Mhz
- Upload Using:	SERIAL
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Hi everyone!! 

Actually what I am trying to do is I am updating the firmware of the nodemcu based on the version which i receive via MQTT and compare the version if the version has changed then I will update the firmware.

My question is:

Why the code never goes to the switch block when ESPhttpUpdate.update() successful done ?
This switch has no default; when ret == 0 it goes to HTTP_UPDATE_FAILED case. It is correct ?

I read due to softrestart it reloads the code again. But Can I get help to stop the softrestart or do some changes to the my code so that  ESPhttpUpdate.update()  enters into switch block therefore it helps to run the main function after update and whenever update comes from MQTT then firmware has to update. 

Or can I know how to update the firmware using mqttota
Looking Forward for help.Thanks in advance.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <HTTPClient.h>
#include <ESP32httpUpdate.h>

#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"

#define WLAN_SSID       ""
#define WLAN_PASS       ""

#define AIO_SERVER      ""
#define AIO_SERVERPORT  1883                   // use 8883 for SSL

WiFiClient client1;

Adafruit_MQTT_Client mqtt(&client1, AIO_SERVER, AIO_SERVERPORT);

Adafruit_MQTT_Subscribe versionData = Adafruit_MQTT_Subscribe(&mqtt, "TestTopic");

// LED pins

String readstring;

String NewactionID;
String OldactionID;

void MQTT_connect() {
  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
         while (1);
       }
  }
  Serial.println("MQTT Connected!");
}

void wifisetup(){
  
  // Connect to WiFi access point.
  Serial.println(); Serial.println();
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.println("WiFi connected");
  Serial.println("IP address: "); Serial.println(WiFi.localIP());
}

void updateFirmware(){
  if((WiFi.status() == WL_CONNECTED)) {

       t_httpUpdate_return ret = ESPhttpUpdate.update("http://172.20.10.3:5555/download/api");
       Serial.println("RETURN_VALUE :");
       Serial.println(ret);

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
//                ESP.restart();
                break;

            case HTTP_UPDATE_NO_UPDATES:
                Serial.printf("HTTP_UPDATE_NO_UPDATES");
//                ESP.restart();
                break;

            case HTTP_UPDATE_OK:
                Serial.printf("HTTP_UPDATE_OK");
                Serial.println("$$$$$$$$$$$$$$$$");
                Serial.println(ret);
//                ESP.restart();
                break;
        }
    }
}

void setup() {
  Serial.begin(115200); 
  pinMode(LED_BUILTIN,OUTPUT);
  
wifisetup();
mqtt.subscribe(&versionData);
}

void loop() {
  MQTT_connect();

  // this is our 'wait for incoming subscription packets' busy subloop
  // try to spend your time here

  Adafruit_MQTT_Subscribe *subscription;
    while ((subscription = mqtt.readSubscription(50))) {
    if (subscription == &versionData) {
      //Serial.print(F("Got: "));
      //Serial.println((char *)onled.lastread);
     readstring =((char*)versionData.lastread);
     NewactionID = readstring;
      Serial.println(readstring);
      }
  }

    if(NewactionID != OldactionID){
      Serial.println("ID has changed");
      OldactionID = NewactionID; 
      updateFirmware();
    }else {
      Serial.println("ID has not changed");
    } 
  
}

### Debug Messages

Using library WiFiClientSecure at version 1.0 in folder: /root/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WiFiClientSecure 
Using library ESP32httpUpdate at version 2.1.145 in folder: /root/Arduino/libraries/ESP32httpUpdate 
Using library Update at version 1.0 in folder: /root/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/Update 
Using library FS at version 1.0 in folder: /root/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/FS 
Using library SPIFFS at version 1.0 in folder: /root/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/SPIFFS 
Using library Adafruit_MQTT_Library-master at version 1.0.3 in folder: /root/Arduino/libraries/Adafruit_MQTT_Library-master 
/root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size -A /tmp/arduino_build_47945/test1.ino.elf
Sketch uses 882670 bytes (44%) of program storage space. Maximum is 1966080 bytes.
Global variables use 39608 bytes (12%) of dynamic memory, leaving 288072 bytes for local variables. Maximum is 327680 bytes.
python /root/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py --chip esp32 --port /dev/ttyUSB1 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /root/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/partitions/boot_app0.bin 0x1000 /root/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 /tmp/arduino_build_47945/test1.ino.bin 0x8000 /tmp/arduino_build_47945/test1.ino.partitions.bin 
esptool.py v2.6
Serial port /dev/ttyUSB1
Connecting........_____....._____..
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 80:7d:3a:a4:4a:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 27172.6 kbit/s)...
Hash of data verified.
Compressed 15328 bytes to 9994...

Writing at 0x00001000... (100 %)
Wrote 15328 bytes (9994 compressed) at 0x00001000 in 0.1 seconds (effective 1039.4 kbit/s)...
Hash of data verified.
Compressed 882784 bytes to 498408...

Writing at 0x00010000... (3 %)
Writing at 0x00014000... (6 %)
Writing at 0x00018000... (9 %)
Writing at 0x0001c000... (12 %)
Writing at 0x00020000... (16 %)
Writing at 0x00024000... (19 %)
Writing at 0x00028000... (22 %)
Writing at 0x0002c000... (25 %)
Writing at 0x00030000... (29 %)
Writing at 0x00034000... (32 %)
Writing at 0x00038000... (35 %)
Writing at 0x0003c000... (38 %)
Writing at 0x00040000... (41 %)
Writing at 0x00044000... (45 %)
Writing at 0x00048000... (48 %)
Writing at 0x0004c000... (51 %)
Writing at 0x00050000... (54 %)
Writing at 0x00054000... (58 %)
Writing at 0x00058000... (61 %)
Writing at 0x0005c000... (64 %)
Writing at 0x00060000... (67 %)
Writing at 0x00064000... (70 %)
Writing at 0x00068000... (74 %)
Writing at 0x0006c000... (77 %)
Writing at 0x00070000... (80 %)
Writing at 0x00074000... (83 %)
Writing at 0x00078000... (87 %)
Writing at 0x0007c000... (90 %)
Writing at 0x00080000... (93 %)
Writing at 0x00084000... (96 %)
Writing at 0x00088000... (100 %)
Wrote 882784 bytes (498408 compressed) at 0x00010000 in 8.1 seconds (effective 874.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 7814.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

d-a-v, 2019-11-21:
`Chip is ESP32D0WDQ6 (revision 1)` ?

Jeroen88, 2019-11-21:
It looks like you can avoid reboot using [this](https://github.com/esp8266/Arduino/blob/ebae47c13aef5ffd1c7464f74502aa83b17bc8ea/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L80) function.

Lakshmi-aloore, 2019-11-21:
@Jeroen88  and @d-a-v 
Thanks for your reply.

My doubt is can we use mqtt for ota update. Or is it possible for ESPHTTPUpdate to get into the switch block once update is successful.

devyte, 2019-11-21:
Your logs show you have an esp32. This is the esp8266 repo.
Closing.

closed 2019-11-21 by devyte