esp8266/Arduino #6829: showing Python was not found during uploading spiffs

ranap040, 2019-11-24:
showing Python was not found during uploading spiffs

earlephilhower, 2019-11-24:
Yes, we've got a PR into the Spiffs uploader to fix this.

ranap040, 2019-11-24:
thanks for the help... Its working fine with the my rgb strip but i want to change the strip with 4 in 1 rgb strip(each led in the strip packed with red,green,blue and a white/warm white). so for that strip should I need to change anything in the code ?
If yes then plz help!
it will enough to have option to toggle the white light on and off and control the brightness doesn't need to change anything in the effects or presets!

earlephilhower, 2019-11-25:
Please try the plugin ZIP embedded in my comment:
https://github.com/esp8266/arduino-esp8266fs-plugin/pull/65#issuecomment-558234472
https://github.com/esp8266/arduino-esp8266fs-plugin/files/3887643/ESP8266FS-0.5.0.zip

You can also try the one that was released in the plugin repo (but it did not work for me w/Arduino 1.8.8).

If you could report back on that PR which one worked/didn't work it would be helpful.

devyte, 2019-11-26:
The relevant PR is merged. Closing.

closed 2019-11-26 by devyte