esp8266/Arduino #6831: UDP Crash with versions above 2.5.2

SvenSiwek, 2019-11-24:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8266]
- Core Version:			[2.6.0 & 2.6.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[LOLIN (WEMOS) D1 R2 & mini]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [other] 

### Problem Description
I am pretty new to this all. So please be polite with me.
I do have a working sketch with version 2.5.2.
I updated to 2.6.1 
The WEMOS does reboot after a certain time (10 to 15 seconds) with heavy ingoing udp traffic.

I downgraded to 2.6.0
The WEMOS does reboot after a certain time (10 to 15 seconds) with heavy ingoing udp traffic.

I downgraded to 2.5.2
The WEMOS does work as expected.

Since I am pretty new to this all (NodeEMU etc),  
is there a "how to" to provide you with additional infos - if needed?

I am willing to help...

Sven

TD-er, 2019-11-24:
Do you "server" the UDP related code with regular intervals?
Just like "serving" the webserver, you must also make regular calls to see what packets have to be processed for UDP.

devyte, 2019-11-24:
@SvenSiwek the issue template you filled out requires an MCVE. Without that there isn't much we can do.
Please edit your OP and add a MCVE sketch with which the issue can be reproduced. If there is additional info involved, such as needing a linux script or similar to reproduce, please add that as well.

SvenSiwek, 2019-11-24:
Okay I found out what MCVE means. I ll  provide one.

It would be fine if you explain what OP means, which I have to edit.

My English is not that good, and as I wrote I am new to this.
Thx
Sven

devyte, 2019-11-24:
Original Post

TD-er, 2019-11-25:
> On Linux do
```
cat /dev/random | netcat -u 172.16.178.41 7777
```

Hmm that sounds like "lots of traffic" and "large packets".
Both are a very good receipt for crashes on these units.
Have you also tried limiting the speed by setting an interval (`-i`) and maybe also use shorter messages.
I don't know what netcat will consider a chunk of data, when processing /dev/random.

devyte, 2019-11-25:
You are processing packets once per loop. You also have delay(100) in the loop. That effectively limits you to 10 packets per second. If you send faster than that, the esp will accumulate them, run out of mem, and crash.
Closing due to user error.

devyte, 2019-11-25:
Also, you're reading at most 1024 bytes into the buffer. If the packet is bigger, there are remaining bytes unread.

SvenSiwek, 2019-11-25:
Why do you close this?

This ist not an user error. You're really doing it very easy.

If I send too many packets too fast, that's my problem. The nodemcu must therefore not crash. If she loses packages that's okay.

I checked my MCVE against 2.5.2 which works as expected without any crash.

So I will stick to that version.

Thank you for taking my time.
I will think for the future again to report errors, if I am helped in this way.
Translated with Google.

Sven

devyte, 2019-11-25:
I closed because I still don't see a core issue. That does not mean end of discussion. 
Also, if what you need is general help, this is not the place for it, because this is an issue tracker, meant for tracking issues in the core, and not a general support forum.
Finally, your reports help yourself before anyone else, because you are the user faced with the issue. In this case, as the primary interested party, it is up to you to continue investigating and providing info, until you either show a core issue, in which case I'll reopen, or you find a mistake in your code, in which case you'll probably immediately realize how to fix it. 

As for 2.5.1, it would intermittently lose incoming packets independently of receiving speed, and that was addressed. If you don't want to process all incoming, it is your responsibility to receive and discard unwanted packets, because there is no way to differentiate between your use case of unwanted packets and somebody else who wants to maximize reception.
And btw, it is true that udp packets can get dropped in traffic, but it is not ok to receive packets and lose them in our software before passing them to the user app.

SvenSiwek, 2019-11-25:
Hi,
I opened this issue, because something that works with 2.5.2 does not work with 2.6.1 and up.

If you arent interested in finding the cause of that, than it is that way,

For me its pretty easy. I generate an udp stream of packets on the server side. 
It is not importand for me that every single udp arrives in the sketch.

But this packet stream should not crash the device - under no circumstances.
This is what happens with version 2.6.1 and up.

So, too much udp packets in a to short timerange crashes the device. That is core.

Sven

TD-er, 2019-11-25:
@SvenSiwek Please read his reply again.
It was an intentional change (aka. bug fix) to change behavior and not drop packets.
I do agree the result is somewhat unfortunate, to cause a crash.

Maybe you can change the function in your code to read the entire packet (read until packet length is processed) and also not waiting for 100 msec every loop.
In a few 100 msec the memory of the ESP is full when receiving at max. speed, especially if you only read 1024 bytes.
Also I am not entirely sure your code does continue to read on the same packet if you go through the next loop. So that may leave the rest of the packet in RAM, causing the RAM to fill up and all kinds of things will then fail as lots of memory allocations will fail.

SvenSiwek, 2019-11-25:
Hi,

the goal, at last for me was, to provide a MCVE, which I did.

If a bug fix results in somewhat like a possible DoS it is not a good one.
Maybe you can install a switch to get the old behavior back.

To the root of that : 
My daughter wanted to have a nice light show with LEDs, which she can control via mobile phone.
As I myself am a programmer (COBOL, Java, PHP), I thought, can not be that hard and I came across the ESP and fastled.
That works great https://github.com/jasoncoon/esp8266-fastled-webserver .
That was not enough for us. It should also give great effects to match music. Then I found https://github.com/scottlawsonbc/audio-reactive-led-strip .

I have combined both in a sketch and it works fine - just up to version 2.5.2.

Now there was the update to 2.6.1 and higher.
It does not work with these versions.
I thought, okay then I'll call it. Something does not seem to work as it should.

The Visualization-Server sends UDP light instructions to the ESP. Always 4 bytes. The first byte Addresses the number of the LED. The other three bytes are RGB color values.

The source code has remained pretty much the same. 

Translated with Google
Sven

SvenSiwek, 2019-11-28:
> @SvenSiwek Please read his reply again.
> It was an intentional change (aka. bug fix) to change behavior and not drop packets.
> I do agree the result is somewhat unfortunate, to cause a crash.
>
> Maybe [...]

I debugged this more: It is not a memory problem visible to the sketch

`  
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
// Wi-Fi network to connect to (if not in AP mode)
 char* ssid = "XXXXXXXXX";
 char* password = "XX";
WiFiUDP port;
// Maximum number of packets to hold in the buffer.
#define BUFFER_LEN 2048

char packetBuffer[ BUFFER_LEN + 1];

IPAddress mc_groupIP = IPAddress(233, 233, 233, 233);
unsigned int mc_localUdpPort = 4711;  // local port to listen on

int len;

void setup() {
  WiFi.setSleepMode(WIFI_NONE_SLEEP);

  Serial.begin(115200);
  delay(100);
  Serial.setDebugOutput(true);
  WiFi.mode(WIFI_STA);
  Serial.printf("Connecting to %s\n", ssid);
  if (String(WiFi.SSID()) != String(ssid)) {
    WiFi.begin(ssid, password);
  }
   while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(500);
  }

  Serial.println("Listening UDP");
  port.beginMulticast(WiFi.localIP(), mc_groupIP, mc_localUdpPort);

  

}

void loop() {
  // put your main code here, to run repeatedly:
  int packetSize = port.parsePacket();
  // If packets have been received, interpret the command
  if (packetSize) {
    Serial.print("P");
    Serial.println(packetSize);
    // Read data over socket
    len = port.read(packetBuffer, BUFFER_LEN);
    Serial.print("L");
    Serial.println(len);
  }  
}`

The only thing this sketch does is println...

And here is the result:
ip:172.16.178.32,mask:255.255.255.0,gw:172.16.178.1
Connecting to SHMS
HTTP web server started
Listening UDP
pm open,type:0 0
P300
L300
P300
L300
P472
L472
P472
L472
P440
L440
P448
L448
P456
L456
P440
L440
P300
L300
P300
L300
P440
L440
[...]
P160
L160
P152
L152
P160
L160
P168
L168
P152
L152
Fatal exception 0(IllegalInstructionCause):
epc1=0x40241abc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x40241abc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffebf0 end: 3fffffb0 offset: 01a0
3fffed90:  400005e1 40102a5f 000000ff 00000000  
3fffeda0:  40224f3c 00000030 00000010 0000c60e  
3fffedb0:  40224e05 00001267 e9e9e9e9 42b210ac  
3fffedc0:  3fff3074 3fff0d94 00000000 3fff3074  
3fffedd0:  00000000 00000006 00000000 00000000  
3fffede0:  00000000 3fff3486 3fff3454 00000000  
3fffedf0:  00000000 00000098 00000098 00000001

d-a-v, 2019-11-28:
Your stack trace is personal, [only you can decode it](https://arduino-esp8266.readthedocs.io/en/2.6.2/Troubleshooting/stack_dump.html).

Thanks, we need to reproduce the issue to fix it.

Do you have any idea of the packet frequency ?

SvenSiwek, 2019-11-28:
To explain I use that python mentioned above. 
https://github.com/scottlawsonbc/audio-reactive-led-strip 

I do have 150Led * 4 Byte * 60Hz. ~ 36.000Bytes/s max.

I will test with lower Hz.

0x40204699: loop() at D:\Users\svens\Documents\Arduino\udp_kaputt/udp_kaputt.ino line 69
0x402046a9: loop() at D:\Users\svens\Documents\Arduino\udp_kaputt/udp_kaputt.ino line 73
0x40209020: loop_wrapper() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 180

Line 69  int packetSize = port.parsePacket();
Line 73 Serial.println(packetSize);

And a second one

Decoding stack results
0x40216ba0: pbuf_alloc_LWIP2 at core/pbuf.c line 186
0x40216ba0: pbuf_alloc_LWIP2 at core/pbuf.c line 186
0x4021cc0c: sntp_init at apps/sntp/sntp.c line 670
0x40212af2: netif_sta_status_callback at glue-lwip/lwip-git.c line 331
0x40216ba0: pbuf_alloc_LWIP2 at core/pbuf.c line 186
0x40216b7c: pbuf_alloc_LWIP2 at core/pbuf.c line 284
0x4021cd60: mem_free at core/mem.c line 237
0x40214a89: sys_check_timeouts at core/timeouts.c line 390
0x40214a84: sys_check_timeouts at core/timeouts.c line 389
0x4021c9a8: sntp_request at apps/sntp/sntp.c line 604
0x40100a4c: malloc(size_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 511
0x4021cd40: mem_malloc at core/mem.c line 210
0x40216b7c: pbuf_alloc_LWIP2 at core/pbuf.c line 284
0x4021c9a8: sntp_request at apps/sntp/sntp.c line 604
0x4021c938: sntp_send_request at apps/sntp/sntp.c line 546
0x4021c9a8: sntp_request at apps/sntp/sntp.c line 604
0x4021c9f8: sntp_request at apps/sntp/sntp.c line 634
0x40213089: memp_free at core/memp.c line 447
0x40214a8c: sys_check_timeouts at core/timeouts.c line 390
0x40210ce2: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40210ce2: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40210c18: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4020ceb8: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 244
0x40210ce2: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40210ecc: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4020e631: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x4010078c: umm_free_core(void*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 316
0x40100a17: free(void*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40100384: vPortFree(void*, char const*, int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\heap.cpp line 308
0x40209f71: uart_write(uart_t*, char const*, size_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\uart.cpp line 509
0x40207f2c: String::copy(char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\WString.cpp line 214
0x40207f62: String::String(char const*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\WString.cpp line 36
0x4020524d: ESP8266WiFiSTAClass::SSID() const at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 659
0x40205268: ESP8266WiFiSTAClass::SSID() const at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 662
0x4020ca66: _printf_common at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 102
0x4020c81c: _printf_float at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_float.c line 327
0x402018fc: esp8266webserver::ESP8266WebServerTemplate ::_parseForm(WiFiClient&, String const&, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/Parsing-impl.h line 396
0x40204f30: ESP8266WiFiSTAClass::config(IPAddress, IPAddress, IPAddress, IPAddress, IPAddress) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 332
0x40204f30: ESP8266WiFiSTAClass::config(IPAddress, IPAddress, IPAddress, IPAddress, IPAddress) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 332
0x40204f30: ESP8266WiFiSTAClass::config(IPAddress, IPAddress, IPAddress, IPAddress, IPAddress) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 332
0x40208f08: __esp_yield() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 100
0x40209516: __delay(unsigned long) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40203751: setup() at D:\Users\svens\Documents\Arduino\udp_kaputt/udp_kaputt.ino line 48
0x40209018: loop_wrapper() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 177

SvenSiwek, 2019-11-28:
To explain what I do. Playing a song on my PC with this python( It's always the same song to reproduce the error - Carol of The Bells by Tommee Profitt ).

Tested with 30Hz sending. It almost made it, but:

Decoding stack results
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x40208ee6: can_yield() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 94
0x40208f5f: optimistic_yield(uint32_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 128
0x40206f97: Print::write(char const*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/Print.h line 56
0x4020715c: Print::print(char const*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\Print.cpp line 122
0x40206be8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 164
0x40206bf4: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 165
0x40206be8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 164
0x40206fbd: Print::write(char const*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/Print.h line 60
0x4020724d: Print::printNumber(unsigned long, unsigned char) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\Print.cpp line 260
0x40208f3a: __yield() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 100
0x40208f5f: optimistic_yield(uint32_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 128
0x4020614a: WiFiServer::available(unsigned char*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\WiFiServer.cpp line 120
0x40100a17: free(void*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40206be8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 164
0x4020aa04: WiFiClient::operator bool() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\WiFiClient.cpp line 350
0x40204599: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 307
0x40208f5f: optimistic_yield(uint32_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 128
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x402046a9: loop() at D:\Users\svens\Documents\Arduino\udp_kaputt/udp_kaputt.ino line 73
0x40209020: loop_wrapper() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 180

SvenSiwek, 2019-11-28:
Whoops:
After pressing the reset button that happend:
Decoding stack results
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x401001b9: esp_schedule() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 113
0x40209029: loop_wrapper() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 182

25Hz:

Decoding stack results
0x402170a4: udp_input at core/udp.c line 404
0x4021bfd8: ip4_input at core/ipv4/ip4.c line 1461
0x40100a17: free(void*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40212f45: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x40212d68: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 469
0x40232ec2: ethernet_input at glue-esp/lwip-esp.c line 365
0x40232ed3: ethernet_input at glue-esp/lwip-esp.c line 373
0x40210ce2: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x4020a9ec: WiFiClient::connected() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\WiFiClient.cpp line 338
0x4020aa04: WiFiClient::operator bool() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\WiFiClient.cpp line 350
0x40209f71: uart_write(uart_t*, char const*, size_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\uart.cpp line 509
0x40206be8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 164
0x40206bf4: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 165
0x40206be8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 164
0x40206fbd: Print::write(char const*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/Print.h line 60
0x4020724d: Print::printNumber(unsigned long, unsigned char) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\Print.cpp line 260
0x40208f3a: __yield() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 100
0x40208f5f: optimistic_yield(uint32_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 128
0x4020614a: WiFiServer::available(unsigned char*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\WiFiServer.cpp line 120
0x40100a17: free(void*) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40206be8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/HardwareSerial.h line 164
0x4020aa04: WiFiClient::operator bool() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src\WiFiClient.cpp line 350
0x40204599: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 307
0x40208f5f: optimistic_yield(uint32_t) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 128
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x401001b9: esp_schedule() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 113
[... cut ...]
0x40208f08: __esp_yield() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 100
0x40209516: __delay(unsigned long) at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40203751: setup() at D:\Users\svens\Documents\Arduino\udp_kaputt/udp_kaputt.ino line 48
0x40209018: loop_wrapper() at C:\Users\svens\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 177

d-a-v, 2019-11-28:
There is a debug option called OOM
I run the UDP example sketch and flooded it:
```sh
while true; do echo hey; done | nc -u 10.0.1.225 8888
```
I got on console the following (see below).
See the `:oom(nnn)` messages starting just before the crash.
When it happens, free heap is below 2KB.

I think we should have some sanity barrier for largest umm free chunk block to not be crossed by the core when possible (here we'd drop the incoming flooding packets) (so the FW does not fail like it is the case here according to the :oom messages not coming from us).

The thing is it has a cost to (re)process the largest free block (or the total free heap). 
I wonder @mhightower83 if there could be a way to maintain at little cost some sort of cache for the heap metrics (or at least the largest block).
I had not followed the latest umm discussion so I ask here where it would be useful to have.

```
22:33:23.666 -> Received packet of size 12 from 10.0.1.7:40953
22:33:23.699 ->     (to 10.0.1.225:8888, free heap = 2160 B)
22:33:23.699 -> Contents:(2160)
22:33:23.699 -> hey
22:33:23.699 -> hey
22:33:23.699 -> hey
22:33:23.699 -> 
22:33:23.699 -> Received packet of size 8 from 10.0.1.7:40953
22:33:23.699 ->     (to 10.0.1.225:8888, free heap = 2016 B)
22:33:23.699 -> Contents:(2016)
22:33:23.699 -> hey
22:33:23.699 -> hey
22:33:23.699 -> 
22:33:23.699 -> :oom(124)@?
22:33:23.699 -> Received packet of size 8 from 10.0.1.7:40953
22:33:23.699 ->     (to 10.0.1.225:8888, free heap = 1872 B)
22:33:23.699 -> Contents:(1872)
22:33:23.732 -> hey
22:33:23.732 -> hey
22:33:23.732 -> 
22:33:23.732 -> :oom(236)@?
22:33:23.732 -> Received packet of size 8 from 10.0.1.7:40953
22:33:23.732 ->     (to 10.0.1.225:8888, free heap = 1888 B)
22:33:23.732 -> Contents:(1888)
22:33:23.732 -> hey
22:33:23.732 -> hey
22:33:23.732 -> 
22:33:23.732 -> :oom(236)@?
22:33:23.732 -> :oom(212)@ieee80211_phy.c:313
22:33:23.732 -> Fatal exception 29(StoreProhibitedCause):
22:33:23.732 -> epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
22:33:23.732 -> 
22:33:23.732 -> Exception (29):
22:33:23.732 -> epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
22:33:23.765 -> 
22:33:23.765 -> >>>stack>>>
```

d-a-v, 2019-11-28:
More interestingly, I replaced `getFreeHeap()` by `getMaxFreeBlockSize()` right next to the `Content:` message.
When we can, we really should prevent the core and the FW from agonizing when *max-free-blocksize* is that low.

```
23:01:41.453 -> hey
23:01:41.453 -> hey
23:01:41.453 -> hey
23:01:41.453 -> 
23:01:41.453 -> :oom(236)@?
23:01:41.453 -> :oom(108)@?
23:01:41.453 -> :oom(1532)@?
23:01:41.486 -> :oom(1532)@?
23:01:41.486 -> Received packet of size 4 from 10.0.1.7:37742
23:01:41.486 ->     (to 10.0.1.225:8888, free heap = 1856 B)
23:01:41.486 -> Contents:(224)
23:01:41.486 -> hey
23:01:41.486 -> 
23:01:41.486 -> :oom(236)@?
23:01:41.486 -> :oom(1532)@?
23:01:41.486 -> Received packet of size 4 from 10.0.1.7:37742
23:01:41.486 ->     (to 10.0.1.225:8888, free heap = 2008 B)
23:01:41.486 -> Contents:(200)
23:01:41.486 -> hey
23:01:41.486 -> 
23:01:41.486 -> :oom(236)@?
23:01:41.486 -> Received packet of size 8 from 10.0.1.7:37742
23:01:41.520 ->     (to 10.0.1.225:8888, free heap = 1728 B)
23:01:41.520 -> Contents:(144)
23:01:41.520 -> hey
23:01:41.520 -> hey
23:01:41.520 -> 
23:01:41.520 -> :oom(236)@?
23:01:41.520 -> Received packet of size 4 from 10.0.1.7:37742
23:01:41.520 ->     (to 10.0.1.225:8888, free heap = 1728 B)
23:01:41.520 -> Contents:(152)
23:01:41.520 -> hey
23:01:41.520 -> 
23:01:41.520 -> :oom(236)@?
23:01:41.520 -> :oom(68)@?
23:01:41.520 -> :oom(40)@?
23:01:41.520 -> :oom(92)@abi.cpp:48
23:01:41.520 -> Contents:(80)
23:01:41.520 -> hey
23:01:41.520 -> 
23:01:41.520 -> :oom(236)@?
23:01:41.553 -> :oom(108)@?
23:01:41.553 -> Received packet of size 8 from 10.0.1.7:37742
23:01:41.553 ->     (to 10.0.1.225:8888, free heap = 1944 B)
23:01:41.553 -> Contents:(192)
23:01:41.553 -> hey
23:01:41.553 -> hey
23:01:41.553 -> 
23:01:41.553 -> :oom(236)@?
23:01:41.553 -> :oom(1532)@?
23:01:41.553 -> :oom(68)@?
23:01:41.553 -> Received packet of size 8 from 10.0.1.7:37742
23:01:41.553 ->     (to 10.0.1.225:8888, free heap = 1800 B)
23:01:41.553 -> Contents:(224)
23:01:41.553 -> hey
23:01:41.553 -> hey
23:01:41.553 -> 
23:01:41.553 -> :oom(236)@?
23:01:41.553 -> :oom(100)@?
23:01:41.553 -> Received packet of size 4 from 10.0.1.7:37742
23:01:41.586 ->     (to 10.0.1.225:8888, free heap = 1944 B)
23:01:41.586 -> Contents:(320)
23:01:41.586 -> hey
23:01:41.586 -> 
23:01:41.586 -> :oom(124)@?
23:01:41.586 -> :oom(1532)@?
23:01:41.586 -> :oom(1532)@?
23:01:41.586 -> Received packet of size 8 from 10.0.1.7:37742
23:01:41.586 ->     (to 10.0.1.225:8888, free heap = 1792 B)
23:01:41.586 -> Contents:(144)
23:01:41.586 -> hey
23:01:41.586 -> hey
23:01:41.586 -> 
23:01:41.586 -> :oom(236)@?
23:01:41.586 -> :oom(1532)@?
23:01:41.586 -> :oom(68)@?
23:01:41.586 -> Received packet of size 8 from 10.0.1.7:37742
23:01:41.586 ->     (to 10.0.1.225:8888, free heap = 1792 B)
23:01:41.619 -> Contents:(144)
23:01:41.619 -> hey
23:01:41.619 -> hey
23:01:41.619 -> 
23:01:41.619 -> :oom(236)@?
23:01:41.619 -> :oom(212)@ieee80211_phy.c:313
23:01:41.619 -> Fatal exception 29(StoreProhibitedCause):
23:01:41.619 -> epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
23:01:41.619 -> 
23:01:41.619 -> Exception (29):
23:01:41.619 -> epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
23:01:41.619 -> 
23:01:41.619 -> >>>stack>>>
23:01:41.619 -> 
23:01:41.619 -> ctx: sys
23:01:41.619 -> sp: 3fffec10 end: 3fffffb0 offset: 01a0
23:01:41.619 -> 3fffedb0:  40220113 3ffed778 00000002 402200a8  
```

devyte, 2019-11-29:
@SvenSiwek Now [that](https://github.com/esp8266/Arduino/issues/6831#issuecomment-559565448) is a good MCVE. It might even serve as a device test.

Alright, we've been discussing internally, here's a summary. 
What we had before was no buffering: if a packet arrived, and wasn't received by the app fast enough, then receiving the next packet meant overwriting the current one, thereby dropping it. There was some bug on top of that. The behavior was not enough for some applications. 
That was changed to do full buffering, which means every packet received would get queued. The responsability for dropping packets was thereby shifted to the user. The current state seems to be that the incoming packet rate is faster than the ability to pass the packets to the user app. I seem to remember that wasn't the case before, but it seems to be the case now. This also doesn't work because it allows DoS type flooding of the ESP causing it to fail.
The solution must then be an intermediate: do limited buffering.

Possible Solutions:
1. Keep a packet counter across all udp queues. Drop packets if total count is above some max. For max ==1 you get the solution before the mentioned fix. For max some huge number, you get the current solution. Con: count doesn't take fragmentation into account. 
2. Keep a total size across all packets across all udp queues. Drop packets if total size is above some max. Con: doesn't take available free heap into account. 
3. Check available heap. Drop packets if available heap is below some min. Pro: simple to implement  Con: doesn't take differing free mem requirements for different apps into account. 
4. Keep a total size across all packets across all udp queues. Drop packets if total size is significant against available heap. Pro: most flexible. Con: more complex to implement. This is essentially a combination of 2 and 3 above.

TD-er, 2019-11-29:
5. Separate drop/queue rules per port for the UDP.  For example: drop newest packets when received on port X, drop oldest packets when received on port Y.

SvenSiwek, 2019-11-29:
Hi there :)
What was the origin to implement a buffer for UDP ? 
The UDP protocoll was invented/implemented with the idea of possible packet loss.

If I wanted to be sure to receive all packets I have to use TCP/IP .

So, please forgive me ... I do not understand the whole buffering concept here.

If you still want buffering - make it transparent to the sketch:

You can determine  size of that buffer in the sketch  and what should happen if the buffer is full.
Ignore new packets arring, or dismiss the oldest in the buffer....

One single question: If I understood you right, an interal buffer was flooded.
Why I can't see that with the  `int packetSize = port.parsePacket(); ` ? 

I never saw a value > 500 in the output of my MCVE 

have a nice weekend
Sven

TD-er, 2019-11-29:
The buffer keeps multiple packets. So it is not a buffer issue for a single packet.

SvenSiwek, 2019-11-29:
> The buffer keeps multiple packets. So it is not a buffer issue for a single packet.

the penny dropped

d-a-v, 2019-11-29:
> What was the origin to implement a buffer for UDP ?

The network stack receives data while your sketch is doing something else, so packet data need to be buffered until sketches ask for them.

> If you still want buffering - make it transparent to the sketch:

It already is

>  Ignore new packets arring, or dismiss the oldest in the buffer....

I'm not sure this kind of setting is good on the arduino side. The same question was asked for serial data. What to do when the input buffer is full?, and the answer was not clear (currently new data are discarded when buffer is full).

> Why I can't see that with the int packetSize = port.parsePacket(); ?

Because you see the oldest first.

devyte, 2019-11-29:
By buffer I really mean a packet list.
The ESP has a somewhat unsmooth way of processing things. Received packets are stored until the sketch asks for them. At this point the sketch is expected to get all packets possible. 
There was a case where a stream of packets was resulting in drops for no good reason. The buffering implemented was meant to simply allow the user to receive the packets despite possibly processing peaks. The point was: packets may be lost, but let's at least not have the core be the cause. Either lose them in transit, or have the user app make the decision not to process some of them. 
As I mentioned, I seem to remember testing at some point that the packet throughput could be handled by the processor. Given your MCVE, that seems to not hold now => packet list grows and eats up all heap. This likely should be investigated further, the esp has plenty of processing power, so seeing it not able to keep up in your MCVE surprises me. 
In any case, a cap on the packet list does make sense. The specifics is still up fir discussion

TD-er, 2019-11-29:
>  so seeing it not able to keep up in your MCVE surprises me.

In the loop() call a lot is written to serial port.
If the buffer of the serial port is full, writing to it will become blocking.
So maybe only aggregate the counts of packets and only print a counter every second, or maybe every 100 packets?

mhightower83, 2019-12-02:
@d-a-v Sorry for the delay. I cannot see a low-cost way to do that. Any largest chunk of memory you have cached can shrink through having additional new allocations carved from it until it is smaller than one of the other free blocks that were not in your large free list cache.

PR #6895 (2019-12-10): udp: limit buffer depth
This commit avoids OOMs on an udp corner case where a delay() in the main
loop would allow memory filling.

A memory leak has been observed with such semantically forbidden delay,
unsolved yet, and preventing to use a simple counter instead of walking
through a linked list.  The count limit is however small.

fixes #6831
closed 2019-12-10 by d-a-v