esp8266/Arduino #6832: Unable to clone full platform (LWIP2 git offline)

TD-er, 2019-11-24:
The repository https://git.savannah.nongnu.org/git/lwip.git is offline.
This means the platform cannot be fetched.

Is it possible to define some archive somewhere as backup?
Now it is this external repository, but it may happen to some external repository some other time, so maybe it is good to be able to define some fallback location somewhere?

devyte, 2019-11-25:
Savannah seems to be online now.

> possible to define some archive somewhere as backup

The only thing I can think of would be to clone it into github, but that would require maintenance to keep it up to date, which we're not willing to do atm. If anyone can think of some other way (mirror? automated update?), please post it here.

Closing for now.

d-a-v, 2019-11-25:
@TD-er 
This repository's submodules should be cloned not recursively,
with `git submodule update --init`
not with `git submodule update --init --recursive`
Upstream lwip being a submodule of lwip2 which is a submodule of this repo, it doesn't matter if savannah is unreachable at the time of cloning. The core uses precompiled version of lwip2 anyway.

TD-er, 2019-11-25:
@d-a-v So that's a configuration error where?
In this repo, or in platformio?

d-a-v, 2019-11-25:
It's the way this repo is cloned. 
What you reported was also an issue in CI, solved [like this](https://github.com/esp8266/Arduino/blob/master/.travis.yml#L10) (unfortunately Travis CI's submodule option is implicitely recursive so it is disabled and made with fingers instead)

Can you point me to the PIO git-setup scripts ?

TD-er, 2019-11-25:
> Can you point me to the PIO git-setup scripts ?

Well, that question alone I cannot truthfully answer, since I hardly understand myself how the PIO stuff is being handled.
I think the magic is happening in the `builder` folder of this repo: https://github.com/platformio/platform-espressif8266

mcspr, 2019-11-25:
See https://github.com/platformio/platformio-core/blob/develop/platformio/vcsclient.py
afaik those are invoked when package points to a git url

d-a-v, 2019-11-25:
@ivankravets 
Do you think `--recursive` can be optional in [these lines](https://github.com/platformio/platformio-core/blob/17b4f0b4dd9a0f9f15b2142277455f383d752908/platformio/vcsclient.py#L169-L174) or per-project disabled ?

ivankravets, 2019-11-26:
@d-a-v Oh, I'm not sure. I think it's better to use PRE script and clone framework before to "packages" folder. See http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

TD-er, 2019-11-30:
The site for the LWIP git may be online, but more than 50% of the time it is queried, it will fail to download.
So a lot of my Travis builds fail on this LWIP install.

d-a-v, 2019-12-01:
@TD-er the reason why we use this external repository is because it's the official one and we prefer to not maintain an up-to-date copy.
I understand your issue and we had it too, like mentioned above (I explained how we solved it)

We can
- try and fix PIO scripts to not recursively download submodule
- modify lwip2 to download lwIP upstream sources by script instead of by submodule

TD-er, 2019-12-01:
@d-a-v 
Is either of these 2 options something I should do/change myself? Or is it something you plan to change in the scripts for this repo?

d-a-v, 2019-12-04:
@TD-er I'd prefer some help from PIO nerds :]

TD-er, 2019-12-04:
Hmm it may be hard to find nerds here.... ;)

TD-er, 2019-12-06:
Can this be re-opened, as it is really a problem.
It takes a lot of time and lots and lots of trial and error to get the repo to be initialized.

```
Submodule path 'tools/sdk/lwip2/builder': checked out '354887a25f83064dc0c795e11704190845812713'
Submodule 'lwip2-src' (https://git.savannah.nongnu.org/git/lwip.git) registered for path 'tools/sdk/lwip2/builder/lwip2-src'
Cloning into 'C:/Users/gijs/.platformio/packages/_tmp_installing-4ht_rpqq-package/tools/sdk/lwip2/builder/lwip2-src'...
fatal: unable to access 'https://git.savannah.nongnu.org/git/lwip.git/': The requested URL returned error: 502
fatal: clone of 'https://git.savannah.nongnu.org/git/lwip.git' into submodule path 'C:/Users/gijs/.platformio/packages/_tmp_installing-4ht_rpqq-package/tools/sdk/lwip2/builder/lwip2-src' failed
Failed to clone 'lwip2-src'. Retry scheduled
Cloning into 'C:/Users/gijs/.platformio/packages/_tmp_installing-4ht_rpqq-package/tools/sdk/lwip2/builder/lwip2-src'...
fatal: unable to access 'https://git.savannah.nongnu.org/git/lwip.git/': The requested URL returned error: 502
fatal: clone of 'https://git.savannah.nongnu.org/git/lwip.git' into submodule path 'C:/Users/gijs/.platformio/packages/_tmp_installing-4ht_rpqq-package/tools/sdk/lwip2/builder/lwip2-src' failed
Failed to clone 'lwip2-src' a second time, aborting
Submodule path 'tools/sdk/ssl/bearssl': checked out '89454af34e3e61ddfc9837f3da5a0bc8ed44c3aa'
Failed to recurse into submodule path 'tools/sdk/lwip2/builder'
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', '2.6.2', 'https://github.com/esp8266/Arduino.git', 'C:\\Users\\gijs\\.platformio\\packages\\_tmp_installing-4ht_rpqq-package']
The terminal process terminated with exit code: 1
```

Not sure why it keeps on trying to fetch it, as I have not removed my framework directories..
Maybe these get thrown away as soon as you upgrade platformIO?

mcspr, 2019-12-06:
Perhaps "modify lwip2 to download lwIP upstream sources by script instead of by submodule" is the correct solution? Maybe even other things related to the tree only tangentially and are useless for a basic user of the repo where the goal is sketch not the Core development.

e.g. things like `[submodule "libraries/SoftwareSerial"]` are, but both bearssl and lwip are already shipped as pre-built libraries

TD-er, 2019-12-06:
@ivankravets Can you give me any pointers here, where to look?
I am willing to clone everything I needed to some server I have, just to get this working again.

PR #6887 (2020-02-18): lwip2 updates: no more git sub-sub-module deps, faster checksum, backlog limitation and other fixes
* upstream lwIP is not sub-sub-moduled anymore (a light `git clone` with proper ref is used instead)
* Allows IPv4 and IPv6 DNS and SNTP server configured via DHCP to co-exist (patch against upstream)

fixes #6832

*edit - other lwip2 submodule updates*:

- Switches to lwIP checksum # 3 (closes #6998 - thanks @rsaxvc for https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/42)
- Fixes `weak` usage (thanks @mcspr for https://github.com/d-a-v/esp82xx-nonos-linklayer/issues/43)
- Allows lwip2 comp [... cut]
closed 2020-02-18 by d-a-v