esp8266/Arduino #6834: esptool write_flash: error: argument <address> <filename>: Must be pairs of an address and the binary filename to write there SPIFFS Upload failed!

CharlesSee, 2019-11-25:
Operating system: Windows 10
Python version: 3.8
ESP hardware in use: Bigtree ESP-01S

I have uploaded the default esp3d.ino to the ESP8266, and have the "ESP3D" appearing in my wifi list on my phone. So the default sketch seems to be running ok.

I'm trying to create the SPIFFS File system section on the other half of the ESP8266 card memory.

In arduino IDE 1.8.10, When I click tools, Sketch Data Upload, I get the message / error text listed below. I have no idea how to correct this message or get the SPIFFS loaded..

I have not changed any specific settings, apart from selecting the card and memory 1M/512M allocation.

I noted that Under the tool "Flash Size" I select : 1M (FS: 512KB OTA :~245KB), but on the youtube videos I always see the wording SPIFFS shown like : 4M( 1M SPIFFS ).  There are NO specific SPIFFS options listed on my Flash Size list.

All the youtube videos and tutorials I have seen just show the upload.py working without errors.
---------------------------------------------------
I reported this to github.com/espressif/esptool/issues#486, but they say :

Hi @CharlesSee ,This has something to do with the command line being generated by Arduino IDE, it's not generating a valid esptool.py command line. There's nothing we can do to fix this from the esptool side.Please report it to the Arduino ESP8266 project: https://github.com/esp8266/arduino/issues
--

Hopefully someone can point me where I can look to try to get the default SPIFFS loaded.
All I want to do is get the default SPIFFS file system loaded onto my ESP-01 card.

thanks.

-----------------------------------------------------
Error message when pressing "sketch data upload" :
---------------------------------------------------
[SPIFFS] data : C:\Users\Charles\AppData\Local\Temp\untitled551208617.tmp\sketch_nov23a\data
[SPIFFS] size : 512
[SPIFFS] page : 256
[SPIFFS] block : 8192
[SPIFFS] upload : C:\Users\Charles\AppData\Local\Temp\arduino_build_457769/sketch_nov23a.spiffs.bin
[SPIFFS] address : 0x7B000
[SPIFFS] reset : --before default_reset --after hard_reset
[SPIFFS] port : COM3
[SPIFFS] speed : 115200
[SPIFFS] python : C:\Users\Charles\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\tools\python.exe
[SPIFFS] uploader : C:\Users\Charles\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\tools\upload.py

usage: esptool write_flash [-h] [--erase-all]

                       [--flash_freq {keep,40m,26m,20m,80m}]
                       [--flash_mode {keep,qio,qout,dio,dout}]
                       [--flash_size FLASH_SIZE]
                       [--spi-connection SPI_CONNECTION] [--no-progress]
                       [--verify] [--encrypt]
                       [--ignore-flash-encryption-efuse-setting]
                       [--compress | --no-compress]
                       <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument

: Must be pairs of an address and the binary filename to write there
SPIFFS Upload failed!

devyte, 2019-11-25:
CC @earlephilhower for confirmation 

The sketch data upload plugin is currently broken, at least for windows. I think this is covered by issue #6829 and the relevant PR #6807. You can see in #6807 that there's a PR pending in the plugin repo. Once that is merged, #6807 can be merged.

If you're feeling adventurous, you could try the plugin with the pending PR plus latest git here with #6807 on top, and report back. It would help move things along.

earlephilhower, 2019-11-25:
Dupe #6829 related to spiffs uploader.  Tracking there and the referenced PR.

closed 2019-11-25 by earlephilhower