esp8266/Arduino #6836: ESP8266 Crashing with exception 28

majorlennox, 2019-11-25:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-8266 12E]
- Core Version:			[newest]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[NodeMCU]
- Flash Mode:			[unknown]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Recently, I've updated my code for my ESP8266, noticed that the code did not work and then installed the previous working code back again. Somehow though, the ESP8266 now is caught in a restart loop, constantly crashing with following stack. I know I am not supposed to paste the whole sketch, however I am not aware of which part is causing the issue.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "xx"
#define FIREBASE_AUTH "xx"
#define WIFI_SSID "xx"
#define WIFI_PASSWORD "xx"
#define WIFI_LED D5
#define STARTUP_STATE 1
#define PATH_LAMP "/tripod/status"
#define PATH_CHECK "/tripod/check"
#define LAMPPIN D2
String returnData = "1";
String returnDataTemp = "1";
int stateint;
//Call to ESpressif SDK
extern "C" {
#include <user_interface.h>
}

FirebaseData firebaseData;
FirebaseData firebaseData2;
uint8_t mac[6] {0xA8, 0xD9, 0xB3, 0x0D, 0xAA, 0xAF};
void setup() {

  pinMode(LAMPPIN, OUTPUT);
  //Lampe Startmodus:
  String state = "0";
  if (STARTUP_STATE == 1) {
    turnOn();
    state = "1";
  }

  else if (STARTUP_STATE == 0) {
    turnOff();
    state = "0";
  }
  WiFi.persistent(false);
  Serial.begin(115200);

  Serial.println(WiFi.macAddress());
  pinMode(WIFI_LED, OUTPUT);
  IPAddress ip(192, 168, 0, 250);
  IPAddress gateway(192, 168, 0, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress dns(8, 8, 4, 4);
  // WiFi.config(ip, gateway, subnet);
  wifi_set_macaddr(0, const_cast<uint8*>(mac));
  Serial.println(WiFi.macAddress());

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    digitalWrite(WIFI_LED, LOW);
    delay(100);
    digitalWrite(WIFI_LED, HIGH);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  digitalWrite(WIFI_LED, HIGH);
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Gateway address: ");
  Serial.println(WiFi.gatewayIP().toString());
  Serial.print("AP subnet mask: ");
  Serial.println(WiFi.subnetMask());

  delay(1000);
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Serial.println("Debug");
  delay(1000);
  if (Firebase.setString(firebaseData, PATH_LAMP, "1"))
  {
    Serial.println("PASSED");
  }
  else {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  stateint = state.toInt();  // initial setup
  while (!Firebase.beginStream(firebaseData, PATH_LAMP))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  Firebase.setReadTimeout(firebaseData, 300);

}

void loop() {
  int firebaseResult = firebaseGet(PATH_LAMP).toInt();
  if (firebaseResult != stateint) {
    if (firebaseResult == 1) {
      turnOn();
    }
    else if (firebaseResult == 0) {
      turnOff();
    }
  }

}
void turnOn() {
  digitalWrite(LAMPPIN, HIGH);
  Serial.println("ON");
  stateint = 1;
}
void turnOff() {
  digitalWrite(LAMPPIN, LOW);
  Serial.println("OFF");
  stateint = 0;
}
String firebaseGet(String s) {
  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
  if (firebaseData.streamAvailable())
  {
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    if (firebaseData.dataType() == "string") {
      returnDataTemp = firebaseData.stringData();
      if (returnDataTemp == "1" || returnDataTemp == "0") {
        returnData = returnDataTemp;
        Firebase.setString(firebaseData2, PATH_CHECK, "check");
      }
    }
    Serial.println("------------------------------------");
    Serial.println();
  }
  return (returnData);
}
```

### Crash Stack

```
Exception (28):
epc1=0x4000e1e0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffca0 end: 3fffffc0 offset: 01a0
3ffffe40:  3fff34cc 3ffeed38 000000ff 4021c238  
3ffffe50:  3fff340a 00000000 3fff30e4 3fff30e4  
3ffffe60:  3fff34cc 3ffeed38 0000000a 40206856  
3ffffe70:  00000000 00000146 00000064 00000010  
3ffffe80:  00285b2e 00000000 00000064 40212cf2  
3ffffe90:  00000136 00000010 00000020 3ffef7a0  
3ffffea0:  3fff49a4 3fff30eb 3ffef090 00000000  
3ffffeb0:  3fff2794 00000001 3ffeedf4 00000001  
3ffffec0:  00000000 00000000 00000001 00000009  
3ffffed0:  3ffeee00 000022ff 00000001 00000001  
3ffffee0:  3ffeee18 00000001 00000000 40226f06  
3ffffef0:  3fffff10 3ffef090 3fff0dd4 3ffef228  
3fffff00:  00000000 3ffef090 3ffeed38 402078a0  
3fffff10:  3ffef7a0 3ffef090 3ffef7a0 40201d41  
3fffff20:  3fffff80 3ffef090 3ffeed38 40207948  
3fffff30:  3ffe8765 3ffef090 3ffeed38 402015f1  
3fffff40:  0000000e 00000000 3ffef1e8 40223040  
3fffff50:  80efeffe 0000000e 3fffff8c 4021261c  
3fffff60:  3fffdad0 3fffff8c 3ffe8765 3ffef228  
3fffff70:  3fffdad0 00000000 3ffef1e8 402018d8  
3fffff80:  00000000 00000000 00000001 3fff2a0c  
3fffff90:  000e000f 00000000 3ffef1e8 40212cc9  
3fffffa0:  3fffdad0 00000000 3ffef1e8 40212dd8  
3fffffb0:  feefeffe feefeffe 3ffe8524 4010025d  
<<<stack<<<
```

Decoding that stack hasn't helped. Up- and downgrading libraries didnt help. Wiping the flash didnt help..
Hope you can help!

earlephilhower, 2019-11-25:
Without a full MCVE, there's little we can do.  But looking at the decoded stack trace I see:
````
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4000e1e0
EXCVADDR: 0x00000000

Decoding stack results
0x4021c238: BearSSL::WiFiClientSecure::read() at C:\Users\Finn\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 374
````

That means you're likely passing a NULL pointer to the read() (i.e. the destination buffer is invalid or the connection you're trying to read from was never made), or one of the object's members is NULL.  That's a common sign of out-of-memory errors, which is where I'd suggest you begin looking.  Turning on full debug options should also provide more info in tracing your problem.

devyte, 2020-02-02:
Closing due to lack of feedback in over two months.

closed 2020-02-02 by devyte