esp8266/Arduino #6838: Exception (28) sometimes...

zejulio, 2019-11-25:
Hello,

I have my project working on several places with different AP's and network configuration.
I also have this working in the same place without any issues, I think it's something related to Enterasys old AP's, when ESP8266 connects to AP immediately trows this exception and resets... in some places (at the same site, with same network configuration) connects and works as expected, this site has several models of Enterasys AP's, this is why I think it's related to OLD AP's, and works with new AP's...
Anyone had this type of issues?
I had issues with Cisco AP with WPA2 TKIP networking setting that also got me connection issues, but did not got exception at that time...

Thanks!

Exception (28):
epc1=0x4021a53f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
ctx: sys 
sp: 3ffffcb0 end: 3fffffb0 offset: 01a0
>>>stack>>>
3ffffe50:  7fffffff 3ffeb8ac 3ffeb8ac 00000001  
3ffffe60:  4021f688 3ffefcb0 3fff352c 3ffed888  
3ffffe70:  3ffed8c0 3ffefcb0 3fff352c 3ffed888  
3ffffe80:  3ffed83e 3ffed848 3fffc200 00000010  
3ffffe90:  00000011 3ffed82a ffffffbe 0000c007  
3ffffea0:  3ffed84e 4021f433 00000010 3fff352c  
3ffffeb0:  3ffed820 3ffed8d8 3fff1b18 3fffd9d0  
3ffffec0:  00000000 00000000 00000000 fffffffe  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 0000001e 00000000  
3fffff30:  3fff38ec 4021efdb 3ffeea30 3fff352c  
3fffff40:  00000000 3ffefcb0 3ffeea30 3ffed820  
3fffff50:  3ffed820 000000bc 00000000 0000001e  
3fffff60:  00000000 00174bc9 4021442b 3ffeea30  
3fffff70:  3ffed814 3fffdcc0 3ffeb060 3ffeb060  
3fffff80:  00000010 3ffeea30 00000000 3fff1b30  
3fffff90:  40213d5b 3fffdab0 00000000 40211abb  
3fffffa0:  3ffeb060 40000f49 3fffdab0 40000f49  
<<<stack<<<
 ets Jan  8 2013,rst cause:1, boot mode:(3,3)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ffea7d3
~ld

earlephilhower, 2019-11-25:
You have ignored the template information requests, not provided the core version, not provided an MCVE, and not decoded the stack trace.  Closing as there's nothing here for the core to look at.

For general questions use https://www.esp8266.com forums or Gitter https://gitter.im/esp8266/Arduino .

closed 2019-11-25 by earlephilhower

zejulio, 2019-11-25:
Hello,

 

How can I can decode the stack trace?

I would like help debugging this issue…

If I know how to do that, I would not asked for help..

 

Thanks,

 

Julio Silva

devyte, 2019-11-25:
@zejulio readthedocs && googleIsYourFriend

zejulio, 2019-11-25:
What docs?
I've searched google for this and was not able to find solution, that was why I've posted...

Pablo2048, 2019-11-26:
So it seems like your google-fu is too weak...
https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html
https://github.com/me-no-dev/EspExceptionDecoder
https://www.esp8266.com/viewtopic.php?f=32&t=11265

zejulio, 2019-11-26:
Indeed :(...
I will lookup those posts looks it will really help.
The main issue I have it's that it only happens sometimes, and I cannot always reproduce this...

Thanks for your help!