esp8266/Arduino #6839: Not uploding code in nodeMCU esp8266 v3

Araf773, 2019-11-26:
NodeMCU esp8266 v3 module in not blinking the builtin led when it is connected to pc. And not uploading code also over heating. and error message is : 
 ""Arduino: 1.8.10 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:

IROM   : 228360          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 26564   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1236  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 1520  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 24896 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 257680 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27652 bytes (33%) of dynamic memory, leaving 54268 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM4
Traceback (most recent call last):
  File "C:\Users\Moznu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1/tools/upload.py", line 69, in <module>
    esptool.main(cmdline)
  File "C:/Users/Moznu/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 2889, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:/Users/Moznu/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/Users/Moznu/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/pyserial\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/Users/Moznu/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/pyserial\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM4': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
serial.serialutil.SerialException: could not open port 'COM4': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.  ""

How can I solve this problem?

Pablo2048, 2019-11-26:
Maybe add COM4: to your PC? Or change the selected com port to valid one?... ;-)

Araf773, 2019-11-26:
When I connecting this board to pc , it starts to overheating very much. What shoult i do now?

Pablo2048, 2019-11-26:
Try another one? ... You give near zero informations (you ignored issue template) so from my point of view it seems like hardware problem now.

TD-er, 2019-11-26:
Also make sure nothing else is connected to the board when you connect it to the PC.
Heating up is a good sign something may break and it would be a shame if it were your PC, so don't leave it connected when it keeps heating up.

devyte, 2019-11-26:
Logs clearly show COM4 not found. You need to fix your serial port.
Closing due to user error.

closed 2019-11-26 by devyte