esp8266/Arduino #6845: ESP8266 always restart after while from code running

smartkodian, 2019-11-27:
#### Platform

- Hardware:			       [ESP8266 12E - Development Board]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE 1.8.10 ]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu 1.0 12E]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

Hi Guys, 

I have sketch upload it in my ESP8266 NodeMcu and after code running around 3 to 5 min then the NodeMcu restart and after I decoded the exceptions code I got this result:

`
Decoding stack results
0x40201d22: raise_exception() at C:\Users\moham\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_postmortem.cpp line 245
0x402017e1: loop_task(ETSEvent*) at C:\Users\moham\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_main.cpp line 190
0x4020184c: esp_yield() at C:\Users\moham\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_main.cpp line 100
0x40201def: __delay(unsigned long) at C:\Users\moham\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40201dfa: __delay(unsigned long) at C:\Users\moham\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266\core_esp8266_wiring.cpp line 57
0x4020109e: ModeTwo(float) at C:\Users\moham\Documents\Arduino\Mu_project_new/Mu_project_new.ino line 34
0x4020114c: ModeOne(float) at C:\Users\moham\Documents\Arduino\Mu_project_new/Mu_project_new.ino line 19
0x402010cd: ModeTwo(float) at C:\Users\moham\Documents\Arduino\Mu_project_new/Mu_project_new.ino line 32
0x4020114c: ModeOne(float) at C:\Users\moham\Documents\Arduino\Mu_project_new/Mu_project_new.ino line 19
0x402010cd: ModeTwo(float) at C:\Users\moham\Documents\Arduino\Mu_project_new/Mu_project_new.ino line 32
0x4020114c: ModeOne(float) at C:\Users\moham\Documents\Arduino\Mu_project_new/Mu_project_new.ino line 19
0x402010cd: ModeTwo(float) at C:\Users\moham\Documents\Arduino\Mu_project_new/Mu_project_new.ino line 32`

and my sketch code is:

`
int SetPoint = 30;
int C = 24;
void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(2000);
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED on (HIGH is the voltage level)
}

void loop() {
  ModeOne(0);
}

void ModeOne(float OutputModeOne) { //This Mode To Increase Hot Water Flow
  while (OutputModeOne <= 180) {
    delay(200);
    if ((C - SetPoint) < -3) {
      OutputModeOne = OutputModeOne + 90;
      if (OutputModeOne > 175 ) {
        OutputModeOne = 175;
      }
      ModeTwo(OutputModeOne);
    }
  }
}

void ModeTwo(float  OutputModeTwo) {
  while (OutputModeTwo >= 0) {
    delay(200);
    if ((C - SetPoint) < -3) {
      OutputModeTwo = OutputModeTwo - 90;
      if (OutputModeTwo < 5 ) {
        OutputModeTwo = 5;
      }
      ModeOne(OutputModeTwo);
    }
  }
}

`

But I don't know what is the issue here is it in my code or from IDE it self!

anyone help me please?

Thanks for all :-)

d-a-v, 2019-11-27:
> around 3 to 5 min

Maybe the time needed, with your delays, to fill the stack.
Do your recursive calls ever get out from the initial `ModeOne(0)` ?

smartkodian, 2019-11-27:
> > around 3 to 5 min
> 
> Maybe the time needed, with your delays, to fill the stack.
> Do your recursive calls ever get out from the initial `ModeOne(0)` ?

Thanks for reply to me :)

yes the code start from `ModeOne(0)` !

Do you mean need a delay?

devyte, 2019-11-27:
@smartkodian your rwcursion doesn't end, it just goes back and forth (slowly) until the ESP runs out of stack.
Closing due to user error.

closed 2019-11-27 by devyte