esp8266/Arduino #6851: SD Card demos do not compile, throw compiler errors

AcuarioCat, 2019-11-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[30-Oct-19]
- Development Env:		[|other]
- Operating System:		[Windows|]

### Settings in IDE

- Module:  			[|Wemos D1 mini r2|]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description
The SD card demos do not compile

Compiler throws the following errors (and lots more...)
FatLib.h:27: In file included from
SdFat.h:33: from
SDFS.h:36: from
SD.h:25: from
SD.cpp:1: from
 
ArduinoFiles.h: 122:7: error: redefinition of 'class fs::File
 class File *: public FatFile, public Stream {
 
SD.h:24: In file included from
SD.cpp:1: from
 
FS.h: 51:7: error: previous definition of 'class fs::File
 class File *: public Stream
 
FatLib.h:28: In file included from
SdFat.h:33: from
SDFS.h:36: from
SD.h:25: from
SD.cpp:1: from
FatFileSystem.h: In member function fs::File FatFileSystem::open(const char*, oflag_t)
 
FatFileSystem.h: 95:13: error: 'class fs::File' has no member named 'open
   tmpFile.open(vwd(), path, oflag)
 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
  SD card datalogger

  This example shows how to log data from three analog sensors
  to an SD card using the SD library.

  The circuit:
   analog sensors on analog ins 0, 1, and 2
   SD card attached to SPI bus as follows:
 ** MOSI - pin 11
 ** MISO - pin 12
 ** CLK - pin 13
 ** CS - pin 4

  created  24 Nov 2010
  modified 9 Apr 2012
  by Tom Igoe

  This example code is in the public domain.

*/

#include <SPI.h>
#include <SD.h>

const int chipSelect = 4;

void setup() {
	// Open serial communications and wait for port to open:
	Serial.begin(9600);
	while (!Serial) {
		; // wait for serial port to connect. Needed for Leonardo only
	}

	Serial.print("Initializing SD card...");

	// see if the card is present and can be initialized:
	if (!SD.begin(chipSelect)) {
		Serial.println("Card failed, or not present");
		// don't do anything more:
		return;
	}
	Serial.println("card initialized.");
}

void loop() {
	// make a string for assembling the data to log:
	String dataString = "";

	// read three sensors and append to the string:
	for (int analogPin = 0; analogPin < 3; analogPin++) {
		int sensor = analogRead(analogPin);
		dataString += String(sensor);
		if (analogPin < 2) {
			dataString += ",";
		}
	}

	// open the file. note that only one file can be open at a time,
	// so you have to close this one before opening another.
	File dataFile = SD.open("datalog.txt", FILE_WRITE);

	// if the file is available, write to it:
	if (dataFile) {
		dataFile.println(dataString);
		dataFile.close();
		// print to the serial port too:
		Serial.println(dataString);
	}
	// if the file isn't open, pop up an error:
	else {
		Serial.println("error opening datalog.txt");
	}
}

```

### Debug Messages

```
Debug messages go here
```

AcuarioCat, 2019-11-28:
Updating to 2.6.2 solved the problem.

closed 2019-11-28 by AcuarioCat

YoussefNassar, 2021-11-11:
Hi :)
I'm facing the same issue.
What did you update to 2.6.2 exactly ?
Thanks

AcuarioCat, 2021-12-14:
It was a while ago, it would have been the esp8266 core. It has now moved on a bit and I haven't tried the code on newer cores.