esp8266/Arduino #6852: Conflicting type definition depending on include order

dok-net, 2019-11-28:
For some reason that shouldn't matter here, if Arduino.h gets (indirectly) included in Schedule.h, this redefinition bug is raised by the compiler:

cores\esp8266\sntp-lwip2.cpp:
```
typedef struct __tzrule_struct
{
  char ch;
  int m;
  int n;
  int d;
  int s;
  time_t change;
  int offset;
} __tzrule_type;
```
tools\sdk\libc\xtensa-lx106-elf\include\time.h:
```
typedef struct __tzrule_struct
{
  char ch;
  int m;
  int n;
  int d;
  int s;
  time_t change;
  long offset; /* Match type of _timezone. */
} __tzrule_type;
```
@d-a-v I was asked by @devyte to refer this one to you.

d-a-v, 2019-11-28:
I suggest removing the struct(s) from sntp-lwip2 and use instead
```cpp
#include <time.h>
```
(in case of issue, try with
```cpp
#include <../include/time.h>
```
)

d-a-v, 2019-11-28:
I suggest removing the struct(s) from sntp-lwip2 and use instead
```cpp
#include <time.h>
```
(in case of issue, try with)
```cpp
#include <../include/time.h>
```

dok-net, 2019-11-28:
@d-a-v This changes `#include <sys/time.h>` to `#include <time.h>`...

PR #6853 (2019-11-28): Conflict with pre-included Arduino.h (the indirect includes thereof)
Fixes #6852
closed 2019-11-28 by devyte