esp8266/Arduino #6858: PROGMEM PROBLEM 2.6.2

laercionit, 2019-11-29:
Good night, I have a problem and I can't identify. I KNOW it is out of the required tracking standards, but I believe the PROGMEM callback function is not finished with all the features it had in previous CORE.
CORE 2.6.2
ESP8266

I did not put all the code here because it is a very large program, but basically I am taking error in all strings used in PROGMEM. This code is functional in CORE 2.4.2, 2.5.0. I think before there was some automatic function that didn't need to use FPSTR ();

static const char LangLess[]               PROGMEM  = "<";
static const char LangGreater[]            PROGMEM  = ">";
static const char LangEqual[]              PROGMEM  = "=";

String RulesGetCompare(byte RulesID) {
  if      (Rules[RulesID].Compare == RulesCompLess   ) return LangLess;
  else if (Rules[RulesID].Compare == RulesCompGreater) return LangGreater;
  else if (Rules[RulesID].Compare == RulesCompEqual  ) return LangEqual;
}

earlephilhower, 2019-12-01:
Sorry, without an MCVE we have no idea what you're seeing and can't reproduce things.  We still allow PROGMEM C-strings (see any of the BearSSL examples), and `F()` macros are used extensively in the core and libraries.

Please post a MCVE that shows the error you're seeing on 2.6+ but not on 2.4.

laercionit, 2019-12-01:
Basically to keep the code working, I had to include the FPSTR () function manually. This worked perfectly, but in core 2.4.2, I think there was a builder that adjusted this automatically.

static const char LangLess[] PROGMEM = "<";
static const char LangGreater[] PROGMEM = ">";
static const char LangEqual[] PROGMEM = "=";

String RulesGetCompare(byte RulesID) {
if (Rules[RulesID].Compare == RulesCompLess ) return FPSTR(LangLess);
else if (Rules[RulesID].Compare == RulesCompGreater) return FPSTR(LangGreater);
else if (Rules[RulesID].Compare == RulesCompEqual ) return FPSTR(LangEqual);
}

aeakin1, 2019-12-05:
I am having the same issue when leveraging a forked WiFiManager Library. https://github.com/kentaylor/WiFiManager

`Compiling .pio\build\esp-12F\src\ESParaSite_WiFiCore.cpp.o
In file included from C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/sys/stdio.h:6:0,
                 from C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/stdio.h:63,     
                 from C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:32,
                 from lib\ArduinoJson\src/ArduinoJson/Configuration.hpp:93,
                 from lib\ArduinoJson\src/ArduinoJson/Namespace.hpp:7,
                 from lib\ArduinoJson\src/ArduinoJson.hpp:15,
                 from lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\ESParaSite_ConfigPortal.cpp:17:
lib\WiFiManager/WiFiManager.h:32:24: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                        ^
In file included from src\ESParaSite_ConfigPortal.cpp:23:0:
C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
Generating LD script .pio\build\esp-12F\ld\local.eagle.app.v6.common.ld
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
In file included from C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/sys/stdio.h:6:0,
                 from C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/stdio.h:63,
                 from C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:32,
                 from src\ESParaSite.cpp:17:
C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))
                                                                                                                                  ^
lib\WiFiManager/WiFiManager.h:32:24: note: in expansion of macro 'PROGMEM'
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                        ^
In file included from src\ESParaSite.cpp:19:0:
C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
*** [.pio\build\esp-12F\src\ESParaSite_ConfigPortal.cpp.o] Error 1
*** [.pio\build\esp-12F\src\ESParaSite.cpp.o] Error 1`

The offending code block - 

`
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <memory>
#undef min
#undef max
#include <algorithm>
extern "C" {
  #include "user_interface.h"
}

#define WFM_LABEL_BEFORE 1
#define WFM_LABEL_AFTER 2
#define WFM_NO_LABEL 0

const char HTTP_200[] PROGMEM             = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n";
const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
const char HTTP_STYLE[] PROGMEM           = "<style>body,textarea,input,select{background: 0;border-radius: 0;font: 16px sans-serif;margin: 0}textarea,input,select{outline: 0;font-size: 14px;border: 1px solid #ccc;padding: 8px;width: 90%}.btn a{text-decoration: none}.container{margin: auto;width: 90%}@media(min-width:1200px){.container{margin: auto;width: 30%}}@media(min-width:768px) and (max-width:1200px){.container{margin: auto;width: 50%}}.btn,h2{font-size: 2em}h1{font-size: 3em}.btn{background: #0ae;border-radius: 4px;border: 0;color: #fff;cursor: pointer;display: inline-block;margin: 2px 0;padding: 10px 14px 11px;width: 100%}.btn:hover{background: #09d}.btn:active,.btn:focus{background: #08b}label>*{display: inline}form>*{display: block;margin-bottom: 10px}textarea:focus,input:focus,select:focus{border-color: #5ab}.msg{background: #def;border-left: 5px solid #59d;padding: 1.5em}.q{float: right;width: 64px;text-align: right}.l{background: url('data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAACAAAAAgCAMAAABEpIrGAAAALVBMVEX///8EBwfBwsLw8PAzNjaCg4NTVVUjJiZDRUUUFxdiZGSho6OSk5Pg4eFydHTCjaf3AAAAZElEQVQ4je2NSw7AIAhEBamKn97/uMXEGBvozkWb9C2Zx4xzWykBhFAeYp9gkLyZE0zIMno9n4g19hmdY39scwqVkOXaxph0ZCXQcqxSpgQpONa59wkRDOL93eAXvimwlbPbwwVAegLS1HGfZAAAAABJRU5ErkJggg==') no-repeat left center;background-size: 1em}input[type='checkbox']{float: left;width: 20px}.table td{padding:.5em;text-align:left}.table tbody>:nth-child(2n-1){background:#ddd}</style>";
[... cut ...]
const char HTTP_SAVED[] PROGMEM           = "<div class=\"msg\"><strong>Credentials Saved</strong><br>Trying to connect ESP to the {x} network.<br>Give it 10 seconds or so and check <a href=\"/\">how it went.</a> <p/>The {v} network you are connected to will be restarted on the radio channel of the {x} network. You may have to manually reconnect to the {v} network.</div>";
const char HTTP_END[] PROGMEM             = "</div></body></html>";
`

compiles fine on Arduino core 2.5.2 and the issues start at Arduino core 2.6.1 (Platformio ESP8266 Platform 2.2.3 and 2.3.0 in , respectively)

d-a-v, 2019-12-09:
@aeakin1 This is a different issue.
Upstream WiFiManager [has been updated](https://github.com/tzapu/WiFiManager/pull/928). @kentaylor's fork was not reported until today.

devyte, 2020-02-02:
No MCVE and no issue template. 
Closing due to not compliant.
@laercionit if you think the problem is real with 2.6.3 or latest git, please make a small MCVE sketch to reproduce, open a new issue, and fill out the required info including the sketch.

closed 2020-02-02 by devyte