esp8266/Arduino #6864: esptool: error: unrecognized arguments: --end

nscnm87, 2019-12-01:
# Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8285 device]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Detailed problem description goes here.
After upgrading the 2.6.2 version i got this error while trying to upload data via "ESP8266 Sketch Data Upload"

### Debug Messages

```
esptool: error: unrecognized arguments: --end

```

jandrassy, 2019-12-01:
update the upload plugin

devyte, 2019-12-01:
Previous comment is correct. A plugin update is needed. 
Closing.

closed 2019-12-01 by devyte

adbensi, 2019-12-18:
I have the same problem.. last **updated plugin ESP8266FS-0.5.0.zip** and SDK 2.6.3

SPIFFS Upload failed!
[SPIFFS] data    : C:\Users\Alexandre\firmware\LockOut\data
[SPIFFS] size    : 3048
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/cadastro.csv

/estado.dat

/formSys.dat

/fundo.svg

/historico.csv

/index.html

/logo.png

[SPIFFS] upload  : C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_37956/LockOut.spiffs.bin
[SPIFFS] address  : 0x100000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : COM3
[SPIFFS] speed    : 115200
[SPIFFS] python   : python.exe
[SPIFFS] uploader : C:\Users\Alexandre\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\tools\upload.py

usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]

               [--before {default_reset,no_reset,no_reset_no_sync}]

               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]

               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]

               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}

               ...

esptool: error: unrecognized arguments: --end

SPIFFS Upload failed!

jandrassy, 2019-12-18:
then use the same solution

adbensi, 2019-12-18:
Hi!

 I try the same solution, updating the *plugin ESP8266FS to last version
0.5.0 *
 But, no luck.. may be change mkspiffs.exe .. I will try it too..
 did You see another thing?

 Best Regards,

ArcAIN6, 2020-10-23:
so.. i ran into this problem today out of the blue.. 
Honestly.. scratching my head here. 

How does one go about updating the uploader (seems like such a generic response)

devyte, 2020-10-24:
get.py?

ArcAIN6, 2020-10-24:
Ended up fixing it this morning.
In case anyone runs into this, 
I had to go to https://github.com/esp8266/arduino-esp8266fs-plugin/releases
and download the latest.
Then extract it into the arduino install location under tools folder. 
in my case, C:\Program Files (x86)\Arduino\tools\

After that, restarting arduino, everything worked.

jandrassy, 2020-10-24:
> Ended up fixing it this morning.
> In case anyone runs into this,
> I had to go to https://github.com/esp8266/arduino-esp8266fs-plugin/releases
> and download the latest.
> Then extract it into the arduino [...]

where else would you expect to find the update?

ArcAIN6, 2020-10-24:
> did somebody else install it for you before?
> where else would you expect to find the update?

Well. .that was COMPLETELY helpful interaction. 

I don't know about anyone else, but i can't remember where i've installed every bit of everything over the past few years. 

Sorry, my memory just isn't up to snuff *eyerolls*