esp8266/Arduino #6866: Add Support for building against NONOS SDK 2.2.x 191122

dirkmueller, 2019-12-01:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			2.6.1
- Development Env:		Platformio
- Operating System:		MacOS

### Problem Description

There is a new phy library in NONOS SDK 2.2.x. Given that a new 2.6.x release was released to revert the default SDK to the version before security fixes for WPA were merged, can we please at least optionally enable the new build? I would hope it fixes the regression.

d-a-v, 2019-12-01:
The old-more-recent FW is still available in the options when generic board is enabled.
Only the default changed.

d-a-v, 2019-12-01:
On PIO, the option is called `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191105`.

dirkmueller, 2019-12-01:
Thanks. My understanding is that this is older (November 5th) than the one I am asking about (November 22). Am I mistaken?

d-a-v, 2019-12-01:
Not mistaken. I misunderstood the problem description.
There are phy updates [since 191105](https://github.com/espressif/ESP8266_NONOS_SDK/commits/release/v2.2.x) so we can try them.

PR #6879 (2019-12-04): sdk:22x-191122
another firmware to test
not default: v2.2.1-119-ga0b1311 (shows as SDK:2.2.2-dev(a58da79) in debug mode)
default unchanged

closes #6866
closed 2019-12-04 by d-a-v