esp8266/Arduino #6870: C++ global object initialization order reversed

earlephilhower, 2019-12-02:
@mikee47 and @slaff noted in the Sming repo that the core is calling the c++ ctors in reverse order:
https://github.com/SmingHub/Sming/pull/1825#discussion_r352540500

https://github.com/esp8266/Arduino/blob/074a8016bfafc23c52368d5cca6880b36f102bc0/cores/esp8266/core_esp8266_main.cpp#L199-L206

I believe w/in a single file/TU, they should be called in the order in which they appear.  So calling them backwards may break something if there are inter-object dependencies in global objects.

igrr, 2019-12-02:
https://github.com/esp8266/Arduino/pull/2074 had a link to MCVE, not sure if it still is an issue with recent GCC.

mikee47, 2019-12-02:
I've added the test case from #2074 and yes, it's still reversed. Thank you.

earlephilhower, 2019-12-02:
Closing as the reverse order is still required.

closed 2019-12-02 by earlephilhower