esp8266/Arduino #6873: ESP8266 Doesn't respond to ARP requests

marrold, 2019-12-03:
Hi,

My ESP8266 regularly stops responding to ARP requests. I see this has been a big issue in the past but was recently "resolved"  in #2330  / #6484 however I'm still getting the exact same behaviour using Core 2.6.1. I am using the latest Tasmota firmware.

How can I troubleshoot this? Is anyone else still seeing issues? 
 
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			2.6.1
- Development Env:		Platformio
- Operating System:		MacOS

d-a-v, 2019-12-03:
More details are needed on when and in which conditions it happens,
also with the source code for wifi setup.
As it is Tasmota, maybe you should ask them first.
You can check with bare metal examples if you use arduino.

devyte, 2019-12-03:
Previous comment is correct. Please discuss at Tasmota and reduce your code to pure Arduino core. If you reproduce the issue with just our core, report back here and we'll discuss further. Or just open a new issue and add the details there. 
Closing in the meantime.

closed 2019-12-03 by devyte

marrold, 2019-12-05:
@devyte I have discussed on the Tasmota github issue (See above link) but they were unable to identify an issue within the Tasmota code itself. One interesting thing that came out of the discussion is this appears to affect Mikrotik Access Points.

I was able to replicate with the following ID code. If `wifi_set_sleep_type` is light or modem ARP requests will consistently fail.  When set to none I reliably get responses to ARP requests.

Code:
```
#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID [YOUR SSID]
#define STAPSK  [YOUR PASSWORD]
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

void setup() {
  Serial.begin(115200);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  wifi_set_sleep_type(NONE_SLEEP_T);
  //wifi_set_sleep_type(LIGHT_SLEEP_T);
  //wifi_set_sleep_type(MODEM_SLEEP_T);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  Serial.println("I'm alive");
  delay(30000);
}
```

ARP responses when sleep type set to none:  (1 lost is acceptable IMO)
```
sudo nping --arp-type ARP 10.0.130.109 -c 20
...
SENT (19.0551s) ARP who has 10.0.130.109? Tell 10.0.130.102
RCVD (19.0663s) ARP reply 10.0.130.109 is at 5C:CF:7F:88:E3:2F
...
Raw packets sent: 20 (840B) | Rcvd: 19 (532B) | Lost: 1 (5.00%)
```

ARP responses when sleep type set to light: 
```
sudo nping --arp-type ARP 10.0.130.109 -c 20
...
SENT (1.0130s) ARP who has 10.0.130.109? Tell 10.0.130.102
RCVD (1.0276s) ARP reply 10.0.130.109 is at 5C:CF:7F:88:E3:2F
...
Raw packets sent: 20 (840B) | Rcvd: 5 (140B) | Lost: 15 (75.00%)
```

ARP responses when sleep type set to modem: 
```
sudo nping --arp-type ARP 10.0.130.109 -c 20
...
SENT (1.0130s) ARP who has 10.0.130.109? Tell 10.0.130.102
RCVD (1.0276s) ARP reply 10.0.130.109 is at 5C:CF:7F:88:E3:2F
...
Raw packets sent: 20 (840B) | Rcvd: 3 (84B) | Lost: 17 (85.00%)
```

Thanks