esp8266/Arduino #6876: No multicast UDP packets are received when I2S is active

JPMJPM, 2019-12-03:
Hi,

I am using board "NodeMCU 1.0 (ESP-12E Module)" 2.6.2

Without i2s_begin(); there is not multicast UDP packet loss.
With i2s_begin(); no multicast UDP packets are received.

Any clue ?. Thanks.

Code :

`WiFiUDP Udp;`
`IPAddress multicastIPaddress(224,1,1,3);`
`unsigned int localUdpPort = 6000;`
`char incomingPacket[1500];`
`int packetSize;`

`void setup() {`
`Udp.beginMulticast(WiFi.localIP(), multicastIPaddress, localUdpPort);`
`i2s_begin();`
`}`

`void loop() {`
`packetSize = Udp.parsePacket();`
`if (packetSize) {`
`int len = Udp.read(incomingPacket, packetSize);`
`Serial.printf("len = %d\n", len);`
`}`	
`}`

gibo77, 2020-01-26:
Hi,
How did you make your multicast work?
I could not make it work and my program is already bare bones.
`
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include "user_interface.h"

WiFiUDP Udp;
IPAddress multicastIPaddress(224,1,1,3);
unsigned int localUdpPort = 4210;  // local port to listen on
char incomingPacket[255];  // buffer for incoming packets
//char  replyPacket[] = "Hi there! Got the message :-)";  // a reply string to send back

void setup()
{
  Serial.begin(115200);
  Serial.println();

  WiFi.mode(WIFI_STA); delay(1000);
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");

  wifi_set_sleep_type(NONE_SLEEP_T);
  delay(1000);
  Udp.beginMulticast(WiFi.localIP(), multicastIPaddress, localUdpPort);
  Serial.printf("Now listening...");
}

void loop()
{
  int packetSize = Udp.parsePacket();
  yield();
  if (packetSize)
  {
    // receive incoming UDP packets
    Serial.printf("Received %d bytes from %s, port %d\n", packetSize, Udp.remoteIP().toString().c_str(), Udp.remotePort());
    int len = Udp.read(incomingPacket, 255);
    if (len > 0)
    {
      incomingPacket[len] = 0;
    }
    Serial.printf("UDP packet contents: %s\n", incomingPacket);

  }
}`

earlephilhower, 2021-01-05:
I just ran the test given on master and had no difficulty when running the UDP multicast sender example on https://gist.github.com/hostilefork/f7cae3dc33e7416f2dd25a402857b6c6 

````
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <i2s.h>

WiFiUDP Udp;
IPAddress multicastIPaddress(224,1,1,3);
unsigned int localUdpPort = 6000;
char incomingPacket[1500];
int packetSize;

void setup() {
  Serial.begin(115200);
  delay(3000);
  Serial.printf("\nUDP begin\n");
  Serial.println(multicastIPaddress);
  Serial.println(localUdpPort);
  Udp.beginMulticast(WiFi.localIP(), multicastIPaddress, localUdpPort);
  i2s_begin();
}

void loop() {
  packetSize = Udp.parsePacket();
  if (packetSize) {
    int len = Udp.read(incomingPacket, packetSize);
    Serial.printf("len = %d\n", len);
  }
}
````

On the Linux host (wired) talking to the UDP broadcast 8266:
````
$ /tmp/sender  224.1.1.3 6000
````

On the ESP8266:
````
UDP begin
224.1.1.3
6000
len = 13
len = 13
len = 13
......
````

Closing for now due to age and non-repeatability under current code.

closed 2021-01-05 by earlephilhower