esp8266/Arduino #6883: general question: preprocessor directive for a board #define MACRO to identify both esp32 and esp8266 Boards ?

dsyleixa, 2019-12-05:
which is the correct preprocessor directive for a board #define MACRO to identify both esp32 and esp8266 Boards ?

e.g.,, I need to declare the preprocessor directive
#if defined (SAM)
#include "avr/dtostrf.h"
#elseif defined (? ESP ?) // <<<<<<<<<<<<< ????
#include <stdlib_noniso.h>
#endif

or will the following  work (for the ESP8266 side) ?
#if defined(ESP32) || defined(ESP8266) // <<<<<<<<<<<<< ????
#include <stdlib_noniso.h>
#endif

d-a-v, 2019-12-05:
I think this is fine
> `#if defined(ESP32) || defined(ESP8266)`

Also: [check this](https://github.com/esp8266/Arduino/blob/2.6.2/cores/esp8266/avr/dtostrf.h)

dsyleixa, 2019-12-05:
thank you!
I have included
#include <stdlib_noniso.h>

what do you mean by "Also check this?"
is there an issue about it?

d-a-v, 2019-12-05:
> what do you mean by "Also check this?"

I mean that you can `#include "avr/dtostrf.h"` on esp8266 with the same effect as on SAM

closed 2019-12-05 by devyte

dsyleixa, 2019-12-06:
thanks again, I was recalling that part; actually it's also for differet other settings specific to ESP boards, in contrast to AVR or SAM.