esp8266/Arduino #6885: OTA update spiffs ERROR[4]: Not Enough Space

Lightwell-bg, 2019-12-06:
Hi
I try update spiffs OTA and get error ERROR[4]: Not Enough Space.
The sketch is updated normally. Problem is only with update SPIFFS
Memory allocated as 4/1. In the previous version of the framework (2.5.x), everything works fine.

    HTTPUpload& upload = HTTP.upload();
    if(upload.status == UPLOAD_FILE_START) {
      Serial.setDebugOutput(true);
      WiFiUDP::stopAll();
      Serial.printf("Update: %s\n", upload.filename.c_str());
      Serial.print("Size: "); Serial.println(upload.contentLength);
      size_t content_len = upload.contentLength;
      int cmd = (upload.filename.indexOf("spiffs") > -1) ? 100 : 0; //U_FS : U_FLASH;
      if(!Update.begin(content_len,cmd)) {//start with max available size
              Update.printError(Serial);
      }
    }
    else if(upload.status == UPLOAD_FILE_WRITE) {
      if(Update.write(upload.buf, upload.currentSize) != upload.currentSize) {
        Update.printError(Serial);
      }
    }
    else if(upload.status == UPLOAD_FILE_END) {
      if(Update.end(true)) { //true to set the size to the current progress
        Serial.printf("Update Success: %u\nRebooting...\n", upload.totalSize);
        String answerUpdate = "Status update: " + String(Update.hasError()?"FAIL":"OK") + ". Please wait while the device reboots";
        HTTP.send(302, "text/html", answerUpdate);
        delay(500);
      } else {
        Update.printError(Serial);
      }
      Serial.setDebugOutput(false);
    }

closed 2019-12-12 by devyte

sascha432, 2019-12-29:
I've got the same issue with 2.6.3. OTA SPIFFS update (U_FS=100) fails with not enough space and is working with 2.5.2. Upload via serial works fine
4MB flash
620kb firmware
1MB spiffs
-Wl,-Teagle.flash.4m1m.ld

allex1978, 2020-02-03:
The same issue.

d-a-v, 2020-02-03:
@Lightwell-bg  Please open a new issue by filling all the fields in the presented issue template.
@allex1978 This is a closed issue. Please open a proper new issue by filling all the fields in the presented issue template.