esp8266/Arduino #6892: Error: maximum number of CA certs added (10)

smartkodian, 2019-12-09:
#### Platform

- Hardware:					[ESP-12|
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

Hi Guys,

I have a strange issue! I am working on this example: HTTPSRequestCACertAxTLS.ino
And If I use the original `cacert.h` which contain the cacert information then the connection is established and no problem. But If I try to use the latest cacert code which I followed the steps from this link:
[https://esp8266-notes.blogspot.com/2018/02/how-to-create-cacerth-for-esp8266.html](url) for the same site for sure `api.github.com` and I edit the file cacert.h with new cacert information but there is no connection established and the after debugging I got this error:

```Current time: Mon Dec  9 23:43:17 2019
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: Invalid X509 ASN.1 file (X509 not ok)
Error: maximum number of CA certs added (10) - change of compile-time configuration required
connecting to api.github.com
Error: No trusted cert is available
ERROR: certificate verification failed!
```
the original cacert.h contain:
```

const unsigned char caCert[] PROGMEM = {
  0x30, 0x82, 0x03, 0xc5, 0x30, 0x82, 0x02, 0xad, 0xa0, 0x03, 0x02, 0x01,
  0x02, 0x02, 0x10, 0x02, 0xac, 0x5c, 0x26, 0x6a, 0x0b, 0x40, 0x9b, 0x8f,
  0x0b, 0x79, 0xf2, 0xae, 0x46, 0x25, 0x77, 0x30, 0x0d, 0x06, 0x09, 0x2a,
  0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x01, 0x05, 0x05, 0x00, 0x30, 0x6c,
  0x31, 0x0b, 0x30, 0x09, 0x06, 0x03, 0x55, 0x04, 0x06, 0x13, 0x02, 0x55,
  0x53, 0x31, 0x15, 0x30, 0x13, 0x06, 0x03, 0x55, 0x04, 0x0a, 0x13, 0x0c,
  0x44, 0x69, 0x67, 0x69, 0x43, 0x65, 0x72, 0x74, 0x20, 0x49, 0x6e, 0x63,
  0x31, 0x19, 0x30, 0x17, 0x06, 0x03, 0x55, 0x04, 0x0b, 0x13, 0x10, 0x77,
  0x77, 0x77, 0x2e, 0x64, 0x69, 0x67, 0x69, 0x63, 0x65, 0x72, 0x74, 0x2e,
  0x63, 0x6f, 0x6d, 0x31, 0x2b, 0x30, 0x29, 0x06, 0x03, 0x55, 0x04, 0x03,
  0x13, 0x22, 0x44, 0x69, 0x67, 0x69, 0x43, 0x65, 0x72, 0x74, 0x20, 0x48,
  0x69, 0x67, 0x68, 0x20, 0x41, 0x73, 0x73, 0x75, 0x72, 0x61, 0x6e, 0x63,
  0x65, 0x20, 0x45, 0x56, 0x20, 0x52, 0x6f, 0x6f, 0x74, 0x20, 0x43, 0x41,
  0x30, 0x1e, 0x17, 0x0d, 0x30, 0x36, 0x31, 0x31, 0x31, 0x30, 0x30, 0x30,
  0x30, 0x30, 0x30, 0x30, 0x5a, 0x17, 0x0d, 0x33, 0x31, 0x31, 0x31, 0x31,
  0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x5a, 0x30, 0x6c, 0x31, 0x0b,
  0x30, 0x09, 0x06, 0x03, 0x55, 0x04, 0x06, 0x13, 0x02, 0x55, 0x53, 0x31,
  0x15, 0x30, 0x13, 0x06, 0x03, 0x55, 0x04, 0x0a, 0x13, 0x0c, 0x44, 0x69,
  0x67, 0x69, 0x43, 0x65, 0x72, 0x74, 0x20, 0x49, 0x6e, 0x63, 0x31, 0x19,
  0x30, 0x17, 0x06, 0x03, 0x55, 0x04, 0x0b, 0x13, 0x10, 0x77, 0x77, 0x77,
  0x2e, 0x64, 0x69, 0x67, 0x69, 0x63, 0x65, 0x72, 0x74, 0x2e, 0x63, 0x6f,
  0x6d, 0x31, 0x2b, 0x30, 0x29, 0x06, 0x03, 0x55, 0x04, 0x03, 0x13, 0x22,
  0x44, 0x69, 0x67, 0x69, 0x43, 0x65, 0x72, 0x74, 0x20, 0x48, 0x69, 0x67,
  0x68, 0x20, 0x41, 0x73, 0x73, 0x75, 0x72, 0x61, 0x6e, 0x63, 0x65, 0x20,
  0x45, 0x56, 0x20, 0x52, 0x6f, 0x6f, 0x74, 0x20, 0x43, 0x41, 0x30, 0x82,
[... 154 lines cut ...]
  0x08, 0xc2, 0x89, 0x2d, 0x0c, 0x06, 0x5c, 0x52, 0xc2, 0xb7, 0xc3, 0x84,
  0xc3, 0xb9, 0x0a, 0xc2, 0x98, 0xc3, 0x91, 0x15, 0x5f, 0xc2, 0x9f, 0x12,
  0xc2, 0xbe, 0x7c, 0x36, 0x63, 0x38, 0xc2, 0xbd, 0x44, 0xc2, 0xa4, 0x7f,
  0xc3, 0xa4, 0x26, 0x2b, 0x0a, 0xc3, 0x84, 0xc2, 0x97, 0x69, 0x0d, 0xc3,
  0xa9, 0xc2, 0x8c, 0xc3, 0xa2, 0xc3, 0x80, 0x10, 0x57, 0xc2, 0xb8, 0xc3,
  0x88, 0x76, 0x12, 0xc2, 0x91, 0x55, 0xc3, 0xb2, 0x48, 0x69, 0xc3, 0x98,
  0xc2, 0xbc, 0x2a, 0x02, 0x5b, 0x0f, 0x44, 0xc3, 0x94, 0x20, 0x31, 0xc3,
  0x9b, 0xc3, 0xb4, 0xc2, 0xba, 0x70, 0x26, 0x5d, 0xc2, 0x90, 0x60, 0xc2,
  0x9e, 0xc2, 0xbc, 0x4b, 0x17, 0x09, 0x2f, 0xc2, 0xb4, 0xc3, 0x8b, 0x1e,
  0x43, 0x68, 0xc3, 0x89, 0x07, 0x27, 0xc3, 0x81, 0xc3, 0x92, 0x5c, 0xc3,
  0xb7, 0xc3, 0xaa, 0x21, 0xc2, 0xb9, 0x68, 0x12, 0xc2, 0x9c, 0x3c, 0xc2,
  0x9c, 0xc2, 0xbf, 0xc2, 0x9e, 0xc3, 0xbc, 0xc2, 0x80, 0x5c, 0xc2, 0x9b,
  0x63, 0xc3, 0x8d, 0xc3, 0xac, 0x47, 0xc2, 0xaa, 0x25, 0x27, 0x67, 0xc2,
  0xa0, 0x37, 0xc3, 0xb3, 0x00, 0xc2, 0x82, 0x7d, 0x54, 0xc3, 0x97, 0xc2,
  0xa9, 0xc3, 0xb8, 0xc3, 0xa9, 0x2e, 0x13, 0xc2, 0xa3, 0x77, 0xc3, 0xa8,
  0x1f, 0x4a
};
const unsigned int caCertLen = 1252;
```

I don't  know where is the issue and why connection not established?

what could the reason for that! 

Thanks for all to try to help :)

earlephilhower, 2019-12-09:
Axtls is deprecated and basically unsupported at this point.  It's letting you know that your chain is too long for the way it's built in the 8266 core, so you're not going to be able to do full cert chain w/what you're sending in.

I would recommend going to BearSSL, which should be able to parse any number of certs embedded in a chain (it does it in a streaming-mode as opposed to a block-based one) and is supported now.  The APIs are almost identical, and you can find many examples in the core and documentations.

closed 2019-12-09 by earlephilhower

smartkodian, 2019-12-10:
@earlephilhower many thanks for advice :) I will go to use BearSSL

Many thanks for support