esp8266/Arduino #6901: Arduino#installing-with-boards-manager

stef-ladefense, 2019-12-12:
hello,

the url https://arduino.esp8266.com/stable/package_esp8266com_index.json
in Arduino#installing-with-boards-manager don't work for install.
ERR_CONNECTION_TIMED_OUT

thank

earlephilhower, 2019-12-12:
Looks to have been a temporary hiccup. It's working now.  Closing.

stef-ladefense, 2019-12-12:
no not work

[image]

sorry but it's wrong 95% of the time,
9 times out of 10 since 2.60 when I use the IDE, I have an error of access to this json and any reference to the 8266 disappears!
it's not usable at all, I'm forced to use the url of the current version, without the possibility of updating

d-a-v, 2019-12-12:
@stef-ladefense It seems to be an issue with your network.
You can open a ping tool, or a terminal with the following command and check responsiveness.
```
ping arduino.esp8266.com
```
Alternatively you can try with the one described there https://d-a-v.github.io/
(the official 2.6.2 version is the same)
(both this source and installable zip files are from github)

stef-ladefense, 2019-12-12:
Not work from France 

 Direct fiber internet

 

[C:\Program Files\JPSoft\TCCLE14x64]ping arduino.esp8266.com

 

Envoi d’une requête 'ping' sur arduino.esp8266.com [104.131.82.128] avec 32 octets de données :

Délai d’attente de la demande dépassé.

Délai d’attente de la demande dépassé.

Délai d’attente de la demande dépassé.

Délai d’attente de la demande dépassé.

 

Statistiques Ping pour 104.131.82.128:

    Paquets : envoyés = 4, reçus = 0, perdus = 4 (perte 100%),

stef-ladefense, 2019-12-12:
But ping esp8266.com is ok

 

[C:\Program Files\JPSoft\TCCLE14x64]ping esp8266.com

 

Envoi d’une requête 'ping' sur esp8266.com [119.9.116.93] avec 32 octets de données :

Réponse de 119.9.116.93 : octets=32 temps=347 ms TTL=54

Réponse de 119.9.116.93 : octets=32 temps=353 ms TTL=54

Réponse de 119.9.116.93 : octets=32 temps=348 ms TTL=54

Réponse de 119.9.116.93 : octets=32 temps=346 ms TTL=54

 

Statistiques Ping pour 119.9.116.93:

    Paquets : envoyés = 4, reçus = 4, perdus = 0 (perte 0%),

Durée approximative des boucles en millisecondes :

    Minimum = 346ms, Maximum = 353ms, Moyenne = 348ms

d-a-v, 2019-12-12:
It seems to be a dns problem.

Here (in France too btw), arduino.esp8266.com is 185.199.111.153 but for you it's 104.131.82.128 which is reversed with `download.igrr.me`.

@igrr might help here.

In the meantime you can try
- https://185.199.111.153/stable/package_esp8266com_index.json
- or (as proposed above) https://d-a-v.github.io/esp8266/Arduino/package_esp8266com_index.json

stef-ladefense, 2019-12-12:
thanks, actually on this machine, my dns were broken, now it works

sorry and thank you again

 

*	https://185.199.111.153/stable/package_esp8266com_index.json

closed 2019-12-12 by d-a-v