esp8266/Arduino #6908: Wifi Connection with WPA2 PEAP Network

Vbansal1996, 2019-12-13:
Hi All

I am trying to connect my esp8266 with WPA2 PEAP Network but not able to connect.

I am using Arduino as a compiler.

I had gone through the Espressif Forum and Issue reported related to this but still not able to resolve the issue I am getting.

Here is the code I am using.

#include  <ESP8266WiFi.h>
#include <EEPROM.h>

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
#include "c_types.h"
}

// SSID to connect to
char ssid[] = "test";
char wlpasswd[] = "pass";
char username[] = "test@test.local";
char password[] = "test44";

unsigned char ca_cert[] = { 
  0x2d, 0x2d, 0x2d, 0x2d, 0x2d, 0x42, 0x45, 0x47, 0x49, 0x4e, 0x20, 0x43, 0x45, 0x52, 0x54, 0x49, 
  0x46, 0x49, 0x43, 0x41, 0x54, 0x45, 0x2d, 0x2d, 0x2d, 0x2d, 0x2d, 0x0a, 0x31, 0x37, 0x61, 0x61, 
  0x31, 0x33, 0x62, 0x62, 0x37, 0x31, 0x33, 0x61, 0x37, 0x65, 0x35, 0x35, 0x66, 0x39, 0x66, 0x63, 
  0x61, 0x37, 0x64, 0x62, 0x61, 0x31, 0x30, 0x32, 0x65, 0x33, 0x35, 0x65, 0x65, 0x33, 0x36, 0x38, 
  0x31, 0x39, 0x31, 0x33, 0x0a, 0x2d, 0x2d, 0x2d, 0x2d, 0x2d, 0x45, 0x4e, 0x44, 0x20, 0x43, 0x45, 
  0x52, 0x54, 0x49, 0x46, 0x49, 0x43, 0x41, 0x54, 0x45, 0x2d, 0x2d, 0x2d, 0x2d, 0x2d, 0x0a, 0x00,
};
  

void setup() {

  WiFi.mode(WIFI_STA);
  Serial.begin(115200);
  delay(1000);
  Serial.setDebugOutput(true);
  Serial.printf("SDK version: %s\n", system_get_sdk_version());
  Serial.printf("Free Heap: %4d\n",ESP.getFreeHeap());
  
  // Setting ESP into STATION mode only (no AP mode or dual mode)

  // Clean up to be sure no old data is still inside
  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();
  wifi_station_set_wpa2_enterprise_auth(0);

  wifi_station_set_enterprise_ca_cert(ca_cert, sizeof(ca_cert));
  
  wifi_station_set_enterprise_username((uint8*)username, strlen(username));
  wifi_station_set_enterprise_password((uint8*)password, strlen((char*)password));
    wifi_station_set_wpa2_enterprise_auth(1);

  WiFi.begin(ssid,password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
  }

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
}

Here are the logs I am getting

SDK version: 2.2.2-dev(38a443e)
Free Heap: 47672
WPA2 ENTERPRISE VERSION: [v2.0] disable
WPA2 ENTERPRISE VERSION: [v2.0] enable
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
pm open,type:2 0
state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
pm open,type:2 0
state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
pm open,type:2 0
state: 5 -> 0 (2)

Any help in that ?

devyte, 2019-12-13:
WPA2 Enterprise isn't fully supported by Espressif's SDK, and we don't have any support for it in our core.
There is an issue requesting examples from Espressif, but it hasn't been answered in a long while, @d-a-v is better aware of the details. 
Closing.

closed 2019-12-13 by devyte

Vbansal1996, 2019-12-16:
Hi

I had checked the logs from the Network side for the device request and getting this error
''
Reason Code:			22
	Reason:				The client could not be authenticated  because the Extensible Authentication Protocol (EAP) Type cannot be processed by the server."

Though using same certificate I am using with windows and able to connect.