esp8266/Arduino #6910: Problem in the Upload WAV file to the server

resfandiari, 2019-12-13:
#### Platform 

- Hardware:				[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[60MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem in the Upload WAV file to the server

Hi Guys,

I try to upload WAV file to server and every time I get HTTP Response code: 415 (Unsupported Media Type)
  , what I should do?

```cpp

#include <SD.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <base64.h>
#include <FS.h>

// Replace with your network credentials
const char *ssid = "TP-LINK_E889BB";
const char *password = "12345678";

//file
File myFile;
//server adress
const char *host = "http://95.216.82.173:8001/api/simple_upload/";

void setup()
{
  //set serial port
  Serial.begin(9600);

  //test connect to wifi
  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  
  // uint8_t max = 16;

  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(250);
  }

   Serial.println("");
   Serial.print("Connected to WiFi network with IP Address: ");
   Serial.println(WiFi.localIP());

  //test connect to sd
  Serial.print("Initializing SD card...");
  if (!SD.begin(D0))
  {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");

  //define file
  myFile = SD.open("TEST.WAV", FILE_READ);

  //test correcting file
  if (myFile)
  {
    Serial.println("test file:ok");
  }
  else
  {
    // if the file didn't open, print an error:
    Serial.println("error opening test.WAV");
  }

}
void loop()
{
  static bool cloud_send = false;
  if (!cloud_send)
  {
    cloud_send = true;

    const auto f_size = myFile.size();

    HTTPClient http;
    http.addHeader("Content-Type", "multipart/form-dat");
    http.addHeader("Content-Length", String(f_size, DEC));
    if (!http.begin(host))
    {
      Serial.println("\nfailed to begin http\n");
      return;
    }

    int http_code;

    http_code = http.sendRequest("POST", &myFile, f_size);

    Serial.println(http_code);
    http.writeToStream(&Serial);
    http.end();
    myFile.close();

    if (http_code > 0)
    {

      Serial.print("HTTP Response code: ");
      Serial.println(http_code);

      String response = http.getString();

      Serial.println(response);
    }
    else
    {
      Serial.print('\n');
      Serial.print("Error code: ");
      Serial.println(http_code);
    }
  }
}

```

### Debug Messages

```
Connecting
..............
Connected to WiFi network with IP Address: 192.168.1.106
Initializing SD card...initialization done.
test file:ok
415
{"detail":"Unsupported media type \"text/plain\" in request."}HTTP Response code: 415

```

[image]

earlephilhower, 2019-12-13:
You need to set a `Content-Type` header in the `httpClient` with the appropriate type before sending the `POST`.

This isn't a core problem, so if you need more assistance either https://esp8266.com or https://gitter.im/esp8266/Arduino are the appropriate places to get it.

closed 2019-12-13 by earlephilhower

resfandiari, 2019-12-13:
> You need to set a `Content-Type` header in the `httpClient` with the appropriate type before sending the `POST`.
> 
> This isn't a core problem, so if you need more assistance either https://esp8266.com or https://gitter.im/esp8266/Arduino are the appropriate places to get it.

thanks for  your answer but i set `Content-Type`  and this problem is still 

`http.addHeader("Content-Type", "multipart/form-data");
`

earlephilhower, 2019-12-13:
Try `audio/x-wav` or a variant.  You are not sending form data in your app, just the raw bits of the WAV file.

resfandiari, 2019-12-15:
> Try `audio/x-wav` or a variant. You are not sending form data in your app, just the raw bits of the WAV file.

I solved my problem and create a repository.thank you for your guide

[https://github.com/resfandiari/esp8266_upload_file_multipart_form_data](url)