esp8266/Arduino #6914: Multiple BSSID for single SSID - cannot connect, works in 2.5.2

jiripospisil-telia, 2019-12-16:
On latest library version (2.6.2) Wi-Fi cannot connect to Wi-Fi network, when there is same Wi-Fi name (SSID) but from multiple AP's (BSSID). (Of course with same password and security settings). Does not work even when specifying correct channel and BSSID (MAC address). Latest working version of library is 2.5.2 (there you not even has to specify channel or BSSID and it will connect).

Tested on: 802.11g (WPA2-PSK), NodeMcu 1.0 V3 (LoLin).

Debug output keep in loop writing:

```
reconnect
wifi evt: 1
STA disconnect: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (12)
wifi evt: 1
STA disconnect: 203
```

d-a-v, 2019-12-16:
There are several SDK versions available in the IDE tool menu. You need to use the generic esp8266 board.
Does it work with another version ?

devyte, 2019-12-16:
I have the exact same setup: multiple APs (4) with the same SSID and different BSSIDs, and I'm using phy set to 802.11g. I have one env with 5 ESPs, and another with over 100.
I've had no problems connecting. My testing:
- ping the ESP non-stop
- kill the AP to which the ESP is connected to
  => the ESP reconnects to the next available AP (ping time goes up x10)
- bring the AP back
  => the ESP jumps back to the recovered AP (ping normalizes)

I suggest the following, in case you haven't tried already:
- use persistent(false)
- do a clean install (delete everything, including temp dirs and .arduino15) and rebuild
- wipe the entire flash before downloading the sketch

Closing due to can't reproduce.

closed 2019-12-16 by devyte