esp8266/Arduino #6920: espsoftwareserial 6.4.0 included in arduino core 2.6.3 is faulty

dirkmueller, 2019-12-18:
### Basic Info

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

### Problem Description

The espsoftwareserial release included in the 2.6.3 release is not working for readBytes() anymore, see 
https://github.com/plerup/espsoftwareserial/issues/128

for details. This is a tracker issue in the hope that a 2.6.4 release can be cut when the issue is fixed.

dirkmueller, 2019-12-18:
Upcoming EspSoftwarSerial 6.5.0 release will fix this issue: https://github.com/plerup/espsoftwareserial/commit/80c312774e941c3d9f18f27df73c758605d02cb9

dirkmueller, 2019-12-19:
Has been released: https://github.com/plerup/espsoftwareserial/releases/tag/6.5.0

dirkmueller, 2019-12-19:
@d-a-v could you please help with a 2.6.4 release?

dok-net, 2019-12-19:
WIP again. Don't merge until further notice, please.

dirkmueller, 2019-12-20:
> WIP again. Don't merge until further notice, please.

This has been fixed by merging 6.5.3. please move ahead with a new release

d-a-v, 2019-12-21:
Next release will be 2.7.0.
If you need a board installer I updated the [alpha-release](https://d-a-v.github.io) including the change

dirkmueller, 2019-12-21:
@d-a-v I understand that this is the plan, I was asking whether we can turn around a quick 2.6.x new releaese as there haven't been any major changes in master git since the last release yet.

dok-net, 2019-12-21:
@dirkmueller @d-a-v For any plans to fulfill this wish, I'd also like to mention that I've made an otherwise relevant, but not for EspSoftwareSerial, fix to the included Delegate.h - it's in the Non-ESP Arduino code. In order to not have different versions of Delegate.h in released CoopTask (via Arduino library manager) and EspSoftwareSerial (submoduled on ESP8266 only), perhaps you could give me the benefit of the doubt and merge EspSoftwareSerial 6.5.4 (#6932) instead.

devyte, 2019-12-26:
Like @d-a-v said, next release is 2.7.0. When this was opened we were already passed the point of no return for it.
The referenced PRs are merged. Closing this one.

closed 2019-12-26 by devyte