esp8266/Arduino #6923: Update docs for Updater/etc. explaining how to use gzip uploads

earlephilhower, 2019-12-18:
#6820 was merged w/o docs to allow @ChocolateFrogsNuts to do his magic on combining the XMC flash patch into eboot without delay, but lacked specific documentation updates.

Documentation the capabilities and process eboot now uses.

TD-er, 2019-12-18:
The active sketch should have support for gzipped files right?
Or is the newly flashed sketch executed?

earlephilhower, 2019-12-18:
Yes.  If you try to upload a GZ sketch to the old updater.cpp, it will disallow it as an invalid app and the eboot present in flash won't understand it, either.  You need one uncompressed upload compiled with the new updater.cpp and then you can just send .GZs up.  Or, if using direct serial upload (esptool.py), then a simple compile and upload will get the new eboot compressed support.

TD-er, 2019-12-18:
Good to know, so I will make a new version of our 2-step OTA image including this.
One GZip'ed and one raw .bin.

earlephilhower, 2019-12-18:
Per @pfalcon:

If you give credit to me, please give credit to the original authors too, from uzlib's README:

uzlib is based on:
    tinf library by Joergen Ibsen (Deflate decompression)
    Deflate Static Huffman tree routines by Simon Tatham
    LZ77 compressor by Paul Sokolovsky

Library integrated and maintained by Paul Sokolovsky.

I guess the same wording as above for LittleFS would work, i.e. "uzlib library written by Joergen Ibsen, Simon Tatham, Paul Sokolovsky, licensed under the ZLib license". (That may be not entirely formally correct, but if Joergen or Simon will ever object claim that they wrote that lib, could be rewritten with "based on code by".)

earlephilhower, 2019-12-18:
@AdrianEddy found a bug in the format string setup.
> "0' is not '\0', is this intentional?

closed 2019-12-20 by earlephilhower

HenkV74, 2019-12-21:
Is it correct that this does not work for the spiffs partition? Would be great as 2mb becomes 60kb after zip.

PR #6924 (2019-12-20): Add documentation for compressed OTA, fix bug
Fixes #6923

Documents the user steps needed to do a compressed upload, and notes the
2-step process needed for deploying compressed uploads to the field for
the first time.

Fixes a bug in serial output formatting discovered by @AdrianEddy.

Adds additional contributors for uzlib, per @pfalcon.
commit 9985a329 (2019-12-20): Add documentation for compressed OTA, fix bug (#6924)

* Add documentation for compressed OTA, fix bug

Fixes #6923

Documents the user steps needed to do a compressed upload, and notes the
2-step process needed for deploying compressed uploads to the field for
the first time.

Fixes a bug in serial output formatting discovered by @AdrianEddy.

Adds additional contributors for uzlib, per @pfalcon.

* Update README for esptool.py, too