esp8266/Arduino #6925: ConfigFile example crash.

BOBILLIER, 2019-12-19:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[1.6.2]
- Development Env:		[Arduino IDE 1.8.10]
- Operating System:		[Windows]
### Settings in IDE

- Module:  			[Generic ESP8266 Module|Nodemcu]
- Flash Mode:			[?]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I have test example ESP8266\ConfigFile, and it not work. Compilation and download OK, but fail in runnig mode. I have Continusly hardware reset and  I never see Serial Message other than reset error message. I have test with Generic ESP12 and NodMCU1.0 and have same problem.
I have last version of ArduinoJson Library 6.13.0 and i have select Flash size 4MB  (FS: 2MB/OTA: 1019KB)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
```
Same than Example ESP8266\ConfigFile . No modifications.

```

### Debug Messages

```
12:34:32.637 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
12:34:32.637 -> 
12:34:32.637 -> load 0x4010f000, len 1392, room 16 
12:34:32.637 -> tail 0
12:34:32.637 -> chksum 0xd0
12:34:32.637 -> csum 0xd0
12:34:32.637 -> v3d128e5c
12:34:32.637 -> ~ld
```

BOBILLIER, 2019-12-19:
Hoops!! It's a mistake all work fine. I have make error by add  : while(!Serial); before Serial.begin than after to wait and see sketch message at start when i open serial monitor. This invertion creat reset cyclicly.
Sorry for false alarm.

closed 2019-12-19 by BOBILLIER

konig87nikkinakki, 2019-12-24:
Hi, please don't disappear i have the same problem, but it crashes when i try to read message from Telegram Bot. here the code:

```
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <TelegramBot.h>

#define LED0 3 //D3
#define LED1 1 //D10
#define LED 2 //D4
#define LED3 10 //SD3
#define DS18B21 0  //D9
 
const char* ssid = "SSID";
const char* password = "wpa-key";
const char BotToken[] = "token:mytoken";

WiFiClientSecure net_ssl;
TelegramBot bot (BotToken, net_ssl);

void setup() {

   

  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
     delay(500);
     Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println(WiFi.localIP());
  bot.begin();  
}

void loop() 
{  

digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("Off");
 delay(1000);               // wait for a second
  digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
  Serial.println("On");
  delay(1000);               // wait for a second
 message m = bot.getUpdates(); // Read new messages
delay(1000);
}
```

It crashes with the same code of yours, after blinking led.
if i cancel
 message m = bot.getUpdates(); // Read new messages
it never crashes.

I tried with +8v and big capacitor in Vin, i tried with another ESP (i was afraid it was bad working).
I m using NodeMCU v3 with "esp8266mod" on it.

i receive this error:
WiFi connected
172.31.189.167
spento
acceso

Exception (3):
epc1=0x40100691 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006e7e8 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd70 end: 3fffffc0 offset: 01a0
3fffff10:  007a1200 6ded136d 00000000 3ffee874  
3fffff20:  3fffdad0 00000000 00000020 4010093b  
3fffff30:  3ffee0d0 3ffee8d0 000003e8 3ffee874  
3fffff40:  3fffdad0 3ffee8d0 3fffff88 4020636e  
3fffff50:  40207135 000003e8 3ffee7ec 40206384  
3fffff60:  3fffdad0 00000000 3ffee7ec 40201174  
3fffff70:  3fffdad0 3ffee790 3ffee6b8 402049a8  
3fffff80:  3fffdad0 3ffee790 3ffee7ec 402010b9  
3fffff90:  40208ed8 a7bd1f00 feefeffe 80efeffe  
3fffffa0:  feefeffe 00000000 3ffee834 40206c90  
3fffffb0:  feefeffe feefeffe 3ffe8538 40100bfd  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

can you please help me?

BOBILLIER, 2019-12-24:
Sorry but I can't help you. It's not because we see same error message, than we have same problem.
Like i have said before, my problem come from a "While(!Serial);" instruction who are place before Serial.begin(); I don't see this in your program. Good luck.

konig87nikkinakki, 2019-12-24:
can you please try to run it into your nodemcu and tell me if it happens to you?

jraiver, 2020-03-29:
I get the same error. I connected the LED strip through the mosfet, but the reset starts after the pinMode(5, OUTPUT) on any of the pins to which the tape is connected (there is no such error on other pins)