esp8266/Arduino #6926: Esp board keep on resets( ets Jan  8 2013,rst cause:4, boot mode:(3,6))

4110, 2019-12-19:
I have made a program to continuously get the weather from open weather and serialprint it.
But after 30 min the board resets . Kindly provide me the solution.

AcuarioCat, 2019-12-20:
Very difficult to know with such lack of information from you. 
Check the free heap size ESP.getFreeHeap(); to make sure it is not too low as when under 3K free it can cause a reset. If it's after an amount of time it sounds like a memory leak somewhere in your program.

closed 2019-12-20 by devyte

4110, 2019-12-20:
As my project is to get weather data from open weather 24/7. Everything works fine but after 50 mins the board resets.Here is m code and any suggestions 
// simple Code for reading information from openweathermap.org 

  #include <ESP8266WiFi.h>
  #include <ArduinoJson.h>
  #include <Wire.h>
  #include <SPI.h>
 
  #include <NTPClient.h>
  #include <WiFiUdp.h>

  const char* ssid     = "K";                 // SSID of local network
  const char* password = "kingkise";                    // Password on network
  String APIKEY = "apikey";                                 
  String CityID = "1275339";                                 //Your City ID

  WiFiClient client;
  char servername[]="api.openweathermap.org";              // remote server we will connect to
  String result;

  int  counter = 300;
  int coun=0;                                      

  String weatherDescription ="";
  String weatherLocation = "";
  String Country;
  float Temperature;
  float Humidity;
  float Pressure;
  float Sunrise;
  float Sunset;
  float Speed;
  float Temp_min;
  float Temp_max;
  float Visibility;
  float Wind_angle;

  void setup()
  {
  Serial.begin(115200);
  int cursorPosition=0;
   yield();
   system_soft_wdt_stop();
  Serial.println("Connecting");
  WiFi.begin(ssid, password);
  
             while (WiFi.status() != WL_CONNECTED) 
            {
            {
            delay(1000);
           
            }

             
  }
  
  
  Serial.println("Connected");
  delay(1000);
    }

    void loop()
    
    {
    yield();
    delay(500);
    system_soft_wdt_stop();
    system_soft_wdt_feed();
    if(counter == 300)                                 //Get new data every 10 minutes
    {
      counter = 0;
      delay(500);
      getWeatherData();
      
    }else
    {
      counter++;
      displayWeather(weatherLocation,weatherDescription);
      delay(500);
 
    }
    }
    
    void getWeatherData()                                //client function to send/receive GET request data.
   {
   if (client.connect(servername, 80))   
          {                                         //starts client connection, checks for connection
          client.println("GET /data/2.5/weather?id="+CityID+"&units=metric&APPID="+APIKEY);
          client.println("Host: api.openweathermap.org");
          client.println("User-Agent: ArduinoWiFi/1.1");
          client.println("Connection: close");
          delay(500);
          client.println();
          } 
  else {
         Serial.println("connection failed");        //error message if no client connect
          Serial.println();
       }

  while(client.connected() && !client.available()) 
  delay(500);                                          //waits for data
  while (client.connected() || client.available())    
       { //connected or data available
         char c = client.read();                     //gets byte from ethernet buffer
         result = result+c;
       }

  client.stop();                                      //stop client
  result.replace('[', ' ');
  result.replace(']', ' ');
  
  Serial.println(result);
  delay(500);
  
  char jsonArray [result.length()+1];
  result.toCharArray(jsonArray,sizeof(jsonArray));
  jsonArray[result.length() + 1] = '\0';
  StaticJsonBuffer<1024> json_buf;
  JsonObject &root = json_buf.parseObject(jsonArray);
delay(500);
  if (!root.success())
  {
    Serial.println("parseObject() failed");
  }

  String location = root["name"];
  String country = root["sys"]["country"];
  float temperature = root["main"]["temp"];
  float humidity = root["main"]["humidity"];
  String weather = root["weather"]["main"];
  String description = root["weather"]["description"];
//  float pressure = root["main"]["pressure"];
 // float sunrise = root["sys"]["sunrise"];
 // float sunset = root["sys"]["sunset"];
 // float temp_min = root["main"]["temp_min"];
  //float temp_max = root["main"]["temp_max"];
 // float speed = root["wind"]["speed"];
  //float visibility = root["visibility"];
  //float wind_angle = root["wind"]["deg"];
  weatherDescription = description;
  weatherLocation = location;
  Country = country;
  Temperature = temperature;
  Humidity = humidity;
 // Pressure = pressure;
//  Sunrise = sunrise;
//  Sunset = sunset;
//  Speed = speed;
 // Temp_min = temp_min;
 // Temp_max = temp_max;
 // Visibility = visibility;
 // Wind_angle = wind_angle;
 }

  void displayWeather(String location,String description)
 {
  Serial.println(description);
  delay(500);
 }

AcuarioCat, 2019-12-20:
Ok, several things:
The problem you have is as I suspected, running out of heap. Each time you read from the weather server you are appending to your result string, not overwriting it. A simple fix; in your getWeatherData function add result = ""; at the beginning (or later if you want to preserve the previous data on a failed call). You should have seen the resulting string grow from your serial output, also adding the heap output I said about earlier would have shown this up.

Secondly (to help you learn..)
- Upgrade to ArduinoJson 6 (from arduinojson.org)
- Don't use loads of delay() calls - use a timer to trigger events
- You shouldn't need yield() calls
- You shouldn't need to feed the watchdog timer
- Don't include loads of unnecessary headers. You don't need spi, wire, ntpclient, wifiudp)

4110, 2019-12-20:
Ok I'll check and update

4110, 2019-12-20:
still same issue. The board resets after 80minutes. 
Here is the modiffied code.
  #include <ESP8266WiFi.h>
  #include <ArduinoJson.h>

  const char* ssid     = "K";                 // SSID of local network
  const char* password = "kingkise";                    // Password on network
  String APIKEY = "84b9927477cbd1cd3f3cd586238f0fca";                                 
  String CityID = "1275339";                                 //Your City ID

  WiFiClient client;
  char servername[]="api.openweathermap.org";              // remote server we will connect to
  String result;

  int  counter = 300;
  int coun=0;                                      

  String weatherDescription ="";
  String weatherLocation = "";
  String Country;
  float Temperature;
  float Humidity;
  float Pressure;
  float Sunrise;
  float Sunset;
  float Speed;
  float Temp_min;
  float Temp_max;
  float Visibility;
  float Wind_angle;

  void setup()
  {
  Serial.begin(115200);
  Serial.println("Connecting");
  WiFi.begin(ssid, password);
  
             while (WiFi.status() != WL_CONNECTED) 
            {
            
            delay(1000);  
            }
  
  
  Serial.println("Connected");
  delay(1000);
    }

    void loop()
    
    {
    delay(500);
    if(counter == 300)                                 //Get new data every 10 minutes
    {
      counter = 0;
      delay(500);
      getWeatherData();
      
    }else
    {
      counter++;
      displayWeather(weatherLocation,weatherDescription);
      delay(500);
 
    }
    }
    
    void getWeatherData()                                //client function to send/receive GET request data.
   {
    result="";
   if (client.connect(servername, 80))   
          {                                         //starts client connection, checks for connection
          client.println("GET /data/2.5/weather?id="+CityID+"&units=metric&APPID="+APIKEY);
          client.println("Host: api.openweathermap.org");
          client.println("User-Agent: ArduinoWiFi/1.1");
          client.println("Connection: close");
          delay(500);
          client.println();
          } 
  else {
         Serial.println("connection failed");        //error message if no client connect
          Serial.println();
       }

  while(client.connected() && !client.available()) 
  delay(500);                                          //waits for data
  while (client.connected() || client.available())    
       { //connected or data available
         char c = client.read();                     //gets byte from ethernet buffer
         result = result+c;
       }

  client.stop();                                      //stop client
  result.replace('[', ' ');
  result.replace(']', ' ');
  
  Serial.println(result);
  delay(500);
  
  char jsonArray [result.length()+1];
  result.toCharArray(jsonArray,sizeof(jsonArray));
  jsonArray[result.length() + 1] = '\0';
  StaticJsonBuffer<1024> json_buf;
  JsonObject &root = json_buf.parseObject(jsonArray);
delay(500);
  if (!root.success())
  {
    Serial.println("parseObject() failed");
  }

  String location = root["name"];
  String country = root["sys"]["country"];
  float temperature = root["main"]["temp"];
  float humidity = root["main"]["humidity"];
  String weather = root["weather"]["main"];
  String description = root["weather"]["description"];
//  float pressure = root["main"]["pressure"];
 // float sunrise = root["sys"]["sunrise"];
 // float sunset = root["sys"]["sunset"];
 // float temp_min = root["main"]["temp_min"];
  //float temp_max = root["main"]["temp_max"];
 // float speed = root["wind"]["speed"];
  //float visibility = root["visibility"];
  //float wind_angle = root["wind"]["deg"];
  weatherDescription = description;
  weatherLocation = location;
  Country = country;
  Temperature = temperature;
  Humidity = humidity;
 // Pressure = pressure;
//  Sunrise = sunrise;
//  Sunset = sunset;
//  Speed = speed;
 // Temp_min = temp_min;
 // Temp_max = temp_max;
 // Visibility = visibility;
 // Wind_angle = wind_angle;
 }

  void displayWeather(String location,String description)
 {
  Serial.println(description);
  delay(500);
 }

Pablo2048, 2019-12-20:
It seems like your heap is too fragmented so here are my suggestions:
1. use "Insert code" when placing code here because it is very difficult to read without proper formatting
2. use `result.reserve(2048)` to preserve space for result string
3. never ever use `result = result + c;` - use `result.concat(c)` instead
4. there is no need for `char jsonArray[...]` IMO - just put `result.c_str()` to parseObject
5. spare your RAM using F macro - all your `root["..."]["..."]` has to be `root[F("...")][F("...")]` so the strings are in PROGMEM to save RAM
6. maybe you can also try to use ESPHTTPClient object (https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266HTTPClient) instead of just pure WiFiCilent

AcuarioCat, 2019-12-20:
change your displayWeather for the following and check the heap size:
void displayWeather(String location, String description)
{
    //Serial.println(description);
    Serial.printf("[%d] %s\n", ESP.getFreeHeap(), description.c_str());
    delay(500);
}
I've been running the modified code (with my api key etc) and a refresh of 10 instead of 300 for the last 2+ hours with no resets.
My heap remains pretty steady at 49448

AcuarioCat, 2019-12-20:
..also if you upgrade to V6 of the json code you can get rid of a load of obsolete stuff:
`   
Serial.printf("Res:%s\n", result.c_str());
    delay(500);

    /////////////////////////////////////////////////
    const size_t capacity = JSON_ARRAY_SIZE(3) + 2 * JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(2) + 3 * JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(12) + 340;
    DynamicJsonDocument root(capacity);

    deserializeJson(root, result);

    /////////////////////////////////////////////////

    //char jsonArray[result.length() + 1];
    //result.toCharArray(jsonArray, sizeof(jsonArray));
    //jsonArray[result.length() + 1] = '\0';
    //StaticJsonBuffer<1024> json_buf;
    //JsonObject& root = json_buf.parseObject(jsonArray);
    //delay(500);
    //if (!root.success())
    //{
    //    Serial.println("parseObject() failed");
    //}

    String location = root["name"];
`

4110, 2019-12-20:
can u attach the modification in the full code and send

AcuarioCat, 2019-12-20:
I am using Arduinojson V6
`
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
//#include <Wire.h>
//#include <SPI.h>

//#include <NTPClient.h>
//#include <WiFiUdp.h>

const char* ssid = "kkk"; // SSID of local network
const char* password = "kkkkk"; // Password on network
String APIKEY = "key";
String CityID = "city"; //Your City ID

WiFiClient client;
char servername[] = "api.openweathermap.org"; // remote server we will connect to
String result;

int counter = 0;
int coun = 0;

String weatherDescription = "";
String weatherLocation = "";
String Country;
float Temperature;
float Humidity;
float Pressure;
float Sunrise;
float Sunset;
float Speed;
float Temp_min;
float Temp_max;
float Visibility;
float Wind_angle;

void setup()
{
    Serial.begin(115200);
    int cursorPosition = 0;
    yield();
    system_soft_wdt_stop();
    Serial.println("Connecting");
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED)
    {
        {
            delay(1000);

        }
    }

    Serial.println("WiFi Connected");
    delay(1000);
    getWeatherData();
}

void loop()

{
    yield();
    delay(500);
    system_soft_wdt_stop();
    system_soft_wdt_feed();
    if (counter == 10)                                 //Get new data every 10 minutes
    {
        counter = 0;
        delay(500);
        getWeatherData();

    }
    else
    {
        counter++;
        displayWeather(weatherLocation, weatherDescription);
        delay(500);

    }
}

void getWeatherData()                                //client function to send/receive GET request data.
{
    result = "";
    if (client.connect(servername, 80))
    { //starts client connection, checks for connection
        client.println("GET /data/2.5/weather?id=" + CityID + "&units=metric&APPID=" + APIKEY);
        client.println("Host: api.openweathermap.org");
        client.println("User-Agent: ArduinoWiFi/1.1");
        client.println("Connection: close");
        delay(500);
        client.println();
    }
    else {
        Serial.println("connection failed"); //error message if no client connect
        Serial.println();
    }

    while (client.connected() && !client.available())
        delay(500); //waits for data
    while (client.connected() || client.available())
    { //connected or data available
        char c = client.read(); //gets byte from ethernet buffer
        result = result + c;
    }

    client.stop(); //stop client
    result.replace('[', ' ');
    result.replace(']', ' ');

    Serial.printf("Res:%s\n", result.c_str());
    delay(500);

    /////////////////////////////////////////////////
    const size_t capacity = JSON_ARRAY_SIZE(3) + 2 * JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(2) + 3 * JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(12) + 340;
    DynamicJsonDocument root(capacity);

    deserializeJson(root, result);

    /////////////////////////////////////////////////

    //char jsonArray[result.length() + 1];
    //result.toCharArray(jsonArray, sizeof(jsonArray));
    //jsonArray[result.length() + 1] = '\0';
    //StaticJsonBuffer<1024> json_buf;
    //JsonObject& root = json_buf.parseObject(jsonArray);
    //delay(500);
    //if (!root.success())
    //{
    //    Serial.println("parseObject() failed");
    //}

    String location = root["name"];
    String country = root["sys"]["country"];
    float temperature = root["main"]["temp"];
    float humidity = root["main"]["humidity"];
    String weather = root["weather"]["main"];
    String description = root["weather"]["description"];
    // float pressure = root["main"]["pressure"];
    // float sunrise = root["sys"]["sunrise"];
    // float sunset = root["sys"]["sunset"];
    // float temp_min = root["main"]["temp_min"];
    //float temp_max = root["main"]["temp_max"];
    // float speed = root["wind"]["speed"];
    //float visibility = root["visibility"];
    //float wind_angle = root["wind"]["deg"];
    weatherDescription = description;
    weatherLocation = location;
    Country = country;
    Temperature = temperature;
    Humidity = humidity;
    // Pressure = pressure;
    // Sunrise = sunrise;
    // Sunset = sunset;
    // Speed = speed;
    // Temp_min = temp_min;
    // Temp_max = temp_max;
    // Visibility = visibility;
    // Wind_angle = wind_angle;
}

void displayWeather(String location, String description)
{
    //Serial.println(description);
    Serial.printf("[%d] %s\n", ESP.getFreeHeap(), description.c_str());
    delay(500);
}
`

4110, 2019-12-20:
Thanks for ur help I solved the problem.