esp8266/Arduino #6935: ESP12-E  MySQL_Cursor, get_next_row() shutdown with Soft WDT reset

finexil, 2019-12-22:
Hello everybody, I'm a newby trying to deal with an ESP-12 to detect home temperature, display it on a local display 1,8" and store it in a Maria DB any 10 minutes in a DB. In the Maria DB I created a table where are stored the temperatures on hourly basis  I'd want the ESP would report on the local display.
So I need to manage two different pages on the display, one to display the temperature and humidity, and a second with a bar graph that reports the central configuration.
Whenever I retreive from the Maria DB, the specific data I use this code:

void db_read() {
  if (conn.connected()) {
    MySQL_Cursor *cur_mem_read = new MySQL_Cursor(&conn);
    if(cur_mem_read->execute(INSERT_SQL)) {  **// INSERT_SQL = 'SELECT * FROM temperature.TermostatSetup where room='PT Camera';', ISERT_SQL is declared as global char array char INSERT_SQL[300];**
      **delay(750);** //I'm not sure the delay is of any use I added it to try to solve the problem
      cols = cur_mem_read->get_columns();
      do {
        Serial.println("ROWS");
        **row = cur_mem_read->get_next_row();**
        Serial.println("ROW get_next_row");
        if(row != NULL) {
          Serial.println(row->values[0]);
          for (int i = 1; i < 25; i++) {
            Serial.println(row->values[i]);
            **a**=row->values[i]; // **a is declare as global String, String a="";**
            t_set[i-1]=a.toFloat();
            }
          }
        } while (row != NULL);
      }
    cur_mem_read->close(); 
    delete cur_mem_read;
    }
}
 Everithing works fine but after 3-4 times I call the routine (I switch the page), the ESP resets at the following instruction:
**row = cur_mem_read->get_next_row();** (row as well as cols are declared as globals as follow row_values *row = NULL;
column_names *cols = NULL;)

with the following error:
**11:25:54.186 -> ROWS        -> This is the last Serial.println before the Soft WDT reset**
11:25:56.168 -> 
11:25:56.168 -> Soft WDT reset
11:25:56.168 -> 
11:25:56.168 -> >>>stack>>>
11:25:56.168 -> 
11:25:56.168 -> ctx: cont
11:25:56.202 -> sp: 3ffffd20 end: 3fffffc0 offset: 01b0
11:25:56.235 -> 3ffffed0:  1c000052 74846cae 00000000 00000000  
11:25:56.303 -> 3ffffee0:  3ffeee68 00000000 3ffeffcc 4020549d  
11:25:56.336 -> 3ffffef0:  3ffeee68 00000000 3ffeffcc 4020559a  
11:25:56.405 -> 3fffff00:  00000000 00000004 3ffeee68 00000000  
11:25:56.439 -> 3fffff10:  3ffeee68 3ffeecec 3ffeffcc 402055e4  
11:25:56.505 -> 3fffff20:  3ffeed28 3ffeecec 3ffeffcc 4020134b  
11:25:56.538 -> 3fffff30:  402088e0 3ffeed28 3ffe874a 00000001  
11:25:56.606 -> 3fffff40:  00000000 00000000 00000001 00000000  
11:25:56.640 -> 3fffff50:  3ffeed28 3ffeed28 00000010 402017ab  
11:25:56.673 -> 3fffff60:  007a1200 6fd98ca3 00000300 3ffeeee0  
11:25:56.706 -> 3fffff70:  00000000 3ffeef3c 000003e8 4020686f  
11:25:56.776 -> 3fffff80:  00000000 00000000 00000001 3ffeeee0  
11:25:56.809 -> 3fffff90:  3fffdad0 3ffee7b4 00000001 40201974  
11:25:56.875 -> 3fffffa0:  3fffdad0 00000000 3ffeeea0 40206984  
11:25:56.909 -> 3fffffb0:  feefeffe feefeffe 3ffe8538 40100f51  
11:25:56.978 -> <<<stack<<<

All the Serial-println are in place to understand where the routine hangs.
I suppose I'm running low of something but I really do not know what to do.

Any help is very much welcomed.

Thanks.

devyte, 2019-12-22:
Most likely mem leak.
This is not the right place to ask for general help, this is an issue tracker meant for tracking issues in the core. Please refer such questions to a community forum like esp8266.com or stackoverflow. 
Also, issue template was ignored and not even the stack is decoded. 
Closing due to non compliant.

closed 2019-12-22 by devyte

finexil, 2019-12-22:
Ok I'll do as suggested. Thanks