esp8266/Arduino #6940: Distorted sound with PT8211 I2S-DAC (regression)

tfry-git, 2019-12-24:
Using the cheap PT8211 I2S-DAC, all I get is highly distorted sound output. Scoping the output it looks like the sign bit is either disregarded or misinterpreted. The output is "cut" on the center line, and the lower half is shifted upwards to be above the upper half.

Output was working fine in a February 2018 development snapshot of the core, but is no longer in version 2.6.3. I suppose the issue may have been introduced with https://github.com/esp8266/Arduino/pull/4574 / https://github.com/esp8266/Arduino/issues/4571 .

The culprit appears to be in the following line of code:

https://github.com/esp8266/Arduino/blob/6c2ab2508734e88b06e0a7633445998044fd151c/cores/esp8266/core_esp8266_i2s.cpp#L474

changing this to ```I2SC |= I2SRF | I2SMR | I2SRMS | (div1 << I2SBD) | (div2 << I2SCD);``` (removing "| I2STMS") fixes the issue for me, but probably breaks stuff elsewhere.

earlephilhower, 2019-12-24:
#4571 is the cause.

The old Arduino code was sending data 1 bit too early ("Japanese format") and was broken for the standard I2S ("Philips" IIRC).  Your DAC datasheet specifies Japanese mode, so it's kind of uncommon in that respect compared to reports I've had from ESP8266Audio.

The current mode isn't going to change, and there is not (yet) and API to allow setting the I2S mode, so you'll need to keep your patch.  For 3.0 we're probably going to reset the I2S API completely, making something closer to the Arduino I2S class (which is, itself, kind of hokey).  We're tracking that in #427 (yes, it's that old!).

Closing this one as wontfix for now, and tracked in #427.

closed 2019-12-24 by earlephilhower

NoNamedCat, 2021-11-28:
How can I use the PT8211 with this library now? I cant use the DAC now because the i2s driver changed in the sourcecode and i cant see how i need to modify the library. Please help?

NoNamedCat, 2021-11-28:
Ok... I find it and is working now:

In the line 493 of the file core_esp8266_i2s.cpp change this line to:

i2sc_temp |= I2SRF | I2SMR | I2SRMS | (div1 << I2SBD) | (div2 << I2SCD);

masm31, 2024-06-30:
it work for esp8266 ,How about in esp32?

masm31, 2024-07-02:
> it work for esp8266 ,How about in esp32?
i found , just add out->SetLsbJustified(true);  and work for me