esp8266/Arduino #6942: is there a simple standard example shipping with this repo about how to get UTC time and Serial.println(time)?

dsyleixa, 2019-12-24:
is there a simple standard example shipping with this repo about how to get UTC time and Serial.println(time)?

jandrassy, 2019-12-24:
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino

closed 2019-12-25 by devyte

dsyleixa, 2019-12-25:
Merry Christmas!
@jandrassy:
thanks, bt that s utterly complciaed, I don't understand anything of that monster code. I mean a simple introductional example, just calling a time server, get time, write time to Serial. No timezone, no setting, no extras.

jandrassy, 2019-12-25:
look better. there is a simple setup()  and a simple loop() and a function to print time

dsyleixa, 2019-12-25:
thanks for your efforts, but I think I resign...

d-a-v, 2019-12-27:
@dsyleixa 

```
struct timeval tv;
gettimeofday(&tv, nullptr);
// tv.tv_sec is the number of seconds since EPOCH, it is UTC time
```
You can use [`gmtime()`](https://linux.die.net/man/3/gmtime) to get details about it.
(used in the example: [in the example](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino#L109))

dsyleixa, 2019-12-28:
yes, thank you, I got it working.
but as to the TOP:
is such an example shipping with the esp8266 standard examples?