esp8266/Arduino #6949: WiFiUDP does not receive packets when in AP mode

prampec, 2019-12-26:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			2.6.3
- Development Env:		Arduino IDE and Platformio
- Operating System:		MacOS

### Settings in IDE

- Module:  			Wemos D1 mini r2, ProMini
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

UDP works in station mode, but not in AP mode.
I only tested with receiving packets by ESP8266. The sender side is an Android device. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

#define AP_SSID "myesp"
//#define AP_PASSWD "82668266"
#define WIFI_SSID "mywifi"
#define WIFI_PASSWD "wifipwd"

//#define AP_MODE

#define UDP_PORT 33543
#define UDP_PACKET_LENGTH 10

WiFiUDP Udp;

char packet[UDP_PACKET_LENGTH];

void setup() {
  Serial.begin(115200);
  Serial.println();

  // AP mode
#ifdef AP_MODE
//  WiFi.softAP(AP_SSID, AP_PASSWD);
  WiFi.softAP(AP_SSID);
  Serial.print("IP address:\t");
  Serial.println(WiFi.softAPIP());
#else
  // Station mode
  WiFi.begin(WIFI_SSID, WIFI_PASSWD);

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
#endif

  Udp.begin(UDP_PORT); // start UDP server
  Serial.println("UDP Server Stared");
}

void loop() {
  int packetSize = Udp.parsePacket();

  if (packetSize)
  {
    // receive incoming UDP packets
    Serial.printf("Received %d bytes from %s, port %d\n", packetSize, Udp.remoteIP().toString().c_str(), Udp.remotePort());
    int len = Udp.read(packet, UDP_PACKET_LENGTH);
    if (len > 0)
    {
      packet[len] = 0;
    }
    Serial.printf("UDP packet contents: %s\n", packet);
  }
}
```

gibo77, 2020-01-24:
I have read in the link below that `WifiUdp.h` only works for STA mode. I am not sure if this is the case since I have not tested it.

[https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/udp-class.html](url)

gibo77, 2020-01-24:
On a different note. Try adding `delay(1000);` just after `WiFi.softAP(AP_SSID);`.

prampec, 2020-02-14:
It might have been an Android security issue. Now it works for me.

closed 2020-02-14 by prampec