esp8266/Arduino #6952: regex crashes

Bmooij, 2019-12-27:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12 (ESP8266MOD)
- Core Version:			698ffc34980191ab422918e3aa22ea90b6596766 (latest master)
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Board                            nodemcuv2
- Flash Mode:			?
- Flash Size: 		        4MB
- lwip Variant:                  ?
- Reset Method:		?
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              ?

### Problem Description

When using a simple regex match, I get an user exception (panic/abort/assert)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <regex>
#include <string.h>

void setup(void) {
    Serial.begin(9600);
    delay(5000);

    Serial.print("start");

    // Target sequence
    std::string s = "I am looking for GeeksForGeeks articles";

    // An object of regex for pattern to be searched
    std::regex r("Geek[a-zA-Z]+");

    // flag type for determining the matching behavior
    // here it is for matches on 'string' objects
    std::smatch m;

    // regex_search() for searching the regex pattern
    // 'r' in the string 's'. 'm' is flag for determining
    // matching behavior.
    std::regex_search(s, m, r);

    // for each loop
    for (auto x : m) {
        Serial.print(x.str().c_str());
        Serial.print(" ");
    }
}

void loop(void) {
}

```

### Debug Messages

```
User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: cont
sp: 3ffffc60 end: 3fffffc0 offset: 0000
3ffffc60:  0000002b 00000001 3ffffefc 4021f806  
3ffffc70:  000000fe 00000000 00000000 00000000  
3ffffc80:  00000000 00000000 00000000 0000000e  
3ffffc90:  00000000 00000001 3fffff04 00000000  
3ffffca0:  3fffff84 3fffff3c 3ffffee0 4020367e  
3ffffcb0:  3fffff84 3fffff3c 3ffffee0 40203694  
3ffffcc0:  00000000 3fffff3c 3ffffee0 40220924  
3ffffcd0:  00000000 3fff0d0c 3ffffee0 40202324  
3ffffce0:  0000006b 3fffff08 00000020 40100933  
3ffffcf0:  00000001 00000000 3fffff08 40201d76  
3ffffd00:  00000064 ffffffff 3ffffd50 00000000  
3ffffd10:  00000000 3fffff3c 3ffffee0 402025d5  
3ffffd20:  00000000 3fffff08 3ffffee0 40204c20  
3ffffd30:  0000006b 3fffff08 3ffffee0 4020249e  
3ffffd40:  3fffff08 00000004 3ffffee0 40202107  
3ffffd50:  3fff0d0c 3fffff08 40201194 3fffff24  
3ffffd60:  3fffff08 3fffff3c 3ffffee0 40202626  
3ffffd70:  3fff0d0c 3fffff08 3ffffee0 40202646  
3ffffd80:  3fffff08 3fffff3c 3ffffee0 4020261b  
3ffffd90:  3fff0bbc 3fffff3c 3ffffee0 40202673  
3ffffda0:  3fffff08 00000004 00000004 ffffffff  
3ffffdb0:  3fff0bbc 3fffff3c 3ffffee0 40202673  
3ffffdc0:  3fffff08 00000003 00000003 ffffffff  
3ffffdd0:  3fff0a5c 3fffff3c 3ffffee0 40202673  
3ffffde0:  3fffff08 00000002 00000002 ffffffff  
3ffffdf0:  c3c2c1c0 3fffff3c 3ffffee0 40202673  
3ffffe00:  3fffff08 00000001 00000001 ffffffff  
3ffffe10:  00000047 00000000 3ffffee0 402023c0  
3ffffe20:  00000004 ffffffff 3ffffe60 40210ef5  
3ffffe30:  00000000 00000000 00000000 40201142  
3ffffe40:  00000000 00000000 3ffffee0 3fffff24  
3ffffe50:  3fffff08 00000000 3ffffee0 40202787  
3ffffe60:  3fff0824 feefeffe 40201110 40204b5c  
3ffffe70:  feefeffe feefeffe feefeffe 3ffef368  
3ffffe80:  00000000 3ffe8020 00000020 4010096a  
3ffffe90:  3fffff08 00000000 00000000 ffffffff  
3ffffea0:  feefeffe 3ffef368 00000007 4020b5cc  
3ffffeb0:  3ffe885b 00000000 feff00ff 402177c8  
3ffffec0:  00000001 3ffef368 000000ff 3fffff88  
3ffffed0:  3fffdad0 00000000 3ffe884e 40202860  
3ffffee0:  3fffff84 40204e68 3ffe885b 3ffe885b  
3ffffef0:  00000010 3ffef8d8 0000000e 3fff0a80  
3fffff00:  00000009 3fff09c8 40204e40 3fff0e4c  
3fffff10:  3fff0f3c 3fff0fcc 00000010 00000000  
3fffff20:  3ffef22c 00000000 00000000 3fffff24  
3fffff30:  3fffff24 00000000 00000001 3fff0a5c  
3fffff40:  3fff0a5c 3fff0a6c 3ffefbfc 402182e8  
3fffff50:  3fffdad0 3fffff98 3fffff84 4021830e  
3fffff60:  3fffff80 00000000 3ffeed34 3ffeed9c  
3fffff70:  3fffdad0 00000000 3ffe884e 40202a10  
3fffff80:  00000010 3ffefbfc feefeffe feefeffe  
3fffff90:  3ffe885b 3ffe884e 3fff0990 feefeffe  
3fffffa0:  feefeffe feefeffe 3ffeed5c 402032d4  
3fffffb0:  feefeffe feefeffe 3ffe85b0 40100c35  
<<<stack<<<
```

#### Decoded stack results
```
0x4021f806: std::string::_Rep::_M_dispose(std::allocator  const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 254
0x40220924: __cxxabiv1::__cxa_bad_cast() at /workdir/repo/gcc/libstdc++-v3/libsupc++/eh_aux_runtime.cc line 33
0x40210ef5: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x4020b5cc: wctype at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/ctype/wctype.c line 136
0x402177c8: std::ctype ::_M_convert_to_wmask(char) const at ctype_members.cc line 110
0x402182e8: std::locale::_S_initialize_once() at /workdir/repo/gcc/libstdc++-v3/src/c++98/locale_init.cc line 267
0x4021830e: std::locale::_S_initialize() at /workdir/repo/gcc/libstdc++-v3/src/c++98/locale_init.cc line 280

```

earlephilhower, 2019-12-27:
`std::regex` requires exceptions to be enabled to work properly.  It uses `throws` to manage things, and so can't work w/o using the full C++ exception option in the Arduino IDE.

earlephilhower, 2019-12-27:
When compiled with `Tools->Exceptions->Enabled` it still throws an error because the regex is invalid:
````
start
User exception (panic/abort/assert)
Unhandled C++ exception: regex_error
````

That said, also be aware that in GCC 4.8, the std::regex support is very limited (due to libstdc++).  If you look at the closed PR/bug reports for std::regex you'll get the full rundown.

earlephilhower, 2019-12-27:
https://github.com/esp8266/Arduino/issues/6198#issuecomment-509847266

Bmooij, 2019-12-27:
Thanks for your (very quick) feedback.

> That said, also be aware that in GCC 4.8, the std::regex support is very limited (due to libstdc++). If you look at the closed PR/bug reports for std::regex you'll get the full rundown.

Because of the (current) limitations I switched to the `<regex.h>` library

closed 2019-12-27 by Bmooij