esp8266/Arduino #6962: missing documentation about how to integrated Webserver example "AdvancedAutothentification.ino" into proprietary webserver/website code

dsyleixa, 2019-12-29:
missing documentation about Webserver example AdvancedAutothentification.ino

issue:
missing documentation about how to integrated Webserver example "AdvancedAutothentification.ino" into proprietary webserver/website code

good news: the original example works, and aoriginal proprietary website/webserver code works, just nor together in 1.

Unfortunatey no information provided in the example code how to use this code when a proprietary code which already uses addional configurations about webserver.on() and proprietary website html code, e.g.
in setup():
```
   webserver.on("/", commClients) ;
   webserver.on("/client/client0/", commClients);  // communication with remote wifi clients
   webserver.on("/client/client1/", commClients);
   webserver.on("/client/client2/", commClients);
   webserver.on("/client/client3/", commClients);
   webserver.begin();
```
in loop():
```
   htmlWebsite();     // html code to construct proprietary website
   webserver.handleClient();  
```
If I c+p the snippets from the example into my far bigger webserver+wifiserver code, the login-window can not be loaded in Firefox (latest version): 
continuous wait loop, rotating circle icon.
OTOH, when I delete the entire authentification code, then my website runs like a charm.

esp8266 core 2.6.3
default settings
Arduino IDE 1.8.9

issue:
missing documentation about how to integrated Webserver example "AdvancedAutothentification.ino" into proprietary webserver/website code

danielschlingmeier, 2019-12-29:
Thats a good one! I would like to use the authention, too, but there is no docu at all how to register user or anything...

devyte, 2019-12-29:
Yes, webserver documentation is lacking. How about instead of asking for documentation to integrate our examples into your proprietary code, you contribute with some part of that missing documentation?
Closing as duplicate of #2811.

closed 2019-12-29 by devyte

dsyleixa, 2019-12-30:
You don't seem to et the point:
I get no program working because a reasonable documentation is missing!

OTOH,  what is provoded, does not work with my code which has no authentification yet!

please reopen the issue because the issue is not solved  !!!